zigpy/zha-device-handlers #2300: [Device Support Request] TuyaSiren _TZE204_t1blo2bj

donaldwpage, 2023-03-25:
TuyaSiren _ZTE204_t1blo2bj appears only as an on/off switch and is missing the other end points.
It appears to be because it is using quirk: zhaquicks.tuya.ts0601.siren.TuyaSirenGPP_NoSensors

I expect it to use quirk TuyaSiren2 as that signature describes a similar set of end points and output clusters

<details>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0403",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0006",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE204_t1blo2bj",
  "model": "TS0601",
  "class": "zhaquirks.tuya.ts0601_siren.TuyaSirenGPP_NoSensors"
}
<summary>Device signature</summary>
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.3.6",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.10",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/London",
    "os_name": "Linux",
    "os_version": "5.15.84-v8",
    "supervisor": "2023.03.2",
    "host_os": "Home Assistant OS 9.5",
    "docker_version": "20.10.22",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "hildebrand_glow_ihd": {
[... 319 lines cut ...]
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context or screenshots about the feature request here.

kfriedenberger, 2023-03-26:
+1; same request from me!

kfriedenberger, 2023-03-26:
@donaldwpage solved: https://github.com/zigpy/zha-device-handlers/issues/1379

donaldwpage, 2023-03-26:
yes this is the quirk I believe it should be using, but it isn't and for reasons that I don't understand, when I tried to make it do so, it still chose to use the NoSensors quirk

kfriedenberger, 2023-03-26:
check your logs? one of the imports in this quirk needed to be upgraded given some HA library changes. i had to move "TuyaDPType" to import from zhaquirks.tuya, not zhaquirks.tuya.mcu

See section below...

from zhaquirks.tuya import ATTR_ON_OFF, TuyaManufCluster, TuyaManufClusterAttributes, TuyaDPType
from zhaquirks.tuya.mcu import (
    TUYA_MCU_COMMAND,
    DPToAttributeMapping,
    TuyaAttributesCluster,
    TuyaClusterData,
    TuyaMCUCluster,
)

javicalle, 2023-03-26:
It seems that you are using a local quirk for your devices.
Do any of you mind to create a PR to update the mainstream? If not doing next change in quirks can broke ir again.

Thanks in advanced.

Juju-no-dev, 2023-04-05:
Good morning,

I also just bought this mermaid and I have the same problem as you @donaldwpage

I have been using Home Aassistant for 2 months, mainly with Sonoff and AQARA devices.

@kfriedenberger 
This is the first time I've encountered a difficulty, I didn't really understand how quirks.
can you tell me which file I need to copy to the home assistant?

Thx

Raspberry Pi3:
Home assistant 2023.3.6
Sonoff Zigbee 3.0 USB

asdkjasdfla, 2023-05-14:
Did you find out how to fix this. I am in the same situation

donaldwpage, 2023-05-14:
Unfortunately not and I haven't had time recently to investigate further. Maybe this week

asdkjasdfla, 2023-05-20:
> Unfortunately not and I haven't had time recently to investigate further. Maybe this week

Thanks.

donaldwpage, 2023-06-03:
So I revisited this device today, removing all ZHA quirks and running the latest HASS 2023.5.4
Pairing is fine but it still appears only as a switch.
Manually managing it the the device info page, Manage Zigbee Device, I can change the volume and melody via the TuyaMCUSiren cluster. Having done this, the on off setting works to play the selected melody.
It still doesn't appear in the Siren section of Services but it is good enough for me to switch it on and sound an alarm via automations.

donaldwpage, 2023-06-03:
Minor addition to the above, I installed the Switch As X integration and used that to turn it into a Siren

pabsi, 2023-12-11:
I am curiours: is this going to be looked at or in any sort of roadmap?
I only have the siren entity and also only after I changed the device type of the switch...it'd be super convenient to have the battery level being reported (I already run out of the batteries shipped with the alarm and found out by surprise).. 
I rather not have zigbee2mqtt just for this one device. 

Happy to try custom quirks or whatever myself with some help. I am OK with Python, just never did anything around HA integrations etc. Just need an initial pointer.

Thanks!

masi, 2024-02-18:
Does ZHA really support manufacturer specific clusters and attributes? From what I have seen in the UI so far only clusters and attributes described by the ZCL are show.

dmulcahey, 2024-02-18:
> Does ZHA really support manufacturer specific clusters and attributes? From what I have seen in the UI so far only clusters and attributes described by the ZCL are show.

There are plenty of examples in this repo of this. Here is an example: https://github.com/zigpy/zha-device-handlers/blob/2a39c07e48a2f3cd1956fffac009ec0c64b0f783/zhaquirks/inovelli/__init__.py#L87

is there something in particular that you have an issue with?

masi, 2024-02-19:
Even more awesome would be if the siren entity had the available tones set.

https://www.home-assistant.io/integrations/siren/
https://zigbee.blakadder.com/Neo_NAS-AB02B2.html

masi, 2024-08-19:
How is the new Zigbee quirks v2 API progressing?
Is it possible for non-ZHA-devs to give a shot at a quirk for this device.

MatejKovacic, 2024-09-13:
I also have the same device. Mine is called `Neo NAS-AB02` and has ID `TS0601 by _TZE204_t1blo2bj`.

I bought it from Aliexpress, unfortunately when you buy it, you can not see the exact model

What is the most simple way to change volume and melody?

PR #3741 (2025-01-21): Convert Tuya sirens to v2 quirks
## Proposed change

This converts the existing Tuya sirens to v2 and add some additional ones.

_TZE200_d0yu2xgi Based on: https://github.com/Koenkk/zigbee-herdsman-converters/blob/f42aa5080ee933e8cde9fc08c67a88bf886d71ff/src/devices/neo.ts#L13

_TZE200_t1blo2bj, _TZE204_t1blo2bj, and _TZE204_q76rtoa9 Based on: https://github.com/Koenkk/zigbee-herdsman-converters/blob/f42aa5080ee933e8cde9fc08c67a88bf886d71ff/src/devices/neo.ts#L48C91-L48C107

_TZE204_hcxvyxa5 Based on: https://github.com/Koenkk/zigbee-herdsman-converters/blob/f42aa5080ee933e8cde9fc08c67a88bf886 [... cut]
closed 2025-01-21 by TheJulianJES