zigpy/zha-device-handlers #2302: New variation of A/S MOSZB-140 ? Had to change the Devolco quirk to get mine to work

erikproper, 2023-03-28:
### Discussed in https://github.com/zigpy/zha-device-handlers/discussions/2301

<div type='discussions-op-text'>

<sup>Originally posted by **erikproper** March 27, 2023</sup>
Dear all,

I have four new Frient A/S MOSZB-140 devices, and am running the latest HASSIO.

The official quirk for these motion devices results in the error message: "Fail because output cluster mismatch on at least one endpoint" when home assistant considers using the quirk.

When looking at the the endpoints as identified in the signature in the motion.py versus the endpoints I can see in the diagnostics that there is indeed a mismatch in the output_clusters for 35 and 38. The "Identify.cluster_id" is positioned in 35's output_clusters on my new "Frients".
My manually changing the motion.py accordingly, it is now recognised, and seems to work ... 

Below you will find the relevant part of the diagnosis file from home assistant.

Cheers,
  Erik

 ```
 "data": {
    "ieee": "**REDACTED**",
    "nwk": 14106,
    "manufacturer": "frient A/S",
    "model": "MOSZB-140",
    "name": "frient A/S MOSZB-140",
    "quirk_applied": false,
    "quirk_class": "zigpy.device.Device",
    "manufacturer_code": 4117,
    "power_source": "Battery or Unknown",
    "lqi": 168,
    "rssi": -58,
    "last_seen": "2023-03-27T18:50:52",
    "available": true,
    "device_type": "EndDevice",
    "signature": {
      "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.NONE: 0>, manufacturer_code=4117, maximum_buffer_size=80, maximum_incoming_transfer_size=80, server_mask=10752, maximum_outgoing_transfer_size=80, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=False, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
      "endpoints": {
        "1": {
          "profile_id": 49353,
          "device_type": "0x0001",
          "in_clusters": [
            "0x0003",
            "0x0005",
            "0x0006"
          ],
          "out_clusters": []
        },
        "34": {
          "profile_id": 260,
          "device_type": "0x0107",
          "in_clusters": [
            "0x0000",
            "0x0003",
            "0x0406"
          ],
          "out_clusters": []
        },
        "35": {
          "profile_id": 260,
          "device_type": "0x0402",
          "in_clusters": [
            "0x0000",
            "0x0001",
            "0x0003",
            "0x000f",
            "0x0020",
            "0x0500"
          ],
          "out_clusters": [
            "0x0003",
            "0x000a",
            "0x0019"
          ]
        },
        "38": {
          "profile_id": 260,
          "device_type": "0x0302",
          "in_clusters": [
            "0x0000",
            "0x0003",
            "0x0402"
          ],
          "out_clusters": []
        },
        "39": {
          "profile_id": 260,
          "device_type": "0x0106",
          "in_clusters": [
            "0x0000",
            "0x0003",
            "0x0400"
          ],
          "out_clusters": []
        },
        "40": {
          "profile_id": 260,
          "device_type": "0x0107",
          "in_clusters": [
            "0x0000",
            "0x0003",
            "0x0406"
          ],
          "out_clusters": []
        },
        "41": {
          "profile_id": 260,
          "device_type": "0x0107",
          "in_clusters": [
            "0x0000",
            "0x0003",
            "0x0406"
          ],
          "out_clusters": []
        }
      }
    },
```
</div>

javicalle, 2023-03-28:
There's no need to change the existing one, is enough to add a new quirk with the proper signature, ie:
```python
class MOSZB140_Var02(CustomDevice):
    """Custom device Develco Motion Sensor Pro (variation 02)."""

    manufacturer_id_override = MANUFACTURER

    signature = {
        # <SimpleDescriptor endpoint=1 profile=49353 device_type=1 device_version=1
        # input_clusters=[3, 5, 6] output_clusters=[]>
        # <SimpleDescriptor endpoint=34 profile=260 device_type=263 device_version=0
        # input_clusters=[0, 3, 1030] output_clusters=[]>
        # <SimpleDescriptor endpoint=35 profile=260 device_type=1026 device_version=0
        # input_clusters=[0, 1, 3, 15, 32, 1280] output_clusters=[3, 10, 25]>
        # <SimpleDescriptor endpoint=38 profile=260 device_type=770 device_version=0
        # input_clusters=[0, 3, 1026] output_clusters=[]>
        # <SimpleDescriptor endpoint=39 profile=260 device_type=262 device_version=0
        # input_clusters=[0, 3, 1024] output_clusters=[]>
        # <SimpleDescriptor endpoint=40 profile=260 device_type=263 device_version=0
        # input_clusters=[0, 3, 1030] output_clusters=[]>
        # <SimpleDescriptor endpoint=41 profile=260 device_type=263 device_version=0
        # input_clusters=[0, 3, 1030] output_clusters=[]>
        MODELS_INFO: [(DEVELCO, "MOSZB-140"), (FRIENT, "MOSZB-140")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: 0xC0C9,
                DEVICE_TYPE: 1,
                INPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                ],
                OUTPUT_CLUSTERS: [],
            },
            34: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.OCCUPANCY_SENSOR,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    OccupancySensing.cluster_id,
                ],
                OUTPUT_CLUSTERS: [],
            },
            35: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Identify.cluster_id,
                    BinaryInput.cluster_id,
                    PollControl.cluster_id,
                    IasZone.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Time.cluster_id,
                    Ota.cluster_id,
                ],
            },
            38: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.TEMPERATURE_SENSOR,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    TemperatureMeasurement.cluster_id,
                ],
                OUTPUT_CLUSTERS: [],
[... 96 lines cut ...]
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    OccupancySensing.cluster_id,
                ],
                OUTPUT_CLUSTERS: [],
            },
            41: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.OCCUPANCY_SENSOR,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    OccupancySensing.cluster_id,
                ],
                OUTPUT_CLUSTERS: [],
            },
        }
    }
```

Could you validate that adding the proposed quirk (instead of replacing) works for your devices?
If it works, can you create a pull request with the new version?

Thanks in advanced.

erikproper, 2023-03-28:
Tested the new version. That worked. 
Also used Git to check in the new version, which results in a fork.
But, I can't figure out how to now do a pull request ....

erikproper, 2023-03-28:
I think I managed to issue the pull request ... first timer on that ;-)

erikproper, 2023-03-28:
That didn't work. Apparently a test called "black" failed.

Below the resulting motion.py file. Not sure what is supposed to be wrong here ...

```
"""Develco Motion Sensor Pro."""

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import (
    Basic,
    BinaryInput,
    Identify,
    OnOff,
    Ota,
    PollControl,
    PowerConfiguration,
    Scenes,
    Time,
)
from zigpy.zcl.clusters.measurement import (
    IlluminanceMeasurement,
    OccupancySensing,
    TemperatureMeasurement,
)
from zigpy.zcl.clusters.security import IasZone

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.develco import DEVELCO, FRIENT, DevelcoIasZone, DevelcoPowerConfiguration

MANUFACTURER = 0x1015

class MOSZB140(CustomDevice):
    """Custom device Develco Motion Sensor Pro."""

    manufacturer_id_override = MANUFACTURER

    signature = {
        # <SimpleDescriptor endpoint=1 profile=49353 device_type=1 device_version=1
        # input_clusters=[3, 5, 6] output_clusters=[]>
        # <SimpleDescriptor endpoint=34 profile=260 device_type=263 device_version=0
        # input_clusters=[0, 3, 1030] output_clusters=[]>
        # <SimpleDescriptor endpoint=35 profile=260 device_type=1026 device_version=0
        # input_clusters=[0, 1, 3, 15, 32, 1280] output_clusters=[10, 25]>
        # <SimpleDescriptor endpoint=38 profile=260 device_type=770 device_version=0
        # input_clusters=[0, 3, 1026] output_clusters=[3]>
        # <SimpleDescriptor endpoint=39 profile=260 device_type=262 device_version=0
        # input_clusters=[0, 3, 1024] output_clusters=[]>
        # <SimpleDescriptor endpoint=40 profile=260 device_type=263 device_version=0
        # input_clusters=[0, 3, 1030] output_clusters=[]>
        # <SimpleDescriptor endpoint=41 profile=260 device_type=263 device_version=0
        # input_clusters=[0, 3, 1030] output_clusters=[]>
        MODELS_INFO: [(DEVELCO, "MOSZB-140"), (FRIENT, "MOSZB-140")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: 0xC0C9,
                DEVICE_TYPE: 1,
                INPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                ],
                OUTPUT_CLUSTERS: [],
[... 308 lines cut ...]
            },
            40: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.OCCUPANCY_SENSOR,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    OccupancySensing.cluster_id,
                ],
                OUTPUT_CLUSTERS: [],
            },
            41: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.OCCUPANCY_SENSOR,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    OccupancySensing.cluster_id,
                ],
                OUTPUT_CLUSTERS: [],
            },
        }
    }

```

erikproper, 2023-03-28:
After some "interpretation" of black's report, I noticed a potential typo on line 354.
A , was missing at the end of that line (Ota.cluster_id,).
All tests are passed now, so the pull request should be ready ;-)

javicalle, 2023-03-28:
Fixed in https://github.com/zigpy/zha-device-handlers/pull/2306

Great job!

closed 2023-03-28 by javicalle