zigpy/zha-device-handlers #2303: [Device Support Request] TS0601 by _TZE200_3ejwxpmu (QOXEZY Zigbee CO2 NDIR Sensor)

Orthodux, 2023-03-28:
Is your feature request related to a problem? Please describe.
This is a NDIR CO2 sensor which measures carbon dioxide (additionally humidity and temperature). While I can connect it to ZHA, it generates no entities. Is it possible to integrate?

Describe the solution you'd like
I would like to be able to see all available entities.

<details>

<summary>Device signature</summary>
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE200_3ejwxpmu",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}

</details>

**Additional context**
Add any other context or screenshots about the feature request here.
[image]

pete1019, 2023-03-28:
Mine is called TS0601 by_TZE200_cirvgep4 but its a totally different device.
Mine is a Humidity and Temperature Sensor.

Model Number: ZTH08-E

[image]

pete1019, 2023-03-28:
> 

Cool. I have no idea. Let's find out or hopefully someone helps us.

pete1019, 2023-03-28:
> I put them all in, then I read what quirk was read by ZHA which was ts0601_co2.TuyaNDIRCO2SensorGPP. The other two I removed to clean up the code.

Is there a way to scan TUYA devices, so we can find the Humidity and Temperature and Battery?

Orthodux, 2023-03-28:
So I found out by looking at the **from zhaquirks.tuya.air import** section, that the temperature and humidity were written down. However, when looking further down in the code, I noticed that I couldn't see any reference of the parameters there, only CO2.

```
    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaCO2ManufCluster,
                    TuyaAirQualityCO2,
```

So by adding the parameters found in the **from zhaquirks.tuya.air import** section, the entities were added to HA.

```
    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaCO2ManufCluster,
                    TuyaAirQualityCO2,
                    TuyaAirQualityTemperature,
                    TuyaAirQualityHumidity,
```

<details>

```
"""Tuya Air Quality sensor."""

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import Basic, GreenPowerProxy, Groups, Ota, Scenes, Time

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya.air import (
    TuyaAirQualityCO2,
    TuyaAirQualityFormaldehyde,
    TuyaAirQualityHumidity,
    TuyaAirQualityTemperature,
    TuyaAirQualityVOC,
    TuyaCO2ManufCluster,
)

class TuyaNDIRCO2SensorGPP(CustomDevice):
    """Tuya NIDR CO2 sensor with GPP."""

    signature = {
        # NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)
        # device_version=1
        # SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=81, device_version=1,
        # input_clusters=[0, 4, 5, 61184],
        # output_clusters=[25, 10])
        MODELS_INFO: [
            ("_TZE200_ogkdpgy2", "TS0601"),
            ("_TZE200_3ejwxpmu", "TS0601"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaCO2ManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                # <SimpleDescriptor endpoint=242 profile=41440 device_type=97
                # input_clusters=[]
                # output_clusters=[33]
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaCO2ManufCluster,
                    TuyaAirQualityCO2,
                    TuyaAirQualityTemperature,
                    TuyaAirQualityHumidity,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }
    }

```
</details>

javicalle, 2023-03-28:
I would suggest to enable the debug logs to see if there are some other information (other DPs) there.
Also the vendor or manufacturer site would tell us which capabilities have the device.

Abdull, 2023-09-25:
> There hasn't been any activity on this issue recently. Due to the high number of incoming GitHub notifications, we have to clean some of the old issues, as many of them have already been resolved [...]

shut up, bot.

Syphdias, 2024-01-30:
Hi, I'd like to pick this up and open a PR. I only skimmed the README with the Contribution Guidelines.
However, I did not find documentation to enabling the debug log, like @javicalle suggested. Could you let me know where I can find it?

PR #3023 (2024-03-18): Add Tuya CO2 sensor `_TZE200_3ejwxpmu`
Link to Amazon (German):
https://www.amazon.de/-/en/Detector-Quality-Detects-Temperature-Humidity/dp/B0BN66C263/ref=sr_1_15?crid=3MUBZEIE60NO2&dib=eyJ2IjoiMSJ9.6d4lrKBQtHphFxK7zUt3DdTm_Ubeway0SCqBwe3N9VmOUbwtfoUaBY7EYIElgGet79JXVL4wZKnuB53bVBUxGPxwkw_SHIpzcpDgPUO3sJFhExDSRXtiOsjVjk4eeAYlb7U_HJCzSO7Kl7OL_VvA7oEOXMynBcX72RLMVhJYtWoErYZEe5z_QZAwyscwo_tGkJWOC_X7iFUsyQRgApKy3uk3MruyIwKLgHaCWWEF5Dg.qOA81PqXb7m3qyBNe2kMmhalpioOETFOnR-BQut724g&dib_tag=se&keywords=Tuya+Smart+Zigbee+CO2+Detector+Air+Quality&qid=1709683021&s=diy&sprefix=tuya+smart+zigbee+co2+detecto [... cut]
closed 2024-03-18 by TheJulianJES