zigpy/zha-device-handlers #2307: [Device Support Request] TS0601 by _TZE200_a0syesf5 (Mercator Ikuü Rotary Dimmer Mechanism SSWRM-ZB)

cwhitmorebeirne, 2023-03-29:
**Device support request**

This is a request to please add support for the **Mercator Ikuü Rotary Dimmer Mechanism SSWRM-ZB**. This device is highly desirable for the Australian market as it fits our standard wall plates and replicates the intuitive rotary dimming capabilities of older (non-smart) dimmer switches. It also does not require a neutral wire, so it is simple to retrofit into most Australian homes.

[image]

The mechanism is not currently listed on the [manufacturer's website](https://www.ikuu.com.au/products/trade/accessories/), but is available for purchase through [local online lighting retailers](https://thelightingoutlet.com.au/products/smart-rotary-dimmer-mechanism-wi-fi-or-zigbee?variant=40439652352042).

As per most of the Mercator Ikuü products, the module appears to be based on a Tuya component. The device can pair when I attempt to add it to Home Assistant via SkyConnect. Still, no controls or entities are made available. I have tried to find an existing custom quirk for a rotary Tuya dimmer, but am yet to experiment fully. 

I'm keen to get the Zigbee version of the device working without needing to purchase yet another hub, or without needing to opt for the wifi option of the module, as Zigbee is so much more reliable.

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_a0syesf5",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.3.6",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.10",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Australia/Sydney",
    "os_name": "Linux",
    "os_version": "5.15.90",
    "supervisor": "2023.03.2",
    "host_os": "Home Assistant OS 9.5",
    "docker_version": "20.10.22",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {},
  "integration_manifest": {
    "domain": "zha",
[... 285 lines cut ...]
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x000a": {
            "endpoint_attribute": "time",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

**Additional context**

Below is a copy of the included device specifications sheet for more details: 

[image]
[image]

javicalle, 2023-03-29:
I would suggest to add your device to the `TuyaSingleSwitchDimmer` quirk:
* https://github.com/zigpy/zha-device-handlers/blob/fa60b9d23a3e94b5a164e6aad1f24f4134544daa/zhaquirks/tuya/ts0601_dimmer.py#L37

Enable the local quirk configuration. copy the current file in your folder and add your device to the quirk like:
```python
class TuyaSingleSwitchDimmer(TuyaDimmerSwitch):
    """Tuya touch switch device."""

    signature = {
        MODELS_INFO: [
            ("_TZE200_a0syesf5", "TS0601"),
            ("_TZE200_dfxkcots", "TS0601"),
            ("_TZE200_whpb9yts", "TS0601"),
            ("_TZE200_ebwgzdqq", "TS0601"),
            ("_TZE200_9i9dt8is", "TS0601"),
            ("_TZE200_swaamsoy", "TS0601"),
            ("_TZE200_0nauxa0p", "TS0601"),
            ("_TZE200_la2c2uo9", "TS0601"),
            ("_TZE200_1agwnems", "TS0601"),  # TODO: validation pending?
            ("_TZE200_9cxuhakf", "TS0601"),  # Added for Mercator IKUU SSWM-DIMZ Device
            ("_TZE200_p0gzbqct", "TS0601"),
            ("_TZE200_w4cryh2i", "TS0601"),
        ],
```

Save changes, restart HA and re pair the device.

Once done, if not working, enable the debug logs, operate the device and attach logs here.

cwhitmorebeirne, 2023-03-29:
Amazing! Thank you. That appears to have worked.
I've submitted the pull request for the change. I hope that helps. 😀

wolken-reiter, 2023-06-02:
Hi @cwhitmorebeirne and all ... did you get the SSWRM-ZB working in ZHA with the quirk, so that one can listen to all the device events (on/off, brightness change)?  I want to use the rotary dimmer purely as a wired-in wall controller to then automate the switching and dimming of linked smart downlights via HA (... and perhaps even double-click or long hold for a scene change).
In ZHA I cannot listen to any events from the device other than a single "on" (i.e. push dimmer button).
Can you listen to all dimmer events (on/off, brightness change)?

Thanks.

---

My set-up:
- for Quirk as per link with line added for device
[image]

- With reference in configuration.yaml
[image]

- Device appears as "light" (I also tried converting it to "switch" trying to expose more entities/attributes) ... whichever way while I can get a single switch "on" event, nothing else logs (no second push for switch "off" or even a second switch "on")
[image]

States / Events: not much ...
[image]
Only "switch on" action from rotary button (i.e. press the button) fires below event, only when device was "off".

```yml
event_type: state_changed
data:
  entity_id: light.tze200_a0syesf5_ts0601_light
  old_state:
    entity_id: light.tze200_a0syesf5_ts0601_light
    state: "off"
    attributes:
      supported_color_modes:
        - brightness
      off_with_transition: false
      off_brightness: 40
      friendly_name: _TZE200_a0syesf5 TS0601 Light
      supported_features: 32
    last_changed: "2023-06-02T09:57:03.725431+00:00"
    last_updated: "2023-06-02T09:57:03.725431+00:00"
    context:
      id: 01H1XRM2338XGA77H35R24T1NG
      parent_id: null
      user_id: e00e76b6eeb94c74b4b322577ecba65f
  new_state:
    entity_id: light.tze200_a0syesf5_ts0601_light
    state: "on"
    attributes:
      supported_color_modes:
        - brightness
      color_mode: brightness
      brightness: 40
      off_with_transition: false
      off_brightness: null
      friendly_name: _TZE200_a0syesf5 TS0601 Light
      supported_features: 32
    last_changed: "2023-06-02T09:57:10.093281+00:00"
    last_updated: "2023-06-02T09:57:10.093281+00:00"
    context:
      id: 01H1XRM8ADF6TF5C7HFF9ETDBC
      parent_id: null
      user_id: null
origin: LOCAL
time_fired: "2023-06-02T09:57:10.093281+00:00"
context:
  id: 01H1XRM8ADF6TF5C7HFF9ETDBC
  parent_id: null
  user_id: null
```

daurnimator, 2024-04-24:
@wolken-reiter did you figure this one?

guidok0127, 2024-04-25:
Hi @daurnimator - I did not pursue this further.

In the end I "went all in" with home automation and installed touchscreens in all our rooms (using Lanbon displays with openHasp, see https://www.youtube.com/watch?v=K-EcN0xnJ6w) which are highly flexible and work reliably.

[image]

wolken-reiter, 2024-04-25:
Hi @daurnimator - I did not pursue this further.

In the end I "went all in" with home automation and installed touchscreens in all our rooms (using Lanbon displays with openHasp, see https://www.youtube.com/watch?v=K-EcN0xnJ6w) which are highly flexible and work reliably.

[image]