zigpy/zha-device-handlers #2308: [Device Support Request]  GL-SD-001 by GLEDOPTO

ShogunMan, 2023-03-29:
The dimmer is of good quality and mostly works as it should. However current state seems not to be reported back to HA correctly.

If changing the dimming level from HA the dimmer dims to the correct level, but the level in HA is not updated to the new level.
It seems to be a timeout issue.

Over at zigbee2mqtt they had the same issue, but solved it: [https://github.com/Koenkk/zigbee2mqtt/issues/8349](url)

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4687, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0101",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x1000"
      ],
      "out_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "GLEDOPTO",
  "model": "GL-SD-001",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant Container",
    "version": "2023.3.6",
    "dev": false,
    "hassio": false,
    "virtualenv": false,
    "python_version": "3.10.10",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Oslo",
    "os_name": "Linux",
    "os_version": "5.15.84-v8+",
    "run_as_root": true
  },
  "custom_components": {
    "hacs": {
      "version": "1.31.0",
      "requirements": [
        "aiogithubapi>=22.10.1"
      ]
    },
    "localtuya": {
      "version": "5.0.0",
      "requirements": []
[... 488 lines cut ...]
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

Anyone able to take a look at this?

TheJulianJES, 2023-04-11:
Please install this as a [custom quirk](https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274) and let me know if it works.

<details>
<summary>Custom Quirk (click to expand)</summary>

```python
"""Quirk for GLEDOPTO GL-SD-001."""

from zigpy.profiles import zha
from zigpy.quirks import CustomCluster, CustomDevice
from zigpy.zcl.clusters.general import (
    Basic,
    Groups,
    Identify,
    LevelControl,
    OnOff,
    Ota,
    Scenes,
)
from zigpy.zcl.clusters.lightlink import (
    LightLink,
)
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.kof.kof_mr101z import NoReplyMixin

class LevelControlNoReply(NoReplyMixin, CustomCluster, LevelControl):
    """LevelControl cluster that does not require default responses."""

    void_input_commands = {cmd.id for cmd in LevelControl.commands_by_name.values()}

class GledoptoGlSd001(CustomDevice):
    """Gledopto GL-SD-001 dimmer custom device implementation."""

    signature = {
        MODELS_INFO: [("GLEDOPTO", "GL-SD-001")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.DIMMABLE_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    LevelControl.cluster_id,
                    LightLink.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    Ota.cluster_id,
                ],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.DIMMABLE_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    LevelControlNoReply,
                    LightLink.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    Ota.cluster_id,
                ],
            },
        },
    }
```

</details>

ShogunMan, 2023-04-11:
@TheJulianJES 
Works like a charm. Thank you very much. So nice to have proper support for these quality triac dimmers.
There are probably a couple of clones around that also could use this quirk. I'll see if I can get hold of them.

TheJulianJES, 2023-04-11:
I'll try to get a PR ready for getting this into zha-quirks. Some refactoring needs to be done though

closed 2023-04-11 by TheJulianJES

TheJulianJES, 2023-04-12:
@ShogunMan Fix should land in 2023.4.3