zigpy/zha-device-handlers #2314: [Device Support Request] Smart button by Zemismart - TS0041 by _TZ3000_3oitjgaa

alexvaltchev, 2023-03-31:
I purchase this button from AliExpress: [https://www.aliexpress.com/item/1005003711313613.html?spm=a2g0o.order_list.order_list_main.30.83f91802xjGrTT](url) and I am trying to make it work in HA. It works with Tuya as 3 functions.

1. Single click
2. Double click
3. Click and hold

My HA system has both ZHA and MQTT running in parallel and I tried both.

ZHA recognize it as **switch**, even after adding a custom quirk **ts0041.py** - https://github.com/zigpy/zha-device-handlers/pull/957/commits/cdf38b4e3c8b7796239b3848347e7eb77144a013
[image]

When clicked nothing registers in the logbook.

I then tried with Zigbee2MQTT and it works better there. 
[image]
[image]

In the logbook it shows all (single click, double click and click hold), all 3 states.

Since it comes as this device - https://www.zigbee2mqtt.io/devices/TS0041.html I tried this Blueprint - https://community.home-assistant.io/t/zigbee2mqtt-1-button-scene-switch-ts0041/475324

I set up everything from the Blueprint as it should but when I clicked, nothing happens. 

[image]

I also check this - https://zigbee.blakadder.com/Tuya_TS0041.html where is says it works with ZHA, but is a different photo, not the one I have. But the model shows as **TS0041 by _TZ3000_3oitjgaa**

**Describe the solution you'd like**
I need a quirk for ZHA or MQTT to make this button works in all 3 states. Maybe another blueprint since this does not seem to work.

<details>

<summary>Device signature</summary>

```yaml
Paste the device signature here.
Don't remove the extra line breaks outside the ``` 
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0401",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0006",
        "0x000a"
      ],
      "out_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3000_3oitjgaa",
  "model": "TS0041",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>
[mqtt-940e4d82963367913f262015136efa0e-Smart button white round-aa41b106c091fd6554c096c3a4cc6d4e.json.txt](https://github.com/zigpy/zha-device-handlers/files/11120123/mqtt-940e4d82963367913f262015136efa0e-Smart.button.white.round-aa41b106c091fd6554c096c3a4cc6d4e.json.txt)
</summary>
<summary>
[zha-1be608adfd5dabaaecaf06df5a0aa201-_TZ3000_3oitjgaa TS0041-573e8ab03fa904085f1a51c5d7d72978.json.txt](https://github.com/zigpy/zha-device-handlers/files/11120178/zha-1be608adfd5dabaaecaf06df5a0aa201-_TZ3000_3oitjgaa.TS0041-573e8ab03fa904085f1a51c5d7d72978.json.txt)
</summary>

```yaml
Paste the diagnostic information here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context or screenshots about the feature request here.

MattWestb, 2023-03-31:
Your remote is not known of ZHA.
It need being added in this quirk https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0041.py in the right class that matching the signature and if its not one its need one new device lass is needed.

The switch is the GUI is because tuya have done the firmware wrongly as one light but its being fixed in the quirk then its being loaded.

alexvaltchev, 2023-03-31:
Hello, I added this quirk https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0041.py but nothing changes. The quirk does not seem to work.
[image]
[image]

javicalle, 2023-03-31:
Try adding to your local file that class:
```python
class TuyaSmartRemote0041TI_Var02(TuyaSmartRemote0041TI):
    """Tuya 1-button remote device with time on in."""

    signature = {
        # SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=0x0401, input_clusters=[0, 10, 1, 6], output_clusters=[25]))
        MODEL: "TS0041",
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.IAS_ANCILLARY_CONTROL,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    OnOff.cluster_id,
                    Time.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id],
            },
        },
    }
```

Save changes, restart HA and repair the device.

alexvaltchev, 2023-03-31:
You mean to add this code to the bottom of my ts0041.py file?

alexvaltchev, 2023-03-31:
Nothing has changed. I added the code, remove the device, restarted HA and added device again. No change.

[image]

javicalle, 2023-03-31:
Can you check if device gets the quirk? You can see it in the device info tab.
Also, you can enable the debug logs during the pairing proccess and attach here?

alexvaltchev, 2023-03-31:
Sorry I am noob. Where to click to get device info tab and debug log? Please give me step by step?
[home-assistant_2023-03-31T10-42-30.446Z.log](https://github.com/zigpy/zha-device-handlers/files/11120941/home-assistant_2023-03-31T10-42-30.446Z.log)
[image]
[image]

javicalle, 2023-03-31:
All we need is the "Zigbee Info" from your screenshoot. Sometime also will ask for the diagnostics from the "Download diagnostics" option.
Your device is saying us that no quirk is loaded (well, it is not saying that has a quirk loaded, that is the same).
You have a few errors in the HA startup.
Let's try with another approach.

Remove the current version of `ts0041.py`.
Copy the version from:
* https://github.com/zigpy/zha-device-handlers/blob/0.0.94/zhaquirks/tuya/ts0041.py
Add the proposed class at the bottom.

Save changes, delete any `__pycache__` folder from your local quirk folder, restart HA and re pair the device again.

alexvaltchev, 2023-03-31:
I cleared the code form ts0041.ty and pasted from the link above. I added the code from https://github.com/zigpy/zha-device-handlers/issues/2314#issuecomment-1491653239 to the very bottom (I use Studio code sever) so no option to save. I belive it is saved automatically.

I checked in my quirk folder there is no __pycache__ files. I restarted HA, removed the device and pair it again. No change
[image]

Still looks the same, nothing change. Am I doing something wrong? 
[image]

alexvaltchev, 2023-03-31:
Do I need to remove other .PY files? Seems like there is error with them? How do I fix it? 
[image]

javicalle, 2023-03-31:
`__pycache__` is a folder and I'm not sure if it 'shows' as a hidden folder.
Most of your quirks files have errors. You need to fix them or remove (or rename to .py_ ie).
Can you check if there are errors from the `ts0041.py` file?

alexvaltchev, 2023-04-06:
Hello, sorry I was not available for a few days.
1) If the __pycache__ folder is hidded, how do I find find/delete it? I have File editor and Studio Code Server installed, but dont see "hidden" option,

2. I removed all files from the "custom_zha_quirks" folder. I only left **ts004.py** file. But how do I tell if is loading it? And if NOT, why? Here is the log after starting HA and info of the button.

[image]
[image]
[image]

What is the problem? Please help :(

alexvaltchev, 2023-04-06:
I see this in the log:

Unexpected exception importing custom quirk 'ts0041'
6:29:28 PM – (ERROR) /usr/local/lib/python3.10/site-packages/zhaquirks/__init__.py

But not sure what it is and how to fix it?

javicalle, 2023-04-06:
You will have an error in the HA startup. Look for the error in the homeassistant.log file. Probably some indent or format error in the file.

alexvaltchev, 2023-04-06:
Error/formating in which file ? The ts0041 ?

Here is the log: home-assistant.log
/config/home-assistant.log
[home-assistant.log](https://github.com/zigpy/zha-device-handlers/files/11172482/home-assistant.log)

javicalle, 2023-04-06:
Yes in the file that is trying to load and the only that is in the folder.
Any error in the log?

alexvaltchev, 2023-04-06:
Yes, here are the errors:
[image]

javicalle, 2023-04-06:
> Yes, here are the errors:

And?
Have you read what is saying the error?
**`File "/config/custom_zha_quirks/ts0041.py", line 185`** 
And:
**`IndentationError: expected an indented block after class definition on line 184`**

You can check your `/config/custom_zha_quirks/ts0041.py` at line 185 and will see that the comment is not indented as the proposed from the previous comment:
https://github.com/zigpy/zha-device-handlers/issues/2314#issuecomment-1491653239

alexvaltchev, 2023-04-06:
Yes I read is line 185. It is a comment "- """Tuya 1-button remote device with time on in."""

I removed it and now shows 186. The indentation seems good.
Here is a screenshot 
[image]

```
    """Tuya 1-button remote device with time on in."""
    signature = {
        # SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=0x0401, input_clusters=[0, 10, 1, 6], output_clusters=[25]))
        MODEL: "TS0041",
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.IAS_ANCILLARY_CONTROL,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    OnOff.cluster_id,
                    Time.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id],
            },
        },
    }
```

alexvaltchev, 2023-04-06:
I was able to load the quirk finally. Here it is:
[image]

However, there is nothing showing. I clicked the button and there is no Logbook. Maybe the  quirk is no good? What do I do from now on? Thanks

javicalle, 2023-04-06:
Enable the debug logs (you can do it from the 3 dots next to the "RECONFIGURE" button).
Once enabled, press the physicals buttons (click, double,...) . When you hit the stop debug logs a file will be downloaded. Attach it here.

alexvaltchev, 2023-04-06:
I dont see "Debug log" When I click the 3 dots I get this:

[image]

Here is Download diagnostics

[zha-1be608adfd5dabaaecaf06df5a0aa201-_TZ3000_3oitjgaa TS0041-573e8ab03fa904085f1a51c5d7d72978.json.txt](https://github.com/zigpy/zha-device-handlers/files/11173966/zha-1be608adfd5dabaaecaf06df5a0aa201-_TZ3000_3oitjgaa.TS0041-573e8ab03fa904085f1a51c5d7d72978.json.txt)

Drmatterpl, 2023-09-03:
Hi,

I have the same device, were you able to make it to work? 

I have followed @javicalle  instructions, downloaded the quirk, added section on the end, restarted - I have it discovered using quirk.

adding part of the log - button press and double press are recorded.

[home-assistant_zha_2023-09-03T16-29-07.105Z_1.log](https://github.com/zigpy/zha-device-handlers/files/12506978/home-assistant_zha_2023-09-03T16-29-07.105Z_1.log)

ps. I was able to pair it under zigbee2mqtt and it was discovered correctly, but I need it working on ZHA - this is the only waterproof remote I was able to find. 
[image]

MattWestb, 2023-09-03:
@Drmatterpl Wish IP-class is you needing ?
If PI44 is enough use IKEA RODRET (OnOff is released and OpenClose and Shortcut is on the way) and is Zigbee certificated = dont need any quirks for working OK and is very likely getting firmware updates to new Zigbee standard ZCL R9,

Drmatterpl, 2023-09-03:
I deleted the post before I saw your response, in fact all was good - I followed the instructions above and the button works as expected. 
I need this remote to open a gate-door (release electric lock) - it will be mounted on the fance - totally outside, IP44 is not enough. My wife also wants to be able to turn on the lights in backyard with it. 

Now we have it all working - thanks to you !!!

closed 2024-06-28 by github-actions[bot]