zigpy/zha-device-handlers #2317: [BUG] IKEA Symfonisk Controller send 3-4 times the commands on ZHA

MrHollowPS, 2023-04-03:
**Describe the bug**
After removing and readding the ikea symfonisk controller a few times, ZHA starts to double triple or quadruple commands executed by the end device

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
If i execute a command it should be executed ONCE not 2-3-4 times like it happens now..

**Screenshots**
You can notice on the right that the command is executed 4 times in a row at the same time.
https://community-assets.home-assistant.io/original/4X/6/d/6/6d6fcd9234f9475d2238de6376c454b2ceeca996.png

<details>

<summary></summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4476, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0006",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0020",
        "0x1000"
      ],
      "out_clusters": [
        "0x0003",
        "0x0004",
        "0x0006",
        "0x0008",
        "0x0019",
        "0x1000"
      ]
    }
  },
  "manufacturer": "IKEA of Sweden",
  "model": "SYMFONISK Sound Controller",
  "class": "zhaquirks.ikea.symfonisk.IkeaSYMFONISK1"
}```

</details>

<details>
<!-- Diagnostic information can be acquired by clicking on the "Download Diagnostics" button in the device settings view -->
<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.3.6",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.10",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Bucharest",
    "os_name": "Linux",
    "os_version": "5.15.90",
[... 333 lines cut ...]
            "endpoint_attribute": "lightlink",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context about the problem here.

MattWestb, 2023-04-03:
If you is not need group binding update you device firmware to the latest production version is doing unicast and you is only getting one event for every command from the remote.

MrHollowPS, 2023-04-03:
I don't have group bindings or anything. Also how do i update the firmware from ZHA? I read that updates are done automatically in ZHA.

MrHollowPS, 2023-04-03:
@MattWestb Thanks, very useful.

closed 2023-10-07 by github-actions[bot]