zigpy/zha-device-handlers #2318: [BUG] Unhandled exception TypeError: ZCLCommandDef object is not subscriptable

adamschmalhofer, 2023-04-06:
**Describe the bug**
After upgrading Homeassistant Core to 2023.4.0 my TRADFRI open/close remote from IKEA stopped reacting to button press events. Battery level is still displayed. 

Looking at the logs with `ha core logs` and searching for ikea reviles the TypeError (see "Additional logs" below)  or alternatively via webinterface searching for zigpy

**To Reproduce**
Steps to reproduce the behavior:
1. Have an automation on the button press of the ikea tradfri open/close remote button with a 2023.3
2. Upgrade to 2023.4
3. Press Button; see no reaction from automation
4. check log with ha core logs and search for ikea

**Expected behavior**
Automation reacts to button presses. No TypeError exception thrown

<details>

<summary>Device signature</summary>

IEEE: 68:0a:e2:ff:fe:55:03:03
Nwk: 0xaa5b
Device Type: EndDevice
LQI: 255
RSSI: -32
Zuletzt gesehen: 2023-04-06T12:28:29
Energiequelle: Battery or Unknown
Eigenart: zhaquirks.ikea.opencloseremote.IkeaTradfriOpenCloseRemote

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4476, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0203",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0009",
        "0x0020",
        "0x1000",
        "0xfc7c"
      ],
      "out_clusters": [
        "0x0003",
        "0x0004",
        "0x0008",
        "0x0019",
        "0x0102",
        "0x1000"
      ]
    }
  },
  "manufacturer": "\u0002KE",
  "model": "TRADFRI open/close remote",
  "class": "zhaquirks.ikea.opencloseremote.IkeaTradfriOpenCloseRemote"
}

```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
[... 355 lines cut ...]
  File "/usr/local/lib/python3.10/site-packages/zigpy_deconz/uart.py", line 84, in data_received
    self._api.data_received(frame)
  File "/usr/local/lib/python3.10/site-packages/zigpy_deconz/api.py", line 373, in data_received
    getattr(self, f"_handle_{command.name}")(data)
  File "/usr/local/lib/python3.10/site-packages/zigpy_deconz/api.py", line 505, in _handle_aps_data_indication
    self._app.handle_rx(
  File "/usr/local/lib/python3.10/site-packages/zigpy_deconz/zigbee/application.py", line 448, in handle_rx
    self.packet_received(
  File "/usr/local/lib/python3.10/site-packages/zigpy/application.py", line 913, in packet_received
    self.handle_message(
  File "/usr/local/lib/python3.10/site-packages/zigpy/application.py", line 434, in handle_message
    return sender.handle_message(
  File "/usr/local/lib/python3.10/site-packages/zigpy/device.py", line 367, in handle_message
    self.endpoints[src_ep].handle_message(
  File "/usr/local/lib/python3.10/site-packages/zigpy/endpoint.py", line 224, in handle_message
    handler(hdr, args, dst_addressing=dst_addressing)
  File "/usr/local/lib/python3.10/site-packages/zigpy/zcl/__init__.py", line 375, in handle_message
    self.handle_cluster_request(hdr, args, dst_addressing=dst_addressing)
  File "/usr/local/lib/python3.10/site-packages/zhaquirks/ikea/opencloseremote.py", line 68, in handle_cluster_request
    cmd_name = self.server_commands.get(hdr.command_id, [hdr.command_id])[0]
TypeError: 'ZCLCommandDef' object is not subscriptable

```

</details>

adamschmalhofer, 2023-04-06:
@javicalle I noticed that you are the only one you changed [zhaquirks/ikea/opencloseremote.py](https://github.com/zigpy/zha-device-handlers/commit/dc3f34823b0c719b36f4aa6b343892044bc2361a#diff-4fedfcf9205c7ad380d0cb4a16f2ff66d76d54a7754866b69501ab78abb3909f) this cycle and did some refactoring. Could you have a look? Is it perhaps as simple as removing the `[0]` from `cmd_name = self.server_commands.get(hdr.command_id, [hdr.command_id])[0]` ?

MattWestb, 2023-04-06:
Im getting errors in HA log then using the remote and the error i have seen some user is having with custom quirks:
```
Logger: bellows.ezsp
Source: /usr/local/lib/python3.10/site-packages/bellows/ezsp/__init__.py:359
First occurred: 15:12:55 (16 occurrences)
Last logged: 15:14:28

Exception running handler
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/bellows/ezsp/__init__.py", line 357, in handle_callback
    handler(*args)
  File "/usr/local/lib/python3.10/site-packages/bellows/zigbee/application.py", line 463, in ezsp_callback_handler
    self._handle_frame(*args)
  File "/usr/local/lib/python3.10/site-packages/bellows/zigbee/application.py", line 508, in _handle_frame
    self.packet_received(
  File "/usr/local/lib/python3.10/site-packages/zigpy/application.py", line 913, in packet_received
    self.handle_message(
  File "/usr/local/lib/python3.10/site-packages/zigpy/application.py", line 434, in handle_message
    return sender.handle_message(
  File "/usr/local/lib/python3.10/site-packages/zigpy/device.py", line 367, in handle_message
    self.endpoints[src_ep].handle_message(
  File "/usr/local/lib/python3.10/site-packages/zigpy/endpoint.py", line 224, in handle_message
    handler(hdr, args, dst_addressing=dst_addressing)
  File "/usr/local/lib/python3.10/site-packages/zigpy/zcl/__init__.py", line 375, in handle_message
    self.handle_cluster_request(hdr, args, dst_addressing=dst_addressing)
  File "/usr/local/lib/python3.10/site-packages/zhaquirks/ikea/opencloseremote.py", line 68, in handle_cluster_request
    cmd_name = self.server_commands.get(hdr.command_id, [hdr.command_id])[0]
TypeError: 'ZCLCommandDef' object is not subscriptable
```
So its not one changes we was doing in the IKEA quirks but somthing in zigpy / ZHA.

Sorry @javicalle for pinging you but its one command that  have start making problem with the last update:
https://github.com/zigpy/zha-device-handlers/blob/db3288dd0387331178b6b7d02a5558424aee6124/zhaquirks/ikea/opencloseremote.py#L68

Can you taking one look if you can see what need being changes for getting it working ??

I think our **P** have doing somthing as normal ;-)))))))

javicalle, 2023-04-06:
Hi there,
 my changes are from 2022. I can't see nothing in the last changes that justifies the error (https://github.com/zigpy/zha-device-handlers/commit/5aeee41925323ea4fd499a394e3be2c2d5643f87) 
The class hierarchy doesn't seems to have changed neither:
* https://github.com/zigpy/zigpy/commits/dev/zigpy/zcl/clusters/closures.py
* https://github.com/zigpy/zigpy/commits/dev/zigpy/quirks/__init__.py

Maybe the issue can be related to https://github.com/zigpy/zigpy/commit/2cbb79f4335947a71c039a37ff26e87940dbd7a7

I would suggest to try with:
```python
cmd_name = self.server_commands.get(hdr.command_id, [hdr.command_id]).name
```

Maybe some code validation should be added to ensure not null values but to test now can be enought.

javicalle, 2023-04-06:
Probably something like this:
https://github.com/zigpy/zha-device-handlers/blob/db3288dd0387331178b6b7d02a5558424aee6124/zhaquirks/__init__.py#L118-L126

MattWestb, 2023-04-06:
https://github.com/zigpy/zha-device-handlers/issues/2318#issuecomment-1499249654  is working great i was patching the HA container and all 4 DA looks working well :-))

Was it our @puddly that was doing something not so nice ?  ;-))

MattWestb, 2023-04-06:
@adamschmalhofer If you like fixing your device you can installing the quirk  as local quirk https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274 and copy the original and changing the line 68 ad proposed and restarting HA and all shall working OK.

puddly, 2023-04-06:
Hmm, this should have been caught with a unit test.

@javicalle the above code seems a little redundant with the outer `if` block. If `self.server_commands.get(hdr.command_id) is not None` is `True`, then the command definition exists, so `self.server_commands[hdr.command_id].name` gets you the command name. The `.get(id, [id])[0]` hack shouldn't exist anymore but it's still around in a few places missed by unit tests.

PR #2320 (2023-04-07): Fix IKEA "open/close remote" events
Fixes #2318
Fixes https://github.com/home-assistant/core/issues/90949
Fixes https://github.com/home-assistant/core/issues/90999

Thanks to @javicalle !!
closed 2023-04-07 by TheJulianJES