zigpy/zha-device-handlers #2339: [BUG] TS011F _TZ3000_cfnprab5 (XENON 4AC+2USB power strip) does not pair

MaxMSchneider, 2023-04-14:
**Describe the bug**
Heya,

I've been trying to pair this power strip to transfer from localtuya to zigbee. The quirk is available locally, but I just can't zha to finish the interview. While pairing it sits on my desk, surrounded by at least 5 zigbee devices within 2m.

What's wrong? I simply don't see it. It seems to leave the network before the interview finishes, so it's just caught in a loop.

Thanks for your help! Using this project has really been amazing so far. Kudos to all of you for such a great work!

<details>
<summary>Additional logs (pairing) </summary>

```
Device 0x315d (70:b3:d5:2b:60:01:76:4c) left the network
Device 0x315d (70:b3:d5:2b:60:01:76:4c) left the network
Device 0x315d (70:b3:d5:2b:60:01:76:4c) left the network
Device 0x315d (70:b3:d5:2b:60:01:76:4c) left the network
Device 0x315d (70:b3:d5:2b:60:01:76:4c) left the network
Device 0x315d (70:b3:d5:2b:60:01:76:4c) left the network
Device 0x315d (70:b3:d5:2b:60:01:76:4c) left the network
Device 0x315d (70:b3:d5:2b:60:01:76:4c) left the network
Device 0x315d (70:b3:d5:2b:60:01:76:4c) left the network
Device 0x315d (70:b3:d5:2b:60:01:76:4c) left the network
Device 0x315d (70:b3:d5:2b:60:01:76:4c) left the network
Device 0x315d (70:b3:d5:2b:60:01:76:4c) left the network
Device 0x315d (70:b3:d5:2b:60:01:76:4c) left the network
Device 0x315d (70:b3:d5:2b:60:01:76:4c) left the network
Device 0x315d (70:b3:d5:2b:60:01:76:4c) left the network
Device 0x315d (70:b3:d5:2b:60:01:76:4c) left the network
Device 0x315d (70:b3:d5:2b:60:01:76:4c) left the network
[0x5610](MS01): Device seen - marking the device available and resetting counter
[0x5610](MS01): Update device availability -  device available: True - new availability: True - changed: False
[0x33D5](LWO001): Device seen - marking the device available and resetting counter
[0x33D5](LWO001): Update device availability -  device available: True - new availability: True - changed: False
Device 0x4ff4 (70:b3:d5:2b:60:01:76:4c) joined the network
Device 70:b3:d5:2b:60:01:76:4c changed id (0x315d => 0x4ff4)
[0x4ff4] Scheduling initialization
Tries remaining: 3
[0x4ff4] Requesting 'Node Descriptor'
Tries remaining: 2
[0x4ff4] Extending timeout for 0xba request
Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x4FF4), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.Broadcast: 15>, address=<BroadcastAddress.ALL_ROUTERS_AND_COORDINATOR: 65532>), dst_ep=0, source_route=None, extended_timeout=False, tsn=185, profile_id=0, cluster_id=19, data=Serialized[b'\x1d\xf4OLv\x01`+\xd5\xb3p\x8e'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=180, rssi=-55)
Device 0x4ff4 (70:b3:d5:2b:60:01:76:4c) joined the network
[0x4ff4] Scheduling initialization
[0x4ff4] Canceling old initialize call
Received frame on uninitialized device <Device model=None manuf=None nwk=0x4FF4 ieee=70:b3:d5:2b:60:01:76:4c is_initialized=False> from ep 0 to ep 0, cluster 19: b'\x1d\xf4OLv\x01`+\xd5\xb3p\x8e'
[0x4ff4:zdo] ZDO request ZDOCmd.Device_annce: [0x4FF4, 70:b3:d5:2b:60:01:76:4c, 142]
Tries remaining: 3
[0x4ff4] Requesting 'Node Descriptor'
Tries remaining: 2
[0x4ff4] Extending timeout for 0xbc request
[0x9C33](TS011F): Device seen - marking the device available and resetting counter
[0x9C33](TS011F): Update device availability -  device available: True - new availability: True - changed: False
[0xDA16](TS0505B): Device seen - marking the device available and resetting counter
[0xDA16](TS0505B): Update device availability -  device available: True - new availability: True - changed: False
Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x33D5), src_ep=11, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=104, profile_id=260, cluster_id=6, data=Serialized[b'\x18\xe9\n\x00\x00\x10\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=192, rssi=-52)
[0x33D5:11:0x0006] Received ZCL frame: b'\x18\xe9\n\x00\x00\x10\x00'
[0x33D5:11:0x0006] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=233, command_id=10, *direction=<Direction.Client_to_Server: 1>)
[0x33D5:11:0x0006] Decoded ZCL frame: OnOff:Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=Bool, value=<Bool.false: 0>))])
[0x33D5:11:0x0006] Received command 0x0A (TSN 233): Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=Bool, value=<Bool.false: 0>))])
[0x33D5:11:0x0006] Attribute report received: on_off=<Bool.false: 0>
Tries remaining: 1
[0x4ff4] Extending timeout for 0xbe request
Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x33D5), src_ep=11, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=105, profile_id=260, cluster_id=8, data=Serialized[b'\x18\xea\n\x00\x00 +'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=192, rssi=-52)
[0x33D5:11:0x0008] Received ZCL frame: b'\x18\xea\n\x00\x00 +'
[0x33D5:11:0x0008] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=234, command_id=10, *direction=<Direction.Client_to_Server: 1>)
[0x33D5:11:0x0008] Decoded ZCL frame: LevelControl:Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=uint8_t, value=43))])
[0x33D5:11:0x0008] Received command 0x0A (TSN 234): Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=uint8_t, value=43))])
[0x33D5:11:0x0008] Attribute report received: current_level=43
[0x33D5:11:0x0008]: received attribute: 0 update with value: 43
[... 50 lines cut ...]
Device 0x4ff4 (70:b3:d5:2b:60:01:76:4c) left the network
Device 0x4ff4 (70:b3:d5:2b:60:01:76:4c) left the network
Device 0x4ff4 (70:b3:d5:2b:60:01:76:4c) left the network
Device 0x4ff4 (70:b3:d5:2b:60:01:76:4c) left the network
Device 0x4ff4 (70:b3:d5:2b:60:01:76:4c) left the network
Device 0x4ff4 (70:b3:d5:2b:60:01:76:4c) left the network
Device 0x4ff4 (70:b3:d5:2b:60:01:76:4c) left the network
Device 0x4ff4 (70:b3:d5:2b:60:01:76:4c) left the network
Device 0x4ff4 (70:b3:d5:2b:60:01:76:4c) left the network
Device 0x4ff4 (70:b3:d5:2b:60:01:76:4c) left the network
Device 0x4ff4 (70:b3:d5:2b:60:01:76:4c) left the network
Device 0x4ff4 (70:b3:d5:2b:60:01:76:4c) left the network
Device 0x4ff4 (70:b3:d5:2b:60:01:76:4c) left the network
Device 0x4ff4 (70:b3:d5:2b:60:01:76:4c) left the network
Device 0x4ff4 (70:b3:d5:2b:60:01:76:4c) left the network
Device 0x4ff4 (70:b3:d5:2b:60:01:76:4c) left the network
Device 0x4ff4 (70:b3:d5:2b:60:01:76:4c) left the network
Device 0x4ff4 (70:b3:d5:2b:60:01:76:4c) left the network
Device 0x4ff4 (70:b3:d5:2b:60:01:76:4c) left the network
Device 0x4ff4 (70:b3:d5:2b:60:01:76:4c) left the network
Device 0x4ff4 (70:b3:d5:2b:60:01:76:4c) left the network
Device 0x4ff4 (70:b3:d5:2b:60:01:76:4c) left the network
```

</details>

davet2001, 2023-04-15:
Given the models and problem symptoms are so similar, I expect this is related to #2259

davet2001, 2023-04-15:
@MaxMSchneider does it work ok in zigbee2mqtt?

MattWestb, 2023-04-15:
@davet2001 Can you posting the first 6 numbers of the device IEEE or all of it then i like to knowing witch chip its inside the device.
The first one posted here i cant find so its one unknown chip that is strange.

davet2001, 2023-04-15:
@MattWestb ok, I have posted them on the above linked issue.

MaxMSchneider, 2023-04-15:
> @MaxMSchneider does it work ok in zigbee2mqtt?

@davet2001 Can't say, I've swapped to ZHA recently when I got the skyconnect. I'll check as soon as I'm back home by plugging the sonoff stick in again and reinstalling Z2M.

@MattWestb Will do as soon as I'm back home tonight.

Thanks guys!

MattWestb, 2023-04-15:
Thanks @davet2001 both is TElink chips.

@MaxMSchneider I have you device in you log `70:b3:d5:2b:60:01:76:4c` and the vendor is not registered so its very strange and you dont need posing it one more time but its little mystic.

davet2001, 2023-04-15:
@MattWestb isn't that [HTL Micro](https://maclookup.app/search/result?mac=70:b3:d5:2b:60:01:76:4c)?

MattWestb, 2023-04-15:
I was also getting 2 hints o them but over 10 with `not allocated` so i must looking if they is doing Zigbee radio chips then have not see any  thing from them.

MaxMSchneider, 2023-04-17:
> I was also getting 2 hints o them but over 10 with `not allocated` so i must looking if they is doing Zigbee radio chips then have not see any  thing from them.

Not sure what that means, could you explain?

MaxMSchneider, 2023-04-27:
@davet2001 @MattWestb Could you guys explain which info you need in order to help me solve this? I'm happy to do some testing with the device.

davet2001, 2023-04-27:
@MaxMSchneider the ideal to debug this would be a decrypted wireshark zigbee wireless packet capture during the successful and unsuccessful pairing.

I think to do this you need a second zigbee stick flashed with special firmware.

But I would recommend trying the new version of HA next week. I am running the dev version and find that my problem device now pairs ok in ZHA.

MaxMSchneider, 2023-04-28:
@davet2001 Ok perfect, thanks for the heads up!

I still have a Sonoff ZigBee stick kicking around here, so I'll keep it for your described procedure, if needed in the future. I'm guessing that it might come in handy now with every manufacturer kind of doing as they please in the ZigBee world.

Thanks a lot!

MattWestb, 2023-04-28:
Is ypur Sonoff ZigBee stick TI or EZSP.
If EZSP you can using it with standard NCP for sniffing and if TI you need flashing one sniffing firmware but i dont knowing if its public for all models but for the old CC-2531 its and its working OK.

Z2M link for sniffing with TI coordinators and also EZSP https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html.

EZSP i have putting some links in the Wiki https://github.com/zigpy/zigpy/wiki/Sniffing-Zigbee-traffic.

MaxMSchneider, 2023-04-28:
@MattWestb Thanks for the links, I'll check when I have the time. Maybe later today.

kkossev, 2023-04-28:
> Is your Sonoff ZigBee stick TI or EZSP ? 

These _TZ3000_cfnprab5 are PIA in Hubitat. There is something odd in the Zigbee 3.0 stack implementation in these devices, they do not fall back to ZHA 1.2 as per the specs.  It never worked with the old Hubitat ZHA 1.2 only hubs, then worked for a while with the new C-8 Texas Instruments CC2652P Zigbee chip, then after several updates it stopped working again.

TS011F _TZ3000_okaz9tjs has precisely the same behavior - after a few seconds, it issues a 'leave' command to the Zigbee coordinator and changes its NWK.

At the same time, _TZ3000_cfnprab5 and _TZ3000_okaz9tjs pair immediately and stay connected without any problems when paired to SmartThings V3 hub ( Samsung ARTIK053 Zigbee chip).

MattWestb, 2023-04-28:
Hi @kkossev thanks for creative feed back of your knowledge as normal !!!
Do you have sniffing of paring with ST and not working ones on other hubs ?
Im interesting trying finding out if they is using some magic like we was finding with the normal tuya ones for the TS004F.

Also have some looking how tuya is doing the paring in there ZBGW then its normally EUSP or perhaps Telink in newer versions ?

Perhaps they is needing one good Zigbee 3 handling coordinator and / or router then not like working in HA 1.X mode and the current TI coordinators have most Zigbee 3 handling broken in the firmware but EZSP shall have all in place and working.

kkossev, 2023-04-28:
@MattWestb I can provide a PCAP file when paired to SmartThings hopefully tomorrow, but in my opinion, the problem is on a lower, Zigbee 3.0 transport/connectivity level that I don't have much knowledge of. The problem is different than the Tuya magic initialization, because when I first paired these devices that use Telink chips, they added to ST as a 'thing', i.e. no system or custom driver has sent any special initialization.

There are a lot of issues with the new TI CC2652P Zigbee chip in Hubitat, fixing the connectivity issues with one type of devices brokes the connectivity with other Zigbee devices.

MattWestb, 2023-04-28:
Great !!
Im also not have the deep knowledge but it shall being somthing that the device is not liking in the first frames like node discretion that the Zigbee stack is not liking like the stack if version is not R21 and its leaving or it trying updating the TC- Link key and its not working then it shall leaving if its policy is demanding secure connection to the trust center.

If you dont like posting the paring sniff online i can pinging you from my git and you is getting my email address.

I have one fight with Z2M then nearly all ZB3 was broken and some user is trying getting it back in the TI coordinator firmware without braking the holy grale (old Xiaomi sensors that can handling device aging setting from ZB).

MaxMSchneider, 2023-07-08:
@kkossev @MattWestb do you guys have an update on this issue?

kkossev, 2023-07-09:
@MaxMSchneider sorry I did not follow up on this earlier... 

I generally gave up chasing the Zigbee connectivity issues in Hubitat new C-8 model, that uses a SiliconLabs EFR32 chipset. (I corrected my previous post). There are still a lot of strange problems, not only with this XENON plug extender, but also with other Zigbee devices. The Zigbee coordinator chip is EFR32MG21A020F1024IM32-BR, and if SkyConnect is using same chip, I could expect similar issues.

Do you have any other Zigbee stick that you can test with?

MaxMSchneider, 2023-10-18:
Update: Updated my Skyconnect firmware. No pairing and connectivity issues anymore. See more here: https://github.com/home-assistant/core/issues/98624#issuecomment-1688222346

kkossev, 2023-10-18:
Thank you for the update!

closed 2023-10-20 by MaxMSchneider