zigpy/zha-device-handlers #2342: [BUG] ikea pir sensor e1745 on_time 1800 change attribute

antoweb, 2023-04-15:
**Describe the bug**
How to change attribute to less delay even the default 1800. I would to set it to 30 secs.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

<details>

<summary>Device signature</summary>

```yaml
Paste the device signature here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
Paste the diagnostic information here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context about the problem here.

MattWestb, 2023-04-15:
IKEA motion sensors is having fixed col down in the firmware then they is made for controlling light directly so its not possible if you not cocking on new firmware for it.

antoweb, 2023-04-16:
How to cooking with new firmware?

MattWestb, 2023-04-16:
https://github.com/SiliconLabs/gecko_sdk

closed 2023-10-20 by github-actions[bot]