zigpy/zha-device-handlers #2344: [Device Support Request]

Horley900, 2023-04-16:
**Is your feature request related to a problem? Please describe.**I
I have an issue with tuya 2 gang and 3 gang light switch's, i was running a zha quirk up untill yesterday, this morning it doesnt seem to be having any effect.
the quirk resolved the issue of when pressing one button on the switch  within HA it would turn all lights on and not treat them as seperat entiteis.
This is on a 3 gang switch , TS0013 by _TZ3000_qewo8dlz and a 2 gang switch, TS0012by _TZ3000_18ejxno0.

**Describe the solution you'd like**
each button to trigger its corresponding light seperatly.

<details>

</details>

<details>

<summary>Diagnostic information</summary>

{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.4.4",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.10",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/London",
    "os_name": "Linux",
    "os_version": "5.15.90",
    "supervisor": "2023.04.0",
    "host_os": "Home Assistant OS 9.5",
    "docker_version": "20.10.22",
    "chassis": "vm",
    "run_as_root": true
  },
  "custom_components": {
    "reolink_discovery": {
      "version": "1.1.1",
      "requirements": []
    },
    "dwains_dashboard": {
      "version": "3.4.3",
      "requirements": []
    },
    "alexa_media": {
      "version": "4.6.2",
      "requirements": [
        "alexapy==1.26.5",
        "packaging>=20.3",
        "wrapt>=1.12.1"
      ]
    },
    "garbage_collection": {
      "version": "4.10.2",
      "requirements": [
        "python-dateutil>=2.8.2"
      ]
    },
    "hacs": {
      "version": "1.32.1",
      "requirements": [
        "aiogithubapi>=22.10.1"
      ]
    },
    "tapo_control": {
      "version": "4.2.1",
      "requirements": [
        "pytapo==3.1.7",
        "onvif-zeep-async==1.2.3"
      ]
[... 766 lines cut ...]
                "attribute_name": "off_wait_time",
                "value": 0
              }
            },
            "unsupported_attributes": {}
          }
        },
        "out_clusters": {}
      }
    }
  }
}

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context or screenshots about the feature request here.

javicalle, 2023-04-16:
Have you checked if you have any errors in the logs, even during the HA startup?
From your report it would seem that the device needs the 'Tuya spell' but this should not happen without a device reset.

Try to remove one of your devices from HA, wait a couple of minutes and pair it again.

Horley900, 2023-04-17:
Hi,  this happened around the same time I was configuring a new home network.  The home assistant server was down for awhile.  I have tried repairing several times buy. With no luck,  I've even setup a clean install of home assistant and added just one of these lights,  still same issue.  I'm still very green to home assistant so will dpmy best to check logs.

javicalle, 2023-04-18:
So it doesn't comes from nowhere, you were making changes in ZHA and the quirk seems not being working now.

> I have tried repairing several times buy

Ok, but do you have tried the way I was suggesting?

> Try to remove one of your devices from HA, wait a couple of minutes and pair it again.

You can enable the debug logs during the pairing of the device and attach it here.

Also, you can check the HA logs even during the startup to see if you catch any ZHA related error:
https://www.home-assistant.io/integrations/logger/#viewing-logs

TheJulianJES, 2023-09-27:
Should be fixed with HA Core 2023.9.0 and later.

closed 2023-09-27 by TheJulianJES