zigpy/zha-device-handlers #2345: [BUG] 2 socket TS011F (_TZ3000_jak16dll)

lukyrys, 2023-04-19:
**Describe the bug**
Hello. i have 2 socket smart plug with energy monitor (Model TS011F manufactured by Immax, Zigbee ID: _TZ3000_jak16dll ) - https://zigbee.blakadder.com/Immax_07752L.html
In homeassistant plug shows ok, but when I want to control via HA individual sockets, both are controlled at the same time no separated. By physical button i can control individual socket separately.
Another problem is with monitoring, I connected an approx. 80W soldering iron to it, the wattage is fine but displayed current is high and Summation delivered too.

**Screenshots**
[image]

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x010a",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0702",
        "0x0b04",
        "0xe000",
        "0xe001"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x010a",
      "in_clusters": [
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe001"
      ],
      "out_clusters": []
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3000_jak16dll",
  "model": "TS011F",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
[... 537 lines cut ...]
          }
        },
        "out_clusters": {}
      },
      "242": {
        "device_type": {
          "name": "unknown",
          "id": 97
        },
        "profile_id": 41440,
        "in_clusters": {},
        "out_clusters": {
          "0x0021": {
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

lukyrys, 2023-04-19:
Solved by adding custom quirk

```
class Plug_Immax_2AC(EnchantedDevice):
    """Immax TS0011F 2 outlet plug."""

    signature = {
        MODEL: "TS011F",
        MODELS_INFO: [("_TZ3000_jak16dll", "TS011F")],
        ENDPOINTS: {
            1: {
                # "profile_id": 260,
                # "device_type": "0x010a",
                # "in_clusters": ["0x0000","0x0003","0x0004","0x0005","0x0006","0x0702","0x0b04","0xe000","0xe001"],
                # "out_clusters": ["0x000a","0x0019"]
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_PLUG_IN_UNIT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    Metering.cluster_id,
                    ElectricalMeasurement.cluster_id,
                    TuyaZBE000Cluster.cluster_id,
                    TuyaZBExternalSwitchTypeCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            2: {
                # "profile_id": 260,
                # "device_type": "0x010a",
                # "in_clusters": ["0x0004","0x0005","0x0006","0xe001"],
                # "out_clusters": []
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_PLUG_IN_UNIT,
                INPUT_CLUSTERS: [
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    TuyaZBExternalSwitchTypeCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                # "profile_id": 41440,
                # "device_type": "0x0061",
                # "in_clusters": [],
                # "out_clusters": ["0x0021"]
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }
    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_PLUG_IN_UNIT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaZBOnOffAttributeCluster,
                    TuyaZBMeteringCluster,
                    TuyaZBElectricalMeasurement,
                    TuyaZBE000Cluster,
                    TuyaZBExternalSwitchTypeCluster,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_PLUG_IN_UNIT,
                INPUT_CLUSTERS: [
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaZBOnOffAttributeCluster,
                    TuyaZBExternalSwitchTypeCluster,
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }
```

javicalle, 2023-04-20:
Good job!
Do you mind to create a PR for your device?
You can do it editing the file from the web browser directly:
https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts011f_plug.py

My comments:
- use a more generic quirk name than `Plug_Immax_2AC` (maybe `Plug_2AC_var04`, 05 or whatever)
- use `MODEL` or `MODELS_INFO`. I would suggest to use `MODEL`
- put your class in a nice place inside the file, but don't try to reorganize all the mess.

Thanks in advance.

PR #2348 (2023-04-26): Add Tuya TS011F `_TZ3000_jak16dll` plug variant
NEO LITE Smart two socket https://zigbee.blakadder.com/Immax_07752L.html
Fixes #2345
closed 2023-04-26 by TheJulianJES