zigpy/zha-device-handlers #2347: [Device Support Request] _TZE200_z1tyspqw TS0601 (Tuya ? Sold as TRV606 on aliexpress)

berowicz, 2023-04-20:
Hello,
I got 3 TRV's from aliexpress 

[https://de.aliexpress.com/item/1005004732463260.html?spm=a2g0o.order_list.order_list_main.59.32f55c5fI542OG&gatewayAdapt=glo2deu](url)

They are sold as TRV606 with OLED Display.

[image]

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_z1tyspqw",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.4.5",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.10",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Berlin",
    "os_name": "Linux",
    "os_version": "6.1.21-v8",
    "supervisor": "2023.04.0",
    "host_os": "Home Assistant OS 10.0",
    "docker_version": "23.0.3",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {},
  "integration_manifest": {
    "domain": "zha",
    "name": "Zigbee Home Automation",
    "after_dependencies": [
      "onboarding",
[... 517 lines cut ...]
2023-04-20 20:10:46.912 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8430fc7e'
2023-04-20 20:10:46.916 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame received sendUnicast: [<EmberStatus.SUCCESS: 0>, 213]
2023-04-20 20:10:47.080 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'40bcb1a96b2a15d653944a04aa5592499d4e277ef3ce6767347e'
2023-04-20 20:10:47.081 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8520dd7e'
2023-04-20 20:10:47.087 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame received messageSentHandler: [<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 2660, EmberApsFrame(profileId=0, clusterId=33, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=213), 30, <EmberStatus.SUCCESS: 0>, b'']
2023-04-20 20:10:47.088 DEBUG (MainThread) [bellows.zigbee.application] Received messageSentHandler frame with [<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 2660, EmberApsFrame(profileId=0, clusterId=33, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=213), 30, <EmberStatus.SUCCESS: 0>, b'']
2023-04-20 20:10:47.112 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'50bcb1a9112a15b259b5ca25aa1593499c1dff7989c49874ffdb638b444c7e'
2023-04-20 20:10:47.113 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8610be7e'
2023-04-20 20:10:47.119 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame received incomingMessageHandler: [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=0, clusterId=32801, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=83), 216, -46, 0x0a64, 255, 255, b'\x1d\x00']
2023-04-20 20:10:47.121 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=0, clusterId=32801, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=83), 216, -46, 0x0a64, 255, 255, b'\x1d\x00']
2023-04-20 20:10:47.124 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0A64), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=0, source_route=None, extended_timeout=False, tsn=83, profile_id=0, cluster_id=32801, data=Serialized[b'\x1d\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=216, rssi=-46)
```

</details>

[image]

I hope my thread is right. Its my first post here on github and Im a home assistant beginner but I have connected and configured some devices succesfull but this TRV I cant controll via HA

Im not sure if I posted the debug log right. If no, please tell me what you need

Can anybody help me to get this TRV working ?

Thank you and Regards
Thomas

MattWestb, 2023-04-20:
Its one OME of ritek that one user have making  one working quirk but you need installing it as local quirk and adding you device ID in the signature.
https://github.com/jacekk015/zha_quirks#ts0601_trv_rtitekpy

berowicz, 2023-04-20:
[image]

You mean this one ? This is _TZE200_**a4bpgplm** mine is _TZE200_**z1tyspqw**

Should I replace the "a4bpgplm" to my "z1tyspqw" ? Or what do you mean ?

Thank you

berowicz, 2023-04-21:
After setting the quirk its looking like this:

[image]

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0301",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0004",
        "0x0005",
        "0x000f",
        "0x0201",
        "0x0204",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0006"
      ],
      "out_clusters": []
    },
    "3": {
      "profile_id": 260,
      "device_type": "0x000d",
      "in_clusters": [
        "0x000d"
      ],
      "out_clusters": []
    },
    "4": {
      "profile_id": 260,
      "device_type": "0x000d",
      "in_clusters": [
        "0x000d"
      ],
      "out_clusters": []
    },
    "5": {
      "profile_id": 260,
      "device_type": "0x000d",
      "in_clusters": [
        "0x000d"
      ],
      "out_clusters": []
    },
    "6": {
      "profile_id": 260,
      "device_type": "0x000d",
      "in_clusters": [
        "0x000d"
      ],
      "out_clusters": []
    },
    "7": {
      "profile_id": 260,
      "device_type": "0x000d",
      "in_clusters": [
        "0x000d"
      ],
      "out_clusters": []
    }
  },
  "manufacturer": "_TZE200_z1tyspqw",
  "model": "TS0601",
  "class": "ts0601_trv_rtitek.Rti"
}
```

When I change the temperature, I hear the TRV for a very short time. So its reacting. But in HA the temperature gets back to 15 degree. 
Also the "connected" logo is blinking in the display the whole. Its looking like the pairing process.

I dont have any idea, what Im making wrong

closed 2023-10-25 by github-actions[bot]