zigpy/zha-device-handlers #2349: [Device Support Request] Aqara P1 - MCCGQ13LM

fabiobrolo, 2023-04-21:
Is it possible to include the Aqara P1 door and window sensor in ZHA?
The device is added, however the sensors and binary sensors are not getting updates.
It´s code is MCCGQ13LM and it is supported in Z2M according to https://www.zigbee2mqtt.io/devices/MCCGQ13LM.html

heroclix, 2023-08-23:
Same issue here, were you able to get it to work?

PabloGux, 2023-09-24:
Same issue here! Aqara P1 contact sensor (lumi.magnet.ac01) pairs with ZHA but the binary_sensor does not get updated nor it have battery information. Please add support in ZHA for this.

TheJulianJES, 2023-10-04:
Another issue: 
- https://github.com/zigpy/zha-device-handlers/issues/2349

Subscribe to both threads. I've ordered a T1 sensor that might be very similar to this P1 sensor, so it's possible that I can provide a custom test quirk in the next few days.

PR #2822 (2023-12-27): Add quirk for Aqara P1 contact sensor
## Proposed change

Title says it all. Fixes: #2349, #1295

TODO:
- [ ] Add config entity for detection distance to ZHA

## Additional information

## Checklist

- [ ] The changes are tested and work correctly
- [ ] `pre-commit` checks pass / the code has been formatted using Black
- [ ] Tests have been added to verify that the new code works
closed 2023-12-27 by dmulcahey