zigpy/zha-device-handlers #2352: [Device Support Request] Tuya TS0203 _TZ3000_au1rjicn Window / Door / Contact Sensor

b2un0, 2023-04-24:
**Is your feature request related to a problem? Please describe.**
The mentioned device above joined the network and than leave the network after a seconds

https://www.aliexpress.com/item/1005004984273100.html

**Describe the solution you'd like**
working Contact Senor

<details>

<summary>Device signature</summary>

```yaml
Paste the device signature here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>

<summary>

</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant Container",
    "version": "2023.4.6",
    "dev": false,
    "hassio": false,
    "virtualenv": false,
    "python_version": "3.10.10",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Berlin",
    "os_name": "Linux",
    "os_version": "6.1.21-v8+",
    "run_as_root": true
  },
  "custom_components": {
    "zha_toolkit": {
      "version": "v0.8.38",
      "requirements": [
        "packaging>=20.8",
        "pytz"
      ]
    },
    "xiaomi_miio_fan": {
      "version": "2022.8.0.0",
      "requirements": [
        "construct==2.10.56",
        "python-miio>=0.5.12"
      ]
    },
    "localtuya": {
      "version": "5.0.0",
      "requirements": []
    },
    "avanza_stock": {
      "version": "1.2.0",
      "requirements": [
        "pyavanza==0.6.0"
      ]
    },
    "hacs": {
      "version": "1.32.1",
      "requirements": [
        "aiogithubapi>=22.10.1"
[... 349 lines cut ...]
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x000a": {
            "endpoint_attribute": "time",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

[home-assistant_zha_2023-04-24T12-26-36.672Z.log](https://github.com/zigpy/zha-device-handlers/files/11310402/home-assistant_zha_2023-04-24T12-26-36.672Z.log)

</details>

[image]

tOOmki, 2023-05-16:
I have 2pcs _TZ3000_au1rjicn
One reports this:
[image]

And the second:
[image]

Qianli-Ma, 2023-06-28:
I believe that the contact sensor is working as expected buy not the tamper sensor?

Malte1992, 2023-06-29:
Is it possible to integrate the sensor via zha in the meantime?

MattWestb, 2023-06-29:
> I believe that the contact sensor is working as expected buy not the tamper sensor?

My experience with LIDL devices is that the tamper is being sent but (Z)HA is not using it and if temper is open and one alarm is being triggered and then the tamper is being retested the alarm reset is being lost and is staying in alarm mode in the HA GUI.
I think ZHA need implanting the tamper and fixing the logic of the reset of it and the alarm for getting it working OK.

All my LIDL devices is working OK if getting the tamper being in sync for not braking the alarm commands sent.

EnricoBettella, 2023-08-11:
> I believe that the contact sensor is working as expected buy not the tamper sensor?

That's the exact issue i'm facing with my ZD08 sensors. Anyway, I think the op problem is with slightly different devices, since those of the aliexpress page linked above don't have the tamper sensor. They seems to appear always as "TS0203" though.

cramer52, 2023-10-15:
I added two TS0203 _TZ3000_au1rjicn contact sensors roughly 6 months ago without any issues and they  have been working flawless since. Today I tried to add a new one (bought from the same vendor) and it's not showing the ias_zone anymore just two 'Opening' sensors which are not responding. I get a "failed to bind the 'ias_zone' cluster" in the Debug.

MaxMSchneider, 2023-10-18:
@cramer52 @b2un0 I had the same issue, also with a Skyconnect. Just today I stumbled across this fix, which made all my pairing problems vanish. A firmware update did the trick.

See more here: https://github.com/home-assistant/core/issues/98624#issuecomment-1688222346

MaxMSchneider, 2023-10-19:
> @cramer52 @b2un0 I had the same issue, also with a Skyconnect. Just today I stumbled across this fix, which made all my pairing problems vanish. A firmware update did the trick.
> 
> See more here: https://github.com/home-assistant/core/issues/98624#issuecomment-1688222346

Add: The pairing issues have been resolved, but I've been a bit overly excited. The door sensor still drops out of the network after some time.

MattWestb, 2023-10-19:
@MaxMSchneider  Look what parent is having in the network it can that is not liking it and trying jumping to one other that also is not working well (like Philips HUE motions is doing all the time).

cramer52, 2023-10-22:
@MaxMSchneider @MattWestb  I use a sonoff cc2652p dongle and with the latest firmware update I paired it (didn't work right away but with a few tries it finally paired) and it seems to be working fine for the past day with several restarts etc.

Jumperm, 2024-03-17:
I'm using the Sonoff Dongle E with EmberZnet NCP 7.4.1.0 and having still problems to add the TS0203 _TZ3000_au1rjicn. It's been detected but the initial configuration fail.
Any solution or alternativ door/windows sensors for ZHA with EZSP?!

Jumperm, 2024-03-21:
After forgot to configure it again it works. I put some batteries instead of battery packs in it. Maybe that's a big problem...

closed 2024-09-24 by github-actions[bot]