zigpy/zha-device-handlers #2354: [Device Support Request] Onesti EasyCodeTouch doorlock

1bard, 2023-04-24:
This works as a lock, but I'm missing battery status.

The device is powered by 3xAA batteries, and as far as I know, mains power is not an option on this model.
The correct battery information seems to be available, but not used.
I believe the issue is that the device reports power source as mains/DC.

<details>

<summary>Device signature</summary>

```{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered: 140>, manufacturer_code=4660, maximum_buffer_size=108, maximum_incoming_transfer_size=127, server_mask=11264, maximum_outgoing_transfer_size=127, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "11": {
      "profile_id": 260,
      "device_type": "0x000a",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0101",
        "0xfea2"
      ],
      "out_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "Onesti Products AS",
  "model": "EasyCodeTouch",
  "class": "zigpy.device.Device"
}
```

</details>

<details>
<summary>Diagnostic information</summary>

```"data": {
    "ieee": "**REDACTED**",
    "nwk": 34502,
    "manufacturer": "Onesti Products AS",
    "model": "EasyCodeTouch",
    "name": "Onesti Products AS EasyCodeTouch",
    "quirk_applied": false,
    "quirk_class": "zigpy.device.Device",
    "manufacturer_code": 4660,
    "power_source": "Mains",
    "lqi": 255,
    "rssi": -81,
    "last_seen": "2023-04-24T20:21:03",
    "available": true,
    "device_type": "EndDevice",
    "signature": {
      "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered: 140>, manufacturer_code=4660, maximum_buffer_size=108, maximum_incoming_transfer_size=127, server_mask=11264, maximum_outgoing_transfer_size=127, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=False, *is_security_capable=False)",
      "endpoints": {
        "11": {
          "profile_id": 260,
          "device_type": "0x000a",
          "in_clusters": [
            "0x0000",
            "0x0001",
            "0x0003",
            "0x0004",
            "0x0005",
            "0x0101",
            "0xfea2"
          ],
          "out_clusters": [
            "0x0019"
          ]
        }
      }
    },
    "active_coordinator": false,
    "entities": [
      {
        "entity_id": "button.onesti_products_as_easycodetouch_identify",
        "name": "Onesti Products AS EasyCodeTouch"
      },
      {
        "entity_id": "lock.onesti_products_as_easycodetouch_doorlock",
        "name": "Onesti Products AS EasyCodeTouch"
      }
    ],
    "neighbors": [],
    "routes": [],
    "endpoint_names": [
      {
        "name": "DOOR_LOCK"
      }
    ],
    "user_given_name": null,
    "device_reg_id": "25cbd8823980768b84764226c66c5449",
    "area_id": "bf6016a107d611eb8e53e78571c4c004",
    "cluster_details": {
      "11": {
        "device_type": {
          "name": "DOOR_LOCK",
          "id": 10
        },
        "profile_id": 260,
        "in_clusters": {
          "0x0000": {
            "endpoint_attribute": "basic",
            "attributes": {
              "0x0004": {
                "attribute_name": "manufacturer",
                "value": "Onesti Products AS"
              },
              "0x0005": {
                "attribute_name": "model",
                "value": "EasyCodeTouch"
              },
              "0x0007": {
                "attribute_name": "power_source",
                "value": 4
              }
            },
            "unsupported_attributes": {
              "0x000e": {
                "attribute_name": "product_label"
              }
            }
          },
          "0x0001": {
            "endpoint_attribute": "power",
            "attributes": {
              "0x0020": {
                "attribute_name": "battery_voltage",
                "value": 45
              },
              "0x0021": {
                "attribute_name": "battery_percentage_remaining",
                "value": 200
              },
              "0x0031": {
                "attribute_name": "battery_size",
                "value": 3
              },
              "0x0033": {
                "attribute_name": "battery_quantity",
                "value": 3
              },
              "0x0034": {
                "attribute_name": "battery_rated_voltage",
                "value": 15
              }
            },
            "unsupported_attributes": {
              "0x0000": {
                "attribute_name": "mains_voltage"
              },
              "0x0041": {
                "attribute_name": "battery_2_percentage_remaining"
              },
              "0x0061": {
                "attribute_name": "battery_3_percentage_remaining"
              },
              "0xfffe": {
                "attribute_name": "attr_reporting_status"
              }
            }
          },
          "0x0003": {
            "endpoint_attribute": "identify",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0005": {
            "endpoint_attribute": "scenes",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0004": {
            "endpoint_attribute": "groups",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0101": {
            "endpoint_attribute": "door_lock",
            "attributes": {
              "0x0000": {
                "attribute_name": "lock_state",
                "value": 1
              },
              "0x0003": {
                "attribute_name": "door_state",
                "value": 4
              }
            },
            "unsupported_attributes": {
              "0x0020": {
                "attribute_name": "enable_logging"
              },
              "0x0021": {
                "attribute_name": "language"
              },
              "0x0006": {
                "attribute_name": "open_period"
              },
              "0x0004": {
                "attribute_name": "door_open_events"
              },
              "0x002a": {
                "attribute_name": "enable_inside_status_led"
              },
              "0x0034": {
                "attribute_name": "zigbee_security_level"
              }
            }
          },
          "0xfea2": {
            "endpoint_attribute": "manufacturer_specific",
            "attributes": {},
            "unsupported_attributes": {}
          }
        },
        "out_clusters": {
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
```

</details>

[image]
[image]
[image]

1bard, 2023-05-03:
I've created a quirk and PR.
By removing the MainsPowered-flag from node discriptor, battery percentage became available in HA.

[image]

Strauman, 2023-06-11:
It also sends open info (user ID, RFID/FingerPring/Code) to 0x101 and pin code to 0x100 on the doorlock endpoint. Also on wrong entry of code. It'd be awesome to have triggers on failed attempts and also know and know who opened the door when it opens.

Strauman, 2023-07-21:
See also the related [z2m](https://github.com/Koenkk/zigbee2mqtt/issues/17205) issue.

linust, 2023-08-02:
It seems version 1.32.2 of z2m has included support. The issue references three relevant commits to herdsman-converters. 

I'm not familiar enough with zha &  herdsman to navigate a port. But perhaps someone else is?

issue :
https://github.com/Koenkk/zigbee2mqtt/issues/17205

referencing the following pulls:

https://github.com/Koenkk/zigbee-herdsman-converters/pull/6009

https://github.com/Koenkk/zigbee-herdsman-converters/pull/6010

https://github.com/Koenkk/zigbee-herdsman-converters/pull/6024

1bard, 2023-08-04:
I've been using my power quirk for a few months now and battery data reported by my lock is useless.
It reports 100% and 4.5 volts for a long time, then it jumps all over the place the day before batteries die.
Not really worth implementing.

Open info and failed attempts sounds more interesting.

DagBertelsen, 2023-08-08:
I have a friend that has this lock in zigbee2mqtt, he say the same thing. The battery goes from 100% ->0% without a warning and is not of use.

oleost, 2024-01-21:
My device has stopped updating Locked / unlocked. It always report’s unlocked. 

And when I press lock it does lock the door but does not update state to locked. 

Is used to work. 

[image]

PR #4138 (2025-06-24): Expose extra entities for Nimly Locks
## Proposed change

- Last action source
- Last action
- Last action user
- Last pin code
- Auto relock
- Sound volume

[image]

## Additional information

Fixes #3095
Fixes #2354

Related to #3465, #3580, #3457

## Checklist

- [x] The changes are tested and work correctly
- [x] `pre-commit` checks pass / the code has been formatted using Black
- [ ] Tests have been added to verify that the new code works
closed 2025-06-24 by TheJulianJES