zigpy/zha-device-handlers #2357: [Device Support Request] TUYA movement sensor _TZ3040_bb6xaihh (TS0202)

domusonline, 2023-04-26:
**Is your feature request related to a problem? Please describe.**
New PIR sensor. Physically similar to others already supported. Device doesn't seem to be correctly recognized. It shows two entities:
a) The IAS zone which seems to work fine
b) One shows as "Opening" and seems to be permanently with the value "Closed"

I cannot understand what is the second entity (could be illuminance, but I'm not sure as the AliExpress link shows it as "optional".
Apparently it also doesn't change the sensitivity level and the "on_time" or "keep_time" seen in other sensors. The attribute change doesn't raise an error, but when reading the value it's always the same

**Describe the solution you'd like**
Take full advantage of the sensor, maybe via a quirk. I understand it may be difficult as I can't really say for sure which are the sensor's functionalities

<details>

<summary>Device signature</summary>

```yaml
  {
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0402",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0004",
        "0x0500"
      ],
      "out_clusters": [
        "0x0004",
        "0x0006",
        "0x000a",
        "0x0019",
        "0x1000"
      ]
    }
  },
  "manufacturer": "_TZ3040_bb6xaihh",
  "model": "TS0202",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
  "data": {
    "ieee": "**REDACTED**",
    "nwk": 48888,
    "manufacturer": "_TZ3040_bb6xaihh",
    "model": "TS0202",
    "name": "_TZ3040_bb6xaihh TS0202",
    "quirk_applied": false,
    "quirk_class": "zigpy.device.Device",
    "manufacturer_code": 4417,
    "power_source": "Battery or Unknown",
    "lqi": 100,
    "rssi": -75,
    "last_seen": "2023-04-26T19:32:44",
    "available": true,
    "device_type": "EndDevice",
    "signature": {
      "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
      "endpoints": {
        "1": {
          "profile_id": 260,
          "device_type": "0x0402",
          "in_clusters": [
            "0x0000",
            "0x0001",
            "0x0003",
            "0x0004",
            "0x0500"
          ],
          "out_clusters": [
            "0x0004",
            "0x0006",
            "0x000a",
            "0x0019",
            "0x1000"
          ]
        }
      }
    },
    "active_coordinator": false,
    "entities": [
      {
        "entity_id": "sensor.pir_tz3040_bb6xaihh_ts0202_battery",
        "name": "_TZ3040_bb6xaihh TS0202"
      },
      {
        "entity_id": "binary_sensor.pir_tz3040_bb6xaihh_ts0202_iaszone",
        "name": "_TZ3040_bb6xaihh TS0202"
      },
      {
        "entity_id": "button.pir_tz3040_bb6xaihh_ts0202_identify",
        "name": "_TZ3040_bb6xaihh TS0202"
      },
      {
        "entity_id": "binary_sensor.pir_tz3040_bb6xaihh_ts0202_opening",
        "name": "_TZ3040_bb6xaihh TS0202"
      }
    ],
    "neighbors": [],
    "routes": [],
    "endpoint_names": [
      {
        "name": "IAS_ZONE"
      }
    ],
    "user_given_name": "PIR_TZ3040_bb6xaihh TS0202",
    "device_reg_id": "a9b28978a896661747ce62a61605ea35",
    "area_id": "geral",
    "cluster_details": {
      "1": {
        "device_type": {
          "name": "IAS_ZONE",
          "id": 1026
        },
        "profile_id": 260,
        "in_clusters": {
          "0x0001": {
            "endpoint_attribute": "power",
            "attributes": {
              "0x0020": {
                "attribute_name": "battery_voltage",
                "value": 30
              },
              "0x0021": {
                "attribute_name": "battery_percentage_remaining",
                "value": 200
              }
            },
            "unsupported_attributes": {
              "0x0031": {
                "attribute_name": "battery_size"
              },
              "0x0033": {
                "attribute_name": "battery_quantity"
              }
            }
          },
          "0x0003": {
            "endpoint_attribute": "identify",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0004": {
            "endpoint_attribute": "groups",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0500": {
            "endpoint_attribute": "ias_zone",
            "attributes": {
              "0x0000": {
                "attribute_name": "zone_state",
                "value": 1
              },
              "0x0001": {
                "attribute_name": "zone_type",
                "value": 13
              },
              "0x0002": {
                "attribute_name": "zone_status",
                "value": 1
              },
              "0x0010": {
                "attribute_name": "cie_addr",
                "value": [
                  247,
                  121,
                  202,
                  254,
                  255,
                  129,
                  142,
                  88
                ]
              },
              "0x0013": {
                "attribute_name": "current_zone_sensitivity_level",
                "value": 1
              }
            },
            "unsupported_attributes": {}
          },
          "0x0000": {
            "endpoint_attribute": "basic",
            "attributes": {
              "0x0004": {
                "attribute_name": "manufacturer",
                "value": "_TZ3040_bb6xaihh"
              },
              "0x0005": {
                "attribute_name": "model",
                "value": "TS0202"
              }
            },
            "unsupported_attributes": {}
          }
        },
        "out_clusters": {
          "0x0004": {
            "endpoint_attribute": "groups",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0006": {
            "endpoint_attribute": "on_off",
            "attributes": {
              "0x0000": {
                "attribute_name": "on_off",
                "value": 0
              }
            },
            "unsupported_attributes": {
              "0x0000": {
                "attribute_name": "on_off"
              },
              "0x4003": {
                "attribute_name": "start_up_on_off"
              },
              "0x4001": {
                "attribute_name": "on_time"
              },
              "0x4002": {
                "attribute_name": "off_wait_time"
              }
            }
          },
          "0x1000": {
            "endpoint_attribute": "lightlink",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x000a": {
            "endpoint_attribute": "time",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Screenshot showing the recognized entities:

[image]

CzarskyM, 2023-04-28:
Hi,
is it possible to add it into Z2M? I've tried this [https://github.com/Koenkk/zigbee2mqtt/issues/9757#issuecomment-1242909728](url) but it doesn't work for me :/

posreg, 2023-05-10:
Good Day!
I have the same sensor, and it is wrongly recognized, same as for the author

DB entry if this helps
```
{"id":13,"type":"EndDevice","ieeeAddr":"0xa4c1380f478120e0","nwkAddr":45992,"manufId":4417,"manufName":"_TZ3040_bb6xaihh","powerSource":"Battery","modelId":"TS0202","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[1,3,4,1280,0],"outClusterList":[4,6,4096,25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0202","manufacturerName":"_TZ3040_bb6xaihh","powerSource":3,"zclVersion":3,"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":""}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b0014da4b33","zoneState":1}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b0014da4b33","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0}],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":491743244},"lastSeen":1683695415537,"defaultSendRequestWhen":"immediate"}```

erelke, 2023-08-18:
@domusonline How get you entities? Mine is aded, but configarition dosn't finish, and no one entites aded, and no one events is available.

julianjwong, 2023-10-02:
I just received mine from AliExpress. Unfortunately was extremely difficult to pair with ZHA and my Sonoff Zigbee USB 3.0 dongle. Literally had to have the sensor next to the dongle and it pair with LQI 25. Never triggered motion events afterwards. Wondering if it's a compatibility issue or just a bad sensor.

scharrrfi, 2023-10-02:
@julianjwong I paired mine with a second Zigbee stick and z2m. This works fine. 
With ZHA it has never worked as expected. Motion was always triggered and never reset. When I reset it with an automation after x seconds, it triggered automatically again after few seconds.

MattWestb, 2023-10-02:
Sonoff Zigbee USB 3.0 dongle need one of the latest firmware (2023.X) for handling end device OK and its the same if running ZHA or Z2M.

srett, 2024-02-02:
I've got 3 of these (and a _TZ3000_msl6wxk9 which seems to be identical apart from the LED being red instead of blue), and run into the issue mentioned, namely that sometimes one of the devices goes unresponsive, and the state stays as "motion detected" for hours in HA. It seems the device just drops off the zigbee network, as least the "Last seen" field in HA doesn't update anymore, but for some reason I still see these lines in the log (and only these, at least grepping for the device address BD22):

```
2024-02-02 17:12:28.926 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xBD22](TS0202): Device seen - marking the device available and resetting counter
2024-02-02 17:12:28.926 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xBD22](TS0202): Update device availability -  device available: True - new availability: True - changed: False
```

I'm new to HA so no idea what makes it say "Device seen"?

This is the log at the very time the device stopped working. 17:10:44 is the last seen time displayed in HA - it's exactly the time it reported motion for the very last time:

```
2024-02-02 17:10:44.651 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(timestamp=datetime.datetime(2024, 2, 2, 16, 10, 44, 651089, tzinfo=datetime.timezone.utc), src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x9C87), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=0, profile_id=260, cluster_id=1280, data=Serialized[b'\t\x13\x00\x01\x00\x00\x00\x00\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=29, non_member_radius=0, lqi=76, rssi=None)
2024-02-02 17:10:44.651 DEBUG (MainThread) [zigpy.zcl] [0x9C87:1:0x0500] Received ZCL frame: b'\t\x13\x00\x01\x00\x00\x00\x00\x00'
2024-02-02 17:10:44.652 DEBUG (MainThread) [zigpy.zcl] [0x9C87:1:0x0500] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), tsn=19, command_id=0, *direction=<Direction.Client_to_Server: 1>)
2024-02-02 17:10:44.653 DEBUG (MainThread) [zigpy.zcl] [0x9C87:1:0x0500] Decoded ZCL frame: IasZone:status_change_notification(zone_status=<ZoneStatus.Alarm_1: 1>, extended_status=<bitmap8: 0>, zone_id=0, delay=0)
2024-02-02 17:10:44.653 DEBUG (MainThread) [zigpy.zcl] [0x9C87:1:0x0500] Received command 0x00 (TSN 19): status_change_notification(zone_status=<ZoneStatus.Alarm_1: 1>, extended_status=<bitmap8: 0>, zone_id=0, delay=0)
2024-02-02 17:10:44.654 DEBUG (MainThread) [homeassistant.components.zha.core.cluster_handlers] [0x9C87:1:0x0500]: Updated alarm state: 1
2024-02-02 17:10:44.654 DEBUG (MainThread) [zigpy.zcl] [0x9C87:1:0x0500] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=19, command_id=<GeneralCommand.Default_Response: 11>, *direction=<Direction.Client_to_Server: 1>)
2024-02-02 17:10:44.655 DEBUG (MainThread) [zigpy.zcl] [0x9C87:1:0x0500] Sending reply: Default_Response(command_id=0, status=<Status.SUCCESS: 0>)
2024-02-02 17:10:44.655 DEBUG (MainThread) [zigpy_znp.zigbee.application] Sending packet ZigbeePacket(timestamp=datetime.datetime(2024, 2, 2, 16, 10, 44, 655279, tzinfo=datetime.timezone.utc), src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x9C87), dst_ep=1, source_route=None, extended_timeout=False, tsn=19, profile_id=260, cluster_id=1280, data=Serialized[b'\x18\x13\x0b\x00\x00'], tx_options=<TransmitOptions.ACK: 1>, radius=0, non_member_radius=0, lqi=None, rssi=None)
2024-02-02 17:10:44.655 DEBUG (MainThread) [zigpy_znp.api] Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0x9C87), DstEndpoint=1, DstPanId=0x0000, SrcEndpoint=1, ClusterId=1280, TSN=19, Options=<TransmitOptions.ACK_REQUEST|SUPPRESS_ROUTE_DISC_NETWORK: 48>, Radius=0, Data=b'\x18\x13\x0B\x00\x00')
2024-02-02 17:10:44.660 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataRequestExt.Rsp(Status=<Status.SUCCESS: 0>)
2024-02-02 17:10:44.779 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.SrcRtgInd.Callback(DstAddr=0xBD22, Relays=[0x0FD8])
2024-02-02 17:10:44.780 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=1280, SrcAddr=0xBD22, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=83, SecurityUse=<Bool.false: 0>, TimeStamp=8216383, TSN=0, Data=b'\x09\xC5\x00\x01\x00\x00\x00\x00\x00', MacSrcAddr=0x0FD8, MsgResultRadius=28)
2024-02-02 17:10:44.781 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(timestamp=datetime.datetime(2024, 2, 2, 16, 10, 44, 781290, tzinfo=datetime.timezone.utc), src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xBD22), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=0, profile_id=260, cluster_id=1280, data=Serialized[b'\t\xc5\x00\x01\x00\x00\x00\x00\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=28, non_member_radius=0, lqi=83, rssi=None)
[... cut ...]
What *might* be contributing to the wonkyness is bad connection to the battery. One of the sensors turned on when I unboxed it, that is, flashed its light a couple times, but didn't go into pairing mode via the button. I took out the battery several times to no avail, until I noticed that the LED would have some flicker to it, as if the battery is low. I carefully bent the battery pins so that they'd push against the battery some more. After that, the device paired just fine. But maybe not good enough sometimes?

Kerber0ss, 2024-04-03:
> I've got 3 of these (and a _TZ3000_msl6wxk9 which seems to be identical apart from the LED being red instead of blue), and run into the issue mentioned, namely that sometimes one of the devices goes unresponsive, and the state stays as "motion detected" for hours in HA. It seems the device just drops off the zigbee network, as least the "Last seen" field in HA doesn't update anymore, but for some reason I still see these lines in the log (and only these, at least grepping for the device address BD22):
> 
> ```
> 2024-02-02 17:12:28.926 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xBD22](TS0202): Device seen - marking the device available and resetting counter
> 2024-02-02 17:12:28.926 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xBD22](TS0202): Update device availability -  device available: True - new availability: True - changed: False
> ```
> 
> I'm new to HA so no idea what makes it say "Device seen"?
> 
> This is the log at the very time the device stopped working. 17:10:44 is the last seen time displayed in HA - it's exactly the time it reported motion for the very last time:
> 
> ```
> 2024-02-02 17:10:44.651 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(timestamp=datetime.datetime(2024, 2, 2, 16, 10, 44, 651089, tzinfo=datetime.timezone.utc), src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x9C87), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=0, profile_id=260, cluster_id=1280, data=Serialized[b'\t\x13\x00\x01\x00\x00\x00\x00\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=29, non_member_radius=0, lqi=76, rssi=None)
> 2024-02-02 17:10:44.651 DEBUG (MainThread) [zigpy.zcl] [0x9C87:1:0x0500] Received ZCL frame: b'\t\x13\x00\x01\x00\x00\x00\x00\x00'
> 2024-02-02 17:10:44.652 DEBUG (MainThread) [zigpy.zcl] [0x9C87:1:0x0500] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), tsn=19, command_id=0, *direction=<Direction.Client_to_Server: 1>)
> 2024-02-02 17:10:44.653 DEBUG (MainThread) [zigpy.zcl] [0x9C87:1:0x0500] Decoded ZCL frame: IasZone:status_change_notification(zone_status=<ZoneStatus.Alarm_1: 1>, extended_status=<bitmap8: 0>, zone_id=0, delay=0)
> 2024-02-02 17:10:44.653 DEBUG (MainThread) [zigpy.zcl] [0x9C87:1:0x0500] Received command 0x00 (TSN 19): status_change_notification(zone_status=<ZoneStatus.Alarm_1: 1>, extended_status=<bitmap8: 0>, zone_id=0, delay=0)
> 2024-02-02 17:10:44.654 DEBUG (MainThread) [homeassistant.components.zha.core.cluster_handlers] [0x9C87:1:0x0500]: Updated alarm state: 1
> 2024-02-02 17:10:44.654 DEBUG (MainThread) [zigpy.zcl] [0x9C87:1:0x0500] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=19, command_id=<GeneralCommand.Default_Response: 11>, *direction=<Direction.Client_to_Server: 1>)
> 2024-02-02 17:10:44.655 DEBUG (MainThread) [zigpy.zcl] [0x9C87:1:0x0500] Sending reply: Default_Response(command_id=0, status=<Status.SUCCESS: 0>)
> 2024-02-02 17:10:44.655 DEBUG (MainThread) [zigpy_znp.zigbee.application] Sending packet ZigbeePacket(timestamp=datetime.datetime(2024, 2, 2, 16, 10, 44, 655279, tzinfo=datetime.timezone.utc), src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x9C87), dst_ep=1, source_route=None, extended_timeout=False, tsn=19, profile_id=260, cluster_id=1280, data=Serialized[b'\x18\x13\x0b\x00\x00'], tx_options=<TransmitOptions.ACK: 1>, radius=0, non_member_radius=0, lqi=None, rssi=None)
> 2024-02-02 17:10:44.655 DEBUG (MainThread) [zigpy_znp.api] Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0x9C87), DstEndpoint=1, DstPanId=0x0000, SrcEndpoint=1, ClusterId=1280, TSN=19, Options=<TransmitOptions.ACK_REQUEST|SUPPRESS_ROUTE_DISC_NETWORK: 48>, Radius=0, Data=b'\x18\x13\x0B\x00\x00')
> 2024-02-02 17:10:44.660 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataRequestExt.Rsp(Status=<Status.SUCCESS: 0>)
> 2024-02-02 17:10:44.779 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.SrcRtgInd.Callback(DstAddr=0xBD22, Relays=[0x0FD8])
> 2024-02-02 17:10:44.780 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=1280, SrcAddr=0xBD22, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=83, SecurityUse=<Bool.false: 0>, TimeStamp=8216383, TSN=0, Data=b'\x09\xC5\x00\x01\x00\x00\x00\x00\x00', MacSrcAddr=0x0FD8, MsgResultRadius=28)
> 2024-02-02 17:10:44.781 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(timestamp=datetime.datetime(2024, 2, 2, 16, 10, 44, 781290, tzinfo=datetime.timezone.utc), src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xBD22), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=0, profile_id=260, cluster_id=1280, data=Serialized[b'\t\xc5\x00\x01\x00\x00\x00\x00\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=28, non_member_radius=0, lqi=83 [... cut ...] the battery is low. I carefully bent the battery pins so that they'd push against the battery some more. After that, the device paired just fine. But maybe not good enough sometimes?
> 
> EDIT: I've now narrowed it down to an incompatibility with my Coordinator, CC2652 (Texas Instruments) running Firmware: Z-Stack 20230507. I've added a repeater since, and when the motion detectors are connecting to it, no such problems arise.

What repeater are you using? Have same problem with this pir sensor.

srett, 2024-04-04:
@Kerber0ss  A [Tuya wall socket](https://www.zigbee2mqtt.io/devices/TS011F_plug_1.html)

wsegatto, 2024-04-14:
I'm with HA Yellow with the embedded radio and I'm getting the same issue. I get "Motion" and "Opening" with get with a fixed status and then at some point go to "Unavailable" and RIP.
A reset gets the cycle started again.

[image]

Any workarounds? I saw some people discussing quirks but not for this exactly same model.
https://github.com/zigpy/zha-device-handlers/issues/1599#issuecomment-1482281897

Eiritj, 2024-04-30:
> A reset gets the cycle started again.

Same here.

closed 2024-11-03 by github-actions[bot]