zigpy/zha-device-handlers #2359: [Device Support Request] Aqara E1 thermostat schedule

Mikaka27, 2023-04-27:
**Is your feature request related to a problem? Please describe.**
Cannot set schedule settings on Aqara E1 thermostat.

**Describe the solution you'd like**
I would like schedule settings to be exposed in zigbee attributes, similar to what zigbee2mqtt does:
https://www.zigbee2mqtt.io/devices/SRTS-A01.html#schedule-settings-text
In reality any way to set the schedules would be good.

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0301",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x000a",
        "0x0201",
        "0xfcc0"
      ],
      "out_clusters": [
        "0x0003",
        "0x0019",
        "0x0201",
        "0xfcc0"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.airrtc.agl001",
  "class": "zhaquirks.xiaomi.aqara.thermostat_agl001.AGL001"
}
```

</details>

<details>

<summary>Diagnostic information</summary>
Cannot get diagnostic information, this is what I get when I click it:

```yaml
2023-04-27 17:42:30.941 ERROR (MainThread) [aiohttp.server] Error handling request
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/aiohttp/web_protocol.py", line 433, in _handle_request
    resp = await request_handler(request)
  File "/usr/local/lib/python3.10/site-packages/aiohttp/web_app.py", line 504, in _handle
    resp = await handler(request)
  File "/usr/local/lib/python3.10/site-packages/aiohttp/web_middlewares.py", line 117, in impl
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 85, in security_filter_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 100, in forwarded_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 28, in request_context_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 80, in ban_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 235, in auth_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/view.py", line 146, in handle
    result = await result
  File "/usr/src/homeassistant/homeassistant/components/diagnostics/__init__.py", line 265, in get
    data = await info.device_diagnostics(hass, config_entry, device)
  File "/usr/src/homeassistant/homeassistant/components/zha/diagnostics.py", line 104, in async_get_device_diagnostics
    device_info[CLUSTER_DETAILS] = get_endpoint_cluster_attr_data(zha_device)
  File "/usr/src/homeassistant/homeassistant/components/zha/diagnostics.py", line 126, in get_endpoint_cluster_attr_data
    ATTR_IN_CLUSTERS: {
  File "/usr/src/homeassistant/homeassistant/components/zha/diagnostics.py", line 129, in <dictcomp>
    **get_cluster_attr_data(cluster),
  File "/usr/src/homeassistant/homeassistant/components/zha/diagnostics.py", line 155, in get_cluster_attr_data
    UNSUPPORTED_ATTRIBUTES: {
  File "/usr/src/homeassistant/homeassistant/components/zha/diagnostics.py", line 156, in <dictcomp>
    f"0x{cluster.find_attribute(u_attr).id:04x}": {
  File "/usr/local/lib/python3.10/site-packages/zigpy/zcl/__init__.py", line 181, in find_attribute
    return self.attributes[name_or_id]
KeyError: 900
```

</details>

ShogunMan, 2023-04-27:
How about using this for even more flexibility:
https://github.com/nielsfaber/scheduler-card

Mikaka27, 2023-04-27:
As far as I know this would only work when Homeassistant is online?

ShogunMan, 2023-04-27:
Well, one of the main points with HA is keeping it online, at least on the LAN to be able to do these things.
Do you ever turn it off?

Mikaka27, 2023-04-27:
I still would like to find a solution for the built in scheduler.

TheJulianJES, 2023-04-27:
An initial attempt for adding the native scheduling option of the device was made here:
- https://github.com/zigpy/zha-device-handlers/pull/2109

At some point in the future, I might get to updating that PR, adding tests, and so on. The [author mentioned that the code can be taken as-is](https://github.com/zigpy/zha-device-handlers/pull/2109#issuecomment-1448713374), so if anyone wants to update it, add tests, and then PR again, feel free to do so.

The goal would be to have a fake attribute which can be written to using the ZHA service call.
In the even further future, it might be nice to have a ZHA configuration text entity to select the schedule too.

TheJulianJES, 2023-04-27:
Also, the issue with not being able to download diagnostic for your device is a ZHA/HA Core issue (which we shouldn't forget about 😄).

Mikaka27, 2023-04-28:
How did you implement some functionality for this TRV before? I mean how did you find what "register" in device did what? Did you look at zigbee2mqtt code or something else?

Mar1usW3, 2023-05-04:
> > is it possible to add an external temp. sensor to the TRV? There is a value in Diagnose but how to set this up?
> 
> There are several ways to do this. Create a virtual thermostat in config.yaml, or you could probably use this: https://community.home-assistant.io/t/target-temperature-adjustment-depending-on-second-temperature-sensor/297287

but this is only a possibilty to set this on a "software" level. I hope for a solution more on the hardware level so that the TRV is directly connected to the temp. sensor

Mar1usW3, 2023-05-05:
@ShogunMan yes

ShogunMan, 2023-05-05:
Well, then you link the sensor with the TRV in the Aqara/smart life app and it will report the temp read by that sensor.

Mar1usW3, 2023-05-05:
but this will only work with an Aqara Hub. In this PR https://github.com/zigpy/zha-device-handlers/pull/2109 there is also the attribute for a temp. sensor included. I ask myself if it will be then possible to link a temp. sensor to the TRV.

ShogunMan, 2023-05-05:
You will need to link the sensor using the hub & app. It is still a software solution, no matter how you look at it.
When the sensor has been linked, you can probably disconnect the hub. It should be possible to send/write the zigbee ID of the external sensor to the appropriate cluster of the TRV using a custom command, but this would need more documentation.

ShogunMan, 2023-05-05:
> Even when linking the sensor, the Aqara hub is still required I'm pretty sure. It's sending a special message to the E1 thermostat. ZHA could implement this as a "fake attribute".

I have a floor heating termostat that I needed the Tuya hub to configure, as all config options was not available in ZHA. These configurations stuck afterwards when removing the hub.

If the external sensor of the TRV is not stored on the device, but is applied through the hub/app then it is a pure software solution anyway and one might just use the blueprint for thermostat with external sensor. Same outcome, no difference.

Mikaka27, 2023-05-07:
> An initial attempt for adding the native scheduling option of the device was made here:
> 
>     * [Expand support for Aqara E1 Thermostat #2109](https://github.com/zigpy/zha-device-handlers/pull/2109)
> 
> 
> At some point in the future, I might get to updating that PR, adding tests, and so on. The [author mentioned that the code can be taken as-is](https://github.com/zigpy/zha-device-handlers/pull/2109#issuecomment-1448713374), so if anyone wants to update it, add tests, and then PR again, feel free to do so.
> 
> The goal would be to have a fake attribute which can be written to using the ZHA service call. In the even further future, it might be nice to have a ZHA configuration text entity to select the schedule too.

Are you sure it's that pull request? Since there seems to be nothing about schedule there (only external temp sensor).
Unless of course you meant that some preparatory work was there only.

closed 2023-11-13 by TheJulianJES