zigpy/zha-device-handlers #2365: [Device Support Request]  Aubess/Tuya TS0505B by _TZ3210_it1u8ahz RGBCC bulb.

ShogunMan, 2023-04-28:
### Problem description

Device is throwing log errors on the color cluster 0X300 (768) (see log)

### Solution description

Device seems to be working, but would be nice to get rid of the errors cluttering the log.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x010d",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x000a",
        "0x0300",
        "0x1000",
        "0x1888",
        "0xffee"
      ],
      "out_clusters": [
        "0x0019",
        "0x1000",
        "0xffee"
      ]
    }
  },
  "manufacturer": "_TZ3210_it1u8ahz",
  "model": "TS0505B",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant Container",
    "version": "2023.4.6",
    "dev": false,
    "hassio": false,
    "virtualenv": false,
    "python_version": "3.10.10",
    "docker": true,
    "arch": "aarch64",
[... 632 lines cut ...]

<details><summary>Logs</summary>

```python
Logger: zigpy.device
Source: runner.py:179
First occurred: 15:56:00 (20 occurrences)
Last logged: 16:50:17

Failed to parse message (b'08fe0a0cf0480c000100000000000000000000') on cluster 768, because Data is too short to contain 1 bytes
Failed to parse message (b'08ff0a09f04800') on cluster 768, because Data is too short to contain 1 bytes
Failed to parse message (b'08b00a09f04800') on cluster 768, because Data is too short to contain 1 bytes
Failed to parse message (b'08b40a0bf04800') on cluster 768, because Data is too short to contain 1 bytes
Failed to parse message (b'08030a0bf04800') on cluster 768, because Data is too short to contain 1 bytes
```

</details>

### Custom quirk

No custom quirks or core quirks are loaded.

### Additional information

Have tried re-pairing the devices with no effect.

ShogunMan, 2023-05-01:
In the diagnostic section i find this:
`          "0x1888": {
            "endpoint_attribute": null,
            "attributes": {},
            "unsupported_attributes": {}
          },`

0x1888 has no matching entries in ZCL lightning.py hence endpoint_attribute is null.
Could this be the cause of the errors filling up the log?

ShogunMan, 2023-05-02:
@TheJulianJES @MattWestb @javicalle Any idea of where I should start looking to fix this?

MattWestb, 2023-05-02:
Sorry im not one code worrier so i cant helping but our friends 110% can do that :-))

ShogunMan, 2023-06-24:
@TheJulianJES & @javicalle Bumping this :-)

TheJulianJES, 2023-09-17:
See: https://github.com/home-assistant/core/issues/98900#issuecomment-1722261213

ShogunMan, 2023-09-17:
@TheJulianJES 
This pull adds a class for the Tuya 0x1888 cluster:
[https://github.com/zigpy/zha-device-handlers/pull/2592](https://github.com/zigpy/zha-device-handlers/pull/2592)

It might fix the log errors. What is the easiest way to test this?

MattWestb, 2023-09-17:
Making one local quirk for your device and adding the signature and importing and use the 0x1888 cluster in it then the system shall trying using the commands from the 0x1888 cluster / attributes of it but i think its a not working one on this device.

closed 2024-03-22 by github-actions[bot]