zigpy/zha-device-handlers #2366: [BUG] Issues with _TZE200_a7sghmms

mkoppanen, 2023-04-28:
### Bug description

The device is a GiEX irrigation valve QT06_2 variant (as seen here: https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/giex.js#LL202C17-L202C23)

When adding the device in ZHA only the on/off switch entity is added and the rest of the values (battery level etc) are not properly displayed. After pairing the device the on/off switch works for about few minutes and then stops controlling the valve. My controller is Sonoff Zigbee 3.0 USB Dongle Plus (e-variant).

In order to exclude device faults I installed a standalone zigbee2mqtt using a separate usb dongle and the device stays connected for several hours (that's as long as I tested).

I've enabled debug logging and tried to isolate relevant parts in the message below. However, if there are any other logs I can produce let me know.

### Steps to reproduce

1. Pair device with ZHA
2. Make sure that on off switch controls the valve
3. Wait a few minutes 
4. The on/off switch no longer controls the valve

### Expected behavior

The valve to work normally and the entities to be added

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0702",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_a7sghmms",
  "model": "TS0601",
  "class": "zhaquirks.tuya.ts0601_valve.GiexValve"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
[... 607 lines cut ...]
[0xdfd0:zdo] ZDO request ZDOCmd.NWK_addr_req: [e0:79:8d:ff:fe:88:54:ed, <AddrRequestType.Single: 0>, 0]
[0x5718](RS 225): Device seen - marking the device available and resetting counter
[0x5718](RS 225): Device seen - marking the device available and resetting counter
[0x5718](RS 225): Update device availability -  device available: True - new availability: True - changed: False
[0x5718](RS 225): Update device availability -  device available: True - new availability: True - changed: False
Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xDFD0), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=0, source_route=None, extended_timeout=False, tsn=11, profile_id=0, cluster_id=2, data=Serialized[b'\x03\x00\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=132, rssi=-67)
Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xDFD0), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=0, source_route=None, extended_timeout=False, tsn=11, profile_id=0, cluster_id=2, data=Serialized[b'\x03\x00\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=132, rssi=-67)
[0xdfd0:zdo] ZDO request ZDOCmd.Node_Desc_req: [0x0000]
[0xdfd0:zdo] ZDO request ZDOCmd.Node_Desc_req: [0x0000]
[0xdfd0:zdo] No handler for ZDO request:ZDOCmd.Node_Desc_req([0x0000])
[0xdfd0:zdo] No handler for ZDO request:ZDOCmd.Node_Desc_req([0x0000])
Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xEAE4), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=219, profile_id=260, cluster_id=1280, data=Serialized[b'\tV\x00\x01\x00\x00\x00\x00\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=132, rssi=-67)
Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xEAE4), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=219, profile_id=260, cluster_id=1280, data=Serialized[b'\tV\x00\x01\x00\x00\x00\x00\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=132, rssi=-67)
[0xEAE4:1:0x0500] Received ZCL frame: b'\tV\x00\x01\x00\x00\x00\x00\x00'
[0xEAE4:1:0x0500] Received ZCL frame: b'\tV\x00\x01\x00\x00\x00\x00\x00'
[0xEAE4:1:0x0500] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), tsn=86, command_id=0, *direction=<Direction.Client_to_Server: 1>)
[0xEAE4:1:0x0500] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), tsn=86, command_id=0, *direction=<Direction.Client_to_Server: 1>)
[0xEAE4:1:0x0500] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=86, command_id=<GeneralCommand.Default_Response: 11>, *direction=<Direction.Client_to_Server: 1>)
[0xEAE4:1:0x0500] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=86, command_id=<GeneralCommand.Default_Response: 11>, *direction=<Direction.Client_to_Server: 1>)
[0xEAE4:1:0x0500] Sending reply: Default_Response(command_id=0, status=<Status.SUCCESS: 0>)
[0xEAE4:1:0x0500] Sending reply: Default_Response(command_id=0, status=<Status.SUCCESS: 0>)

```

</details>

mkoppanen, 2023-05-05:
Some additional information: I've been running a lot of tests with this device and zha. The behaviour seems to be consistent: When I pair the device it keeps working for a couple of minutes and then the HA switch stops controlling the valve. 

After 6 hours of pairing the device becomes unavailable (6 hours is my configured time for battery powered devices to be seen offline).

I don't know whether it's relevant but during the time the device is connected it keeps changing id (Log entries about device changed id). 

I've tried to see if there are any debug logging events around the time when the valve becomes unresponsive but I cannot see anything relevant. 

Any tips for debugging this issue further?

closed 2023-11-08 by github-actions[bot]

MartinNeurol, 2024-07-22:
Hi, Martin here. I know this is a while ago, but... I own three of those valves and am facing the same issue, they are unusable with ZHA. Have you been able to solve this last year, or gave up? Thank you, Martin

mkoppanen, 2024-07-22:
@MartinNeurol I switched to zigbee2mqtt and haven't had similar issues since.

MartinNeurol, 2024-07-22:
oh, I see... thank you...