zigpy/zha-device-handlers #2367: [Device Support Request]/Quirks Bug/New Firmware

Drizzt321, 2023-04-28:
### Problem description

I recently got some new Aqara Opple remote switches purchased from Aliexpress (where I purchased my original ones), WXCJKG11LM (2 button, lumi.remote.b286opcn01) and WXCJKG12LM (4 button, lumi.remote.b486opcn01). I have multiple WXCJKG12LM that I've been using for a year or more now, just great, I have `single`, `double`, `hold`, `release` events.

However, the new ones, both WXCJKG11LM and WXCJKG12LM, don't give me those.

WXCJKG11LM gets pulled in with Quirks `zhaquirks.xiaomi.aqara.opple_remote.RemoteB286OPCN01V2`, and WXCJKG12LM with Quirks `zhaquirks.xiaomi.aqara.opple_remote.RemoteB486OPCN01V2`. The original WXCJKG12LM remotes are identified with the same quirks, which leads me to believe some kind of firmware or other similar change inside.

The ZHA events I get from WXCJKG11LM are _only_ when I double-press the buttons. No single, no hold/release, nothing. The only difference is the 1st `args` is 0 or 1, and `step_mode` is 0 or 1. 
```
event_type: zha_event
data:
  device_ieee: <SNIP>
  unique_id: <SNIP>:1:0x0008
  device_id: <SNIP>
  endpoint_id: 1
  cluster_id: 8
  command: step
  args:
    - 1
    - 85
    - 7
    - 0
    - 0
  params:
    step_mode: 1
    step_size: 85
    transition_time: 7
    options_mask: 0
    options_override: 0
origin: LOCAL
time_fired: "2023-04-28T22:22:52.070701+00:00"
context:
  id: 01GZ4ZCGN6Z00V5KJT6TSKGSGW
  parent_id: null
  user_id: null
```

The WXCJKG12LM ZHA events from the new ones only occur from apparently buttons 0 & 1, not from 2 & 3. Which are the buttons not on the switch with the LED light that shows through. However, I get the exact same data when I single press, _or_ hold, but not on release. The only difference between the 2 buttons is first `args` parameter, and the `step_mode` are either 0 or 1, just like the 2-button remote.

```
event_type: zha_event
data:
  device_ieee: <SNIP>
  unique_id: <SNIP>:1:0x0008
  device_id: <SNIP>
  endpoint_id: 1
  cluster_id: 8
  command: step
  args:
    - 1
    - 85
    - 7
    - 0
    - 0
  params:
    step_mode: 1
    step_size: 85
    transition_time: 7
    options_mask: 0
    options_override: 0
origin: LOCAL
time_fired: "2023-04-28T22:25:12.097045+00:00"
context:
  id: 01GZ4ZGSD124443X3B3RE6K7J6
  parent_id: null
  user_id: null
```

Anything further you need me to lookup/attach from HomeAssistant/ZHA? Anything I can try with the quirks? 
[... 1082 lines cut ...]
            "unsupported_attributes": {}
          }
        },
        "out_clusters": {}
      }
    }
  }
}
```
</details>

### Logs

<details><summary>Logs</summary>
I did 2 button remote, single click on each button, then double click on each button, then hold and release on each button. Then switched to the 4 button starting with the 0 & 1 buttons that I got ZHA events for, doing single, double, hold & release, and then went to the other 2 buttons and did single, double, hold & release. 

See log file [zha debug.log](https://github.com/zigpy/zha-device-handlers/files/11357683/zha.debug.log)

</details>

### Custom quirk

<details><summary>Custom quirk</summary>
No custom quirk
</details>

Drizzt321, 2023-05-02:
@tube0013 here's the Aqara Opple issue I was talking about elsewhere.

TheJulianJES, 2023-05-02:
You should be able to change the switch_mode on the OppleCluster through the clusters UI.

Drizzt321, 2023-05-02:
@TheJulianJES So I've looked at that now, OppleCluster, Endpoint ID: 1, Id: 0xfcc0, only see 1 Attribute `mode` Id: 0x0009. Don't see any `switch_mode` attribute. 

Tried to do a Read Attribute on `mode`, got no value returned. I've done some searches, there's nothing out there that I see, or in https://github.com/zigpy/zha-device-handlers/blob/fb5d256299dd789536034ca33a91afc646a07251/zhaquirks/xiaomi/aqara/opple_remote.py, to say what values are allowed to set. Maybe I'm not looking in the right places, or the right search terms.

TheJulianJES, 2023-05-02:
It's probably `mode` then. You'll need to wake the remote by pressing a button on it just after you've pressed read/write.

TheJulianJES, 2023-05-02:
Try to write it opposite of what it currently is set to. Then see if the events are different.

Drizzt321, 2023-05-02:
So the existing, working ones have that a `1`, but I can't seem to get it to read from the new one. I click a button and click Read Attribute, I see the event come in via Logbook, but it gives me an error. The write does say it succeeds, but I can't tell if it did for sure. 

I tried removing the device, and re-pairing. Had to reload ZHA, and then do re-pair again, and huzzah! It works!

Must just have been set in mode = 1 the other ones I received. PITA. But thanks so much!

closed 2023-05-02 by Drizzt321