zigpy/zha-device-handlers #2368: [Device Support Request] TuYa TS011F Smart Plug by _TZ3000_8a833yls

patrickli, 2023-04-30:
### Problem description

Recently got a TuYa Smart Plug by _TZ3000_8a833yls. This has energy metering functions. ZHA can detect all of its controls and sensors but the sensors return no data, eg, 0W, 0.0A and 0V with a load.

### Solution description

Make the metering sensors work so I can monitor energy consumption.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x010a",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0702",
        "0x0b04",
        "0xe000",
        "0xe001"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3000_8a833yls",
  "model": "TS011F",
  "class": "zhaquirks.tuya.ts011f_plug.Plug"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Unsupported Third Party Container",
    "version": "2023.4.6",
    "dev": false,
    "hassio": false,
    "virtualenv": false,
    "python_version": "3.10.10",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Pacific/Auckland",
    "os_name": "Linux",
    "os_version": "6.2.9-1-pve",
[... 496 lines cut ...]
                "value": 0
              }
            },
            "unsupported_attributes": {}
          }
        },
        "out_clusters": {
          "0x000a": {
            "endpoint_attribute": "time",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

TheJulianJES, 2023-05-01:
Can you try to reconfigure the plug through the UI via the device settings page?

patrickli, 2023-05-01:
Did that. All green ticks, but still no values.

[image]

patrickli, 2023-05-01:
Alright this device actually doesn't come with energy metering. I just double checked that I ordered the wrong type. https://github.com/Koenkk/zigbee2mqtt/issues/11505 also mentioned this doesn't come with measuring. Maybe we should add a quirk to disable metering sensors.

MattWestb, 2023-05-01:
@TheJulianJES Its possible making one device class without power measuring that is matched on the model ID but it must being loaded before the general class as we have done with the TS004F remotes.

closed 2023-11-04 by github-actions[bot]