zigpy/zha-device-handlers #2369: [BUG] TS0012, TS0013 3gang switch cannot switch single gang, all gangs always turned on/off together

jkraemer, 2023-05-01:
### Bug description

I have successfully paired two Tuya compatible switches (the brand is Zemismart) that show the problem that whenever I toggle one gang in HA, all gangs get toggled. Switching single gangs manually on the switches works fine. Quirks are loaded but do not seem to fix it for these models.

### Steps to reproduce

Toggle any one gang of the switch in HA dashboard and all gangs will be toggled.

### Expected behavior

I'd expect to be able to toggle each gang separately.

### Device signature

<details><summary>Device signature</summary>

**2-gang**

```json
Nwk: 0x8b27
Device Type: EndDevice
LQI: 54
RSSI: Unknown
Last Seen: 2023-05-01T10:53:52
Power Source: Battery or Unknown
Quirk: zhaquirks.tuya.ts001x.TuyaDoubleNoNeutralSwitch_2

```

**3-gang**

```json
Nwk: 0x4d27
Device Type: EndDevice
LQI: 51
RSSI: Unknown
Last Seen: 2023-05-01T10:58:18
Power Source: Battery or Unknown
Quirk: zhaquirks.tuya.ts001x.TuyaTripleNoNeutralSwitch_2
```
</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

**2-gang**
```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.4.6",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.10",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Asia/Manila",
    "os_name": "Linux",
    "os_version": "6.1.21-v8",
    "supervisor": "2023.04.1",
    "host_os": "Home Assistant OS 10.1",
    "docker_version": "23.0.3",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {},
  "integration_manifest": {
    "domain": "zha",
[... 727 lines cut ...]
            "endpoint_attribute": "on_off",
            "attributes": {
              "0x0000": {
                "attribute_name": "on_off",
                "value": 0
              },
              "0x4001": {
                "attribute_name": "on_time",
                "value": 0
              },
              "0x4002": {
                "attribute_name": "off_wait_time",
                "value": 0
              }
            },
            "unsupported_attributes": {}
          }
        },
        "out_clusters": {}
      }
    }
  }
}
```
</details>

MattWestb, 2023-05-01:
This problem shall being fixed from HA 2023.1 then implanted the tuya magic for this devices.
Ar you running one clean HA or have you installed custom quirks for the devices ?

jkraemer, 2023-05-01:
Hi, this is a fresh install of Home Assistant (on a RaspberryPI). I did not install any custom quirks, the quirk that apparently is loaded for the devices came stock with the HA image.

MattWestb, 2023-05-01:
Then it shall working.
The devcie need one "tuya magic" being casted then its joining and then it shall doing all things OK but its have being some refactoring of the code in the last release with the logic.
I dont have the device for testing if its working OK if was having it i have sniffing the joining so can see if its casting is working OK that we was doing with some device then testing the last changes.

@TheJulianJES do you have some thinking about this problem ?
The device is loading the quirk OK and the device class is using the `EnchantedDevice, TuyaSwitch` so it shall working OK if not new is being broken after the last refactoring.
Also the devices is being new joined on one clean system so the casting shall working OK.

jkraemer, 2023-05-04:
Please let me know if there is anything I can do to help debug this.

thewolf-oz, 2023-05-06:
I have the same issue, happy to assist with anything too

hoshyK2, 2023-05-13:
For what it's worth I got this working as follows:

1. Install "ZHA Toolkit" - I did this via HACS
2. Go to "Developer Tools -> Services" in HA
3. Select Service "ZHA Toolkit: execute"
4. Select Command "tuya_magic"
5. In the ieee section select first Tuya entity and click call service - do for each Tuya entity

I had to re-run the tuya_magic command after changing the entity names (and / or repairing, not certain).
With a little fiddleing this should work.

NicolasIonut, 2023-05-13:
Thank you for your help! It's working fine now!

lanturn07, 2023-05-19:
Thank you!!!
I'm using TS0012 2-gang (model: ds-1021zc)
_TZ3000_zmlunnhy
It helped me too!

Danixu, 2023-06-25:
> For what it's worth I got this working as follows:
> 
>     1. Install "ZHA Toolkit" - I did this via HACS
> 
>     2. Go to "Developer Tools -> Services" in HA
> 
>     3. Select Service "ZHA Toolkit: execute"
> 
>     4. Select Command "tuya_magic"
> 
>     5. In the ieee section select first Tuya entity and click call service - do for each Tuya entity
> 
> 
> I had to re-run the tuya_magic command after changing the entity names (and / or repairing, not certain). With a little fiddleing this should work.
> 
> Edit: I'm using TS0013 3-gang btw

Two days fighting against this problem... Your comment has saved my day. Thank you very much!.

MattWestb, 2023-06-25:
Hola @javicalle Looks like new devices if this type is using the tuya magic.
Shall we rolling all classes to using it ?
Im most after the last changes with class that must being used so i cant do it and i dont have the device.

Thanks in advance !!!

NW

javicalle, 2023-06-28:
Hi Matt, sorry to be absent, but no spare time on my side.

> Shall we rolling all classes to using it ?

But the classes for these devices are using the `EnchantedDevice` already:

https://github.com/zigpy/zha-device-handlers/blob/9e210ba946069eeb9e2ae4d41dba362f0a399925/zhaquirks/tuya/ts001x.py#L285

https://github.com/zigpy/zha-device-handlers/blob/9e210ba946069eeb9e2ae4d41dba362f0a399925/zhaquirks/tuya/ts001x.py#L359

And both are using the `TuyaZBOnOffAttributeCluster` which is an `TuyaEnchantableCluster`:
https://github.com/zigpy/zha-device-handlers/blob/9e210ba946069eeb9e2ae4d41dba362f0a399925/zhaquirks/tuya/__init__.py#L989C35-L996

So the Tuya spell must be casted for these devices.
I can't detect any problem in hierarchy or classes.

Maybe these devices don't trigger the bind for the OnOff cluster?

MattWestb, 2023-06-29:
Hi JC thanks for replay !!
Yes i was looking on the wrong quirk and some one was changing it for over a half year ago https://github.com/zigpy/zha-device-handlers/pull/2049.

Back to the problem users is having:
The quirk shall sending the tuya magic one time then the device is being added new to the system and then remember it until its being retested.
If its loosing it you must deleting in in ZHA and waiting one minute so the system is deleting it in the DB (or restarting HA) and then adding it as new device and it shall working OK.
Using ZHA-Toolkit shall also working OK then the device can being casted all the time and not only in init mode like some controllers is needing.

TheJulianJES, 2023-06-29:
> If its loosing it you must deleting in in ZHA and waiting one minute so the system is deleting it in the DB (or restarting HA) and then adding it as new device and it shall working OK.

This shouldn’t be the case anymore. You can just reconfigure the device to cast the spell or just re-pair it (WITHOUT having to delete it from HA).

MattWestb, 2023-06-29:
> This shouldn’t be the case anymore. You can just reconfigure the device to cast the spell or just re-pair it (WITHOUT having to delete it from HA).

I was testing reconfigure my TS004F DMS and sniffing it and indeed ZHA is casting the spell so it shall working for all devices that is not need getting it in INIT mode then they need being repaired for getting it in time for changing the hardware config of the device.

Thanks @TheJulianJES for correcting my !

setterlee, 2023-07-23:
> For what it's worth I got this working as follows:
>
> 1. Install "ZHA Toolkit" - I did this via HACS
> 2. Go to "Developer Tools -> Services" in HA
> 3. Select Service "ZHA Toolkit: execute"
> 4. Select [...]

Every day we learn something new... And in this case I learned something new and fixed a problem. Thanks for share your knowledge..

CobraDunn, 2023-08-05:
> For what it's worth I got this working as follows:
>
> 1. Install "ZHA Toolkit" - I did this via HACS
> 2. Go to "Developer Tools -> Services" in HA
> 3. Select Service "ZHA Toolkit: execute"
> 4. Select [...]

Searched for solutions for an hour before finding this one. Thank you.  Confirmed working for me on 2-Gang - Moes TS0012 - _TZ3000_18ejxno0

smokerxf, 2023-08-23:
I have the same problem with ts0012. My 2-gang switch cannot switch single gang by HA.
I've tried using custom quirks from this repo = the situation is unchanged.
                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                      
` "data": {
    "ieee": "**REDACTED**",
    "nwk": 54326,
    "manufacturer": "_TZ3000_e98krvvk",
    "model": "TS0012",
    "name": "_TZ3000_e98krvvk TS0012",
    "quirk_applied": true,
    "quirk_class": "ts001x.TuyaDoubleNoNeutralSwitch_2",
    "manufacturer_code": 4417,
    "power_source": "Battery or Unknown",
    "lqi": 216,
    "rssi": -46,
    "last_seen": "2023-08-23T23:01:36",
    "available": true,
    "device_type": "EndDevice",
    "signature": {
      "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
      "endpoints": {
        "1": {
          "profile_id": "0x0104",
          "device_type": "0x0100",
          "input_clusters": [
            "0x0000",
            "0x0003",
            "0x0004",
            "0x0005",
            "0x0006"
          ],
          "output_clusters": [
            "0x0019"
          ]
        },
        "2": {
          "profile_id": "0x0104",
          "device_type": "0x0100",
          "input_clusters": [
            "0x0006"
          ],
          "output_clusters": []
        }
      },
      "manufacturer": "_TZ3000_e98krvvk",
      "model": "TS0012"
    },`

fabiosaboia, 2023-08-27:
> For what it's worth I got this working as follows:
>
> 1. Install "ZHA Toolkit" - I did this via HACS
> 2. Go to "Developer Tools -> Services" in HA
> 3. Select Service "ZHA Toolkit: execute"
> 4. Select [...]

thanks. woks for me.

Informação Dispositivo
TS0013
por _TZ3000_snq47izk

Zigbee info
IEEE: 
Nwk: 0x83f1
Device Type: EndDevice
LQI: 94
RSSI: Desconhecido
Visto pela Última Vez: 2023-08-27T18:46:27
Fonte de Energia: Battery or Unknown

etmmafl, 2023-10-07:
> For what it's worth I got this working as follows:
>
>     1. Install "ZHA Toolkit" - I did this via HACS
>
>     2. Go to "Developer Tools -> Services" in HA
>
>     3. Select Service "ZHA Toolkit: [...]

Searched for solutions for multiple days before finding this one. T
Thank you. 

Confirmed working for me on 2-Gang - Moes TS0012 - _TZ3000_18ejxno0

TheJulianJES, 2023-10-08:
@etmmafl This should no longer be required when re-pairing/reconfiguring with recent Home Assistant Core versions.
Please attach the "device diagnostic information" and/or device signature from the device you had issues with here.
It can be downloaded on the device page.

corcaioch, 2023-12-20:
> For what it's worth I got this working as follows:
>
> 1. Install "ZHA Toolkit" - I did this via HACS
> 2. Go to "Developer Tools -> Services" in HA
> 3. Select Service "ZHA Toolkit: execute"
> 4. Select [...]

This worked for me, thanks so much.

saadiego, 2024-03-13:
Worked flawlessly with my Zemismart 4 Gang Light Switch + 2 Gang Scene Switch - KES-606US-Z4+2 (TS0726 by _TZ3000_s678wazd)

closed 2024-09-16 by github-actions[bot]

dev-huiya, 2024-12-08:
> For what it's worth I got this working as follows:
>
> 1. Install "ZHA Toolkit" - I did this via HACS
> 2. Go to "Developer Tools -> Services" in HA
> 3. Select Service "ZHA Toolkit: execute"
> 4. Select [...]

This solution worked perfectly for me, even in 2024. 
My switches were _TZ3000_lrgccsxm TS0013, TS0012 and with this solution they now work perfectly. 

God bless you
Thankyou so much