zigpy/zha-device-handlers #2375: [Device Support Request] Tuya Neo Outdoor Siren with Solar (TZE200_nlrfgpny)

JMoratelli, 2023-05-03:
### Problem description

Hello good afternoon! I recently bought this solar siren from aliexpress, very well built product, however I couldn't get it to work within the ZHA, I checked and found a support request for zigbee2mqtt, but I didn't find anything about adding it to the ZHA, not even a device handler that was compatible with her, I attached her diagnosis.

this was the thread I found about this equipment (but zigbee2mqtt) https://github.com/Koenkk/zigbee2mqtt/discussions/17325

This is the product i bought: https://pt.aliexpress.com/item/1005005264809268.html
And this is the product on the manufacturer's website https://www.szneo.com/en/products/show.php?id=251 (but mine is zigbee version).

### Solution description

The device has a battery, is powered by solar energy, however if I could at least trigger the siren, that would be great for me, would anyone be able to help me?

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_nlrfgpny",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.4.6",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.10",
    "docker": true,
    "arch": "x86_64",
    "timezone": "America/Cuiaba",
    "os_name": "Linux",
    "os_version": "6.1.25",
[... 342 lines cut ...]
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

Cityboy3, 2023-05-21:
There is a solution using zigbee2mqtt:[https://github.com/junbing2006/zigbee_siren_TZE200_nlrfgpny/blob/main/zigbee_siren_TZE200_nlrfgpny.js]
I have a Tuya Gateway so I am still looking for a solution to get the entities visible by using local tuya in HACS.

JMoratelli, 2023-05-22:
> There is a solution using zigbee2mqtt:[https://github.com/junbing2006/zigbee_siren_TZE200_nlrfgpny/blob/main/zigbee_siren_TZE200_nlrfgpny.js] I have a Tuya Gateway so I am still looking for a solution to get the entities visible by using local tuya in HACS.

Yes, I had seen this solution, but I only use the ZHA, as it is more integrated with the home assistant. Unfortunately at ZHA there is no discussion about it, probably because it is a newly launched product, I had difficulty finding any information about it even on google. Anyway, thank you for your attention.

JMoratelli, 2023-05-22:
I clicked the wrong button...

coasttech, 2023-06-14:
Any on how to address in Home Assistant using zigbee2mqtt?

JMoratelli, 2023-06-15:
> Anyone how to add this in Home Assistant using zigbee2mqtt?

This is the external converter file to support zigbee siren _TZE200_nlrfgpny on zigbee2mqtt, there's just nothing in the zha.
https://github.com/junbing2006/zigbee_siren_TZE200_nlrfgpny/tree/main

JMoratelli, 2023-07-08:
I bought a zigbee2mqtt compatible adapter and migrated to it. It is working without any convert on it.

wappinghigh, 2023-07-23:
I'd also like a ZHA version if it could be done.
Thanks

nric, 2023-08-14:
Is this not planned or completed? Says it is both sequentially. Would be great if this could get a ZHA version.

JMoratelli, 2023-08-14:
not planned, I could only make it work by z2m only

ioangogo, 2023-08-29:
Picking this up by making a quirk, I have ordered one and i am waiting for it to arrive, but If someone with one already could get the Simple Descriptor as [mentioned here ](https://github.com/zigpy/zha-device-handlers#simple-descriptor) that would help a lot in getting the quirk written

ioangogo, 2023-08-31:
also could this issue get reopened, it was prematurely closed because the reporter was not getting attention, its still a issue as the device and should have been left open even if you had found another solution in diffrent software

JMoratelli, 2023-08-31:
ok, i'll leave it open. hope it helps you.

nric, 2023-09-01:
> but If someone with one already could get the Simple Descriptor as [mentioned here ](https://github.com/zigpy/zha-device-handlers#simple-descriptor) that would help a lot in getting the quirk written.

I'm new to this but is this what you need?
[zha-db386a20c9c68f5ceaa30276e1ec743c-_TZE200_nlrfgpny TS0601-34a5825e72845a4d109167aad3342f11.json.txt](https://github.com/zigpy/zha-device-handlers/files/12501708/zha-db386a20c9c68f5ceaa30276e1ec743c-_TZE200_nlrfgpny.TS0601-34a5825e72845a4d109167aad3342f11.json.txt)

ioangogo, 2023-09-10:
nah, it's something that comes up in the logs while pairing, mines arrived now so it can get it there

still not fully sure on how I can use the zigbee2mqtt converter to get the information needed to write the quirk.

I think I'll dedicate sometime to it next weekend to try and get it semi working

Cityboy3, 2023-09-10:
Great you keep working on it and keep us updated!

ioangogo, 2023-09-11:
@FrnchFrgg sorry for pinging you out of the blue 2 years after your contributions, where did [these values](https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_siren.py#L39-L51) come from?

Puppetmaster91, 2023-10-30:
@ioangogo i used your temporary quirk and it gets some functionality out of the product. i am starting to look into how to read attributes on zigbee devices because i want this device to work. Outdoor sirens are apperently rare in the smart home scene. 
I hope once i get some more info i can help alittle with it.

Are you still working on a quirk for this?

ioangogo, 2023-11-11:
yeah, although I currently don't have much time to fully investigate the other attributes

Puppetmaster91, 2023-11-11:
> yeah, although I currently don't have much time to fully investigate the other attributes

Sounds awesome, i wrote an e-mail to szneo in order to get the technical documentation describing which attributes and clusters they use for the product, hopefully i will get a response. I have tried looking through the configuration of the device in Z2M in order to find the right cluster attributes, but without much luck. 

They got it to somewhat work in Z2M so in someway there must be some code where we can find the right attributes so we can make it work in ZHA.

liouma, 2023-11-17:
I also would be very interested by a custom quirck for this siren :)

dominikamann, 2023-12-07:
Thank you for opening the thread I would also be very interested in an integration. I bought the sierene a while ago and unfortunately can't get it integrated. Triggering the siren and stopping it would be enough for me. Is there perhaps already a way via HACS? Thank you!

[image]

nric, 2023-12-07:
@dominikamann I looked for it last time a couple of weeks ago and couldn't find any :( 

It would be wonderful if someone would make it work, I just really don't want to throw it away. Happy to donate.

harishvelli, 2023-12-12:
Quirk would be great interim solution if possible!

liouma, 2023-12-12:
> Quirk would be great interim solution if possible!

I've been waiting too long to keep hope !
I gave up and installed zigbee2mqtt on a pi zero 2w with a conbee2. One more device to maintain but it work

JMoratelli, 2023-12-13:
> > Quirk would be great interim solution if possible!
> 
> I've been waiting too long to keep hope ! I gave up and installed zigbee2mqtt on a pi zero 2w with a conbee2. One more device to maintain but it work

I did exactly the same thing and I didn't regret it. ZHA is great, works well. But Z2M has much more community support overall

masi, 2024-01-13:
> But Z2M has much more community support overall
Sadly all questions on how to write a quirk are either ignored or answered with "read the README". I've read the thing but it's so short that I really wonder how the devs of ZHA expect anyone outside their group to understand it.

Puppetmaster91, 2024-01-18:
So someone has made a quirk for Z2M in order for it to work there. 

What ioangogo is needing is to know which clusters that was used where he defined 0x0000 ini the quirk. i have tried looking through the config in Z2M in order to find the right clusters, unfortunately the TUYA config for Z2M is hugh and cotains a huge amount of lines of code.

I think the best bet to get the info of the clusters needed is to ask the code owner of the Z2M TUYA config.

nils-82, 2024-01-26:
Interested in thisas well, That would be great to make this work natively with zha.

constapel, 2024-02-10:
Please continue working on the zha integration of NAS-AB06B2 _TZE204_nlrfgpny TS0601
https://nl.aliexpress.com/item/1005005827808088.html

masi, 2024-02-17:
Got mine yesterday. Hope to figure out at least so much I can send "raw Zigbee data" to make the sirene sing.

voyagerft, 2024-03-22:
[image]
I succeeded, I created a quirks for ZHA at least now the TS0601 _TZE204_nlrfgpny siren has the ability to activate and deactivate it from the Home assistant. "!Long life and prosperity!"

liouma, 2024-03-22:
> I succeeded, I created a quirks for ZHA at least now the TS0601 _TZE204_nlrfgpny siren has the ability to activate and deactivate it from the Home assistant. "!Long life and prosperity!"

Congrats! that's a great news!!!
I'm looking forward testing it !

lboue, 2024-03-22:
> I succeeded, I created a quirks for ZHA at least now the TS0601 _TZE204_nlrfgpny siren has the ability to activate and deactivate it from the Home assistant. "!Long life and prosperity!"

@voyagerft Can you tell us how to do it?

voyagerft, 2024-03-22:
Sharing is life..
sorry for my English I studied French at school.

Yesterday I received this siren but I realized that it didn't expose any entity, then I saw that many of you had been fighting it for about 1 year, so this morning after coffee (I'm more active in the morning after coffee) I wrote my first quirks, in 20 minutes everything resolved, at least for the activation/deactivation of the alarm, in the future I would like to try to expose the volume and melody clusters.

In the meantime, I'll give you the procedure for installing ZHA in home assistant

- create a "custom_zha_quirks" folder in config/
- add the following code to configuration.yaml

zha:
   enable_quirks: true
   custom_quirks_path: config/custom_zha_quirks/

- upload the attached files to the config/custom_zha_quirks/ folder
__init__.py and ts601_siren.py

- Delete the siren from the Zigbee devices, if previously configured and restart Home Assistant

- Perform the Zigbee pairing again with the siren and the switch will come out

I remind everyone that it works exclusively with TS0601 _TZE204_nlrfgpny

If you help me I could insert volume level and melody
[Quirks_Ts601_by_Voyagerft_ver_1_0.zip](https://github.com/zigpy/zha-device-handlers/files/14720716/Quirks_Ts601_by_Voyagerft_ver_1_0.zip)

dominikamann, 2024-03-22:
@voyagerft awesome. Made my day! Thank you. I will test as soon as possible 💪👍

lboue, 2024-03-22:
@voyagerft 
Thanks a lot. It works for me: 

[image]

[image]

dominikamann, 2024-03-22:
Working fine for me as well. Just damaged my hearing 📢😃

voyagerft, 2024-03-22:
Good.. code on github
https://github.com/voyagerft/ZHA-TS0601-_TZE204_nlrfgpny

liouma, 2024-03-22:
@voyagerft : how can I help you?
I have 2 models of this siren. Zigbee and also wifi.
Do you think I can get interesting info for you from tuya wifi integration?
Or from tuya dev platform?

voyagerft, 2024-03-22:
From ZHA's device management it is possible to set the ringtone, volume and duration and many other things... try it! I set ringtone 2 to be the loudest ever, now I'd like to read the battery level

voyagerft, 2024-03-26:
can anyone read the battery level?

Puppetmaster91, 2024-04-02:
@voyagerft Nice work, if we want it to be auto recognizeable in the futrue. Commit it to this 
https://github.com/zigpy/zha-device-handlers/pull/2557/commits/3306f730361f5f02a6efbd6c6d77ea388964d7d7

@ioangogo already added a couple of clusters 

```
TUYA_BATTERY_ATTR = 0x020F  # [0, 0, 0, 100] battery percentage
TUYA_ALARM_ATTR = 0x0168  # [0]/[1] Alarm!
TUYA_ALARM_DURATION_ATTR = 0x0267  # [0,0,0,10] alarm duration in seconds: 0-1800
TUYA_MELODY_ATTR = 0x0466  # [5] Melody
TUYA_VOLUME_ATTR = 0x0474  # [0]/[1]/[2] Volume 0-low, 2-high
TUYA_TAMPER_ATTR = 0x0000                           <-------- missing
TUYA_TAMPER_ALARM_ATTR = 0x0000             <-------- missing
TUYA_ALARM_MODE_ATTR = 0x0000                <-------- missing
TUYA_CHARGE_STATE_ATTR = 0x0000               <-------- missing
TUYA_ALARM_STATE_ATTR = 0x0000                 <-------- missing
```

I have tried to reach out to the company who makes this to get more info regarding this sirens cluster but without any luck.
We're missing a way to read these clusters of the device, but i do not persist the proper skill to reverse engineer it.

piggei, 2024-04-17:
I think the battery sensor is fake and always set to 100%. I tried with a half charged battery and the app shows full charge.
BTW the app shows when I charge the battery via usb but not via solar panel.
Did anyone do some tries about it?

remoke1985, 2024-08-16:
Hi there,

I am a bit of a Noob in this matter and maybe you could help me out? 
After a few hours of trying I manage to pair the sirene and button. But my problem is, that when I turn the button "on" in HA (Lovelcae card)  it wont make a sound or shows me any sign of life.
So what I did to test it if the device was ok, is: push the button on the device (wall sensor, on the back of the device). This makes a really loud noise, so the device is working when its tempered. 

I have no clue what to do to make it work with a alarm card or otherwise. 

Thanks in advance

masi, 2024-08-19:
@remoke1985 You need to use zha.set_zigbee_cluster_attribute to configure the device, eg to set the melody type 13
```
    service: zha.set_zigbee_cluster_attribute
    data:
      # change <MAC-address> in the line below with the address of the siren
      ieee: <MAC-address>
      endpoint_id: 1
      cluster_id: 6
      cluster_type: in
      attribute: 0x466
      value: 0x0D
```
You find the mac address as "IEEE" under "Zigbee info".

Use the postings as starting points for clusters, endpoints, attributes and their possible values.
"Manage Zigbee device" will open the dialogue seen in the screenshots above.

To play a sound (turn on):
```
  - service: zha.issue_zigbee_cluster_command
    data:
      ieee: <MAC-address>
      endpoint_id: 1
      cluster_id: 6
      cluster_type: in
      command: 0x01
      command_type: server
      params: {}
```

asiersan, 2024-09-04:
I can confirm that modern sirens comes with Tuya Neo Outdoor Siren with Solar (TZE204_nlrfgpny) same as this configuration quirks is runing, 
Only i have to change:
MODELS_INFO: [
            ("_TZE200_nlrfgpny", "TS0601"),  
to 
MODELS_INFO: [
            ("_TZE204_nlrfgpny", "TS0601"),
in quirks

wappinghigh, 2024-10-13:
This is great but I don't want to edit all this in my yaml 
I look forward to a formal integration 
Thanks

tapionx, 2025-02-03:
hello there, is anyone working on this device? I'm available to pay a reward if this will become officially supported in ZHA

masi, 2025-02-03:
Is the documentation for quirks v2 ready? Last time I checked I found nothing I could understand :)

prairiesnpr, 2025-02-03:
> hello there, is anyone working on this device? I'm available to pay a reward if this will become officially supported in ZHA

Are you able to test a quirk?

closed 2025-02-20 by TheJulianJES

tapionx, 2025-02-20:
hey @prairiesnpr sorry I didn't answer. I don't have the siren right now but I ordered it and I should be able to test it soon.
I saw the PR, what the timeline for having it live in home assistant? I'm not familiar with the release process here.
If that will turn out to actually work, as I said, I'm happy to pay you a couple of beers 🍻

pleone, 2025-02-25:
hi @prairiesnpr i saw #3885 was merged in dev. Any plan to merge it to master? 
Or, alternative question, how can i use the dev branch of zha? 

thanks!

pleone, 2025-03-13:
@prairiesnpr updated HA to 2025.3.2 but still I can only discover the device but I have no actions, triggers or info.
any hint?

tapionx, 2025-03-14:
@pleone is yours TZE200 or TZE284?
I received a TZE284 and I managed to make it work with this change: https://github.com/zigpy/zha-device-handlers/pull/3945

pleone-DQ, 2025-03-14:
ehy @tapionx  thanks! I don't remember right now, buy should be 284. 
How can I try your change ?  Just editing the quirks? 

Thanks!

tapionx, 2025-03-14:
just pick the file from my PR and put it in the quirks.
You need to configure home assistant to use ZHA quirks ([like this](https://community.home-assistant.io/t/how-to-setup-local-zha-quirks/341226/2), I don't understand why this is not clearly explained in the documentation) and then restart home assistant.

pleone, 2025-03-14:
Oh finally, i was struggling on how to enable/configure quirks! Thanks for sharing, I will try it!

PR #3885 (2025-02-20): Add Tuya siren `TZE200_nlrfgpny` variant
## Proposed change

Add Tuya Siren `TZE200_nlrfgpny`

## Additional information

Closes: #2375
## Checklist

- [x] The changes are tested and work correctly
- [x] `pre-commit` checks pass / the code has been formatted using Black
- [ ] Tests have been added to verify that the new code works