zigpy/zha-device-handlers #2377: Tuya water valve (TS0049 by _TZ3210_0jxeoadc)

pcxsam, 2023-05-04:
### Problem description

Please can a handler be created to support a Tuya water valve controller (TS0049).  The device can be added to home assistant via zha but no entities are available to control the valve.

### Solution description

Creation of a new handler to support the device

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0000",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3210_0jxeoadc",
  "model": "TS0049",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.4.6",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.10",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/London",
    "os_name": "Linux",
    "os_version": "6.1.25",
    "supervisor": "2023.04.1",
    "host_os": "Home Assistant OS 10.1",
    "docker_version": "23.0.3",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
[... 321 lines cut ...]
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

idem2lyon, 2023-05-11:
It would be great, I have same device !!

fouram, 2023-05-19:
I just bought one of these as well, hoping to get it working. Seems like not yet! Would love to have support, as a renter its nearly impossible to get a hose valve I can simply switch on and off without HACS and cloud services and timings and checking weather schedules (looking at you, Orbit). My setup would greatly benefit from the simplicity of my end points just being on/off switches!

DonnySus, 2023-05-23:
I also bought one of these irrigation valves. ZHA has not a quirk for a moment (at least I found nothing), but Zigbee2MQTT already developed some workaround. Can anybody from development team check following link it the provided code helo to build up a quirk for that device?
The link for Zigbee2MQTT code: [https://github.com/Koenkk/zigbee2mqtt/issues/15124](url)

MiguelAngelLV, 2023-06-07:
I have one too. 

I try create a quirks but only get a show a useless switch. The switch stage change but it don't do nothing.

pazzotranquillo, 2023-07-05:
Have you guys found a custom solution for this? It would really great if this valve could be added to ZHA.

Margriko, 2023-07-10:
I tried to put the pieces together for a custom quirk but was not successful. The quirk was applied successfully, switch status is shown in the UI, it reacts to the physical button press, but it is not possible to control it from the UI. Zigbee2MQTT implementation had to send some custom command to activate it, but I am a complete novice with ZHA quirks and Zigbee itself.
Anyway, if anyone wants to continue, here is the starting point for a quirk.
Battery status is not shown correctly as it has 3 states instead of percentage and it needs to be implemented. Also I listed all data points I found in the logs.

`from typing import Dict

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl import foundation
from zigpy.zcl.clusters.general import Basic, Groups, Identify, OnOff, Ota, Scenes, Time
from zigpy.zcl.clusters.smartenergy import Metering

from zhaquirks import DoublingPowerConfigurationCluster
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    TUYA_SET_DATA,
    TuyaLocalCluster,
    TuyaNewManufCluster,
    TuyaZBExternalSwitchTypeCluster,
    TuyaPowerConfigurationCluster2AA,
)
from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    EnchantedDevice,
    TuyaMCUCluster,
    TuyaOnOff,
    TuyaOnOffNM,
    TuyaPowerConfigurationCluster,
)

class TuyaFJKZYellowValveManufCluster(TuyaMCUCluster):
    """Manufacturer Specific Cluster for the _TZ3210_0jxeoadc water valve sold as RAIN SEER FJKZ005C-Y."""

    # class BatteryState(t.enum8):
    #     """Battery state option."""
        
    #     Low = 0x00
    #     Medium = 0x01
    #     High = 0x02
        
    
    attributes = TuyaMCUCluster.attributes.copy()
    
    dp_to_attribute: Dict[int, DPToAttributeMapping] = {
        101: DPToAttributeMapping(
            TuyaOnOff.ep_attribute,
            "on_off",
        ),
        102: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "dp_2",
        ),
        103: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "dp_3",
        ),
        105: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "dp_5",
        ),
        106: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "dp_6",
        ),
        109: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "dp_9",
        ),
        110: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "dp_110",
        ),
        111: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "dp_111",
        ),
        115: DPToAttributeMapping(
            TuyaPowerConfigurationCluster.ep_attribute,
            "battery_status",
        ),
    }

    data_point_handlers = {
        101: "_dp_2_attr_update",
        102: "_dp_2_attr_update",
        103: "_dp_2_attr_update",
        105: "_dp_2_attr_update",
        106: "_dp_2_attr_update",
        109: "_dp_2_attr_update",
        110: "_dp_2_attr_update",
        111: "_dp_2_attr_update",
        115: "_dp_2_attr_update",
    }
    

class TuyaFJKZYellowValve(EnchantedDevice):
    signature = {
        MODELS_INFO: [("_TZ3210_0jxeoadc", "TS0049")],
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=260 device_type=266
            # device_version=1
            # input_clusters=[0, 61184]
            # output_clusters=[10, 25]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    TuyaFJKZYellowValveManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    TuyaOnOff,
                    TuyaPowerConfigurationCluster,
                    TuyaFJKZYellowValveManufCluster,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
        },
    }
`

martinw72, 2023-07-17:
log output:

"data": {
    "ieee": "**REDACTED**",
    "nwk": 64536,
    "manufacturer": "_TZ3210_0jxeoadc",
    "model": "TS0049",
    "name": "_TZ3210_0jxeoadc TS0049",
    "quirk_applied": false,
    "quirk_class": "zigpy.device.Device",
    "manufacturer_code": 4417,
    "power_source": "Battery or Unknown",
    "lqi": 180,
    "rssi": -55,
    "last_seen": "2023-07-16T22:01:13",
    "available": false,
    "device_type": "EndDevice",
    "signature": {
      "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
      "endpoints": {
        "1": {
          "profile_id": "0x0104",
          "device_type": "0x0000",
          "input_clusters": [
            "0x0000",
            "0xef00"
          ],
          "output_clusters": [
            "0x000a",
            "0x0019"
          ]
        }
      },
      "manufacturer": "_TZ3210_0jxeoadc",
      "model": "TS0049"
    },
    "active_coordinator": false,
    "entities": [],
    "neighbors": [],
    "routes": [],
    "endpoint_names": [
      {
        "name": "ON_OFF_SWITCH"
      }
    ],
    "user_given_name": "Glasshausbew\u00e4sserung",
    "device_reg_id": "5d5ba450476da71385ab19e332022105",
    "area_id": "glashaus",
    "cluster_details": {
      "1": {
        "device_type": {
          "name": "ON_OFF_SWITCH",
          "id": 0
        },
        "profile_id": 260,
        "in_clusters": {
          "0xef00": {
            "endpoint_attribute": null,
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0000": {
            "endpoint_attribute": "basic",
            "attributes": {
              "0x0004": {
                "attribute_name": "manufacturer",
                "value": "_TZ3210_0jxeoadc"
              },
              "0x0005": {
                "attribute_name": "model",
                "value": "TS0049"
              }
            },
            "unsupported_attributes": {}
          }
        },
        "out_clusters": {
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x000a": {
            "endpoint_attribute": "time",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }

coutadeurf, 2023-07-21:
Hi,

They had the same issue in Z2M, and this was because of that:
"However, the main point that they found was that the valve wouldn't work until the sendDataPoint...() function was called with cmd = 0x04 ("Send commands")".
After adding that value, it worked. I am not fluent enough in quirks to propose something but that may help someone to update the proposed solution

zenalien, 2023-07-22:
hi everyone, any idea, if this will be ever available please ?

pnneil, 2023-07-27:
FWIW I would like to add my voice to this request :)

coutadeurf, 2023-07-27:
I have a test to do but as I am not close to the device this week, so I cannot test it myself now. 
If you want, you can do it: in the given quirks, replace TuyaOnOff by TuyaOnOffNM in the 2 places where it appears: I found a generic tuya valve quirks that is using TuyaOnOffNM. If it doesn't work, other option would be to understand what the quirk I found is doing differently but it isnquite complex...

pnneil, 2023-07-27:
Thanks, but the behaviour is the same as described above for me...

coutadeurf, 2023-07-28:
I finally decided to go with Z2M because I have another device that is not supported, I was just at the beginning of my setup and estimated that I will spend less time setting up Z2M than trying to blindly fix the quirks.

pnneil, 2023-07-30:
> I finally decided to go with Z2M because I have another device that is not supported, I was just at the beginning of my setup and estimated that I will spend less time setting up Z2M than trying to blindly fix the quirks.

I can understand that, for me this is the only device out of quite a few which is not yet supported, so I'm sticking with ZHA for the time being.  Good luck!

pnneil, 2023-08-01:
So I found this over at Z2MQTT, which is the converter for the device.  I can see that a special conversion has been set up for the Switch and Timer for the valve, but I have no idea how to convert this into the required code for a quirk.

Any ideas?
 
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

**const TS0049_ValueConverter = {
    Switch:{
        to: async (value, meta) => {
            const entity = meta.device.endpoints[0];
            await tuya.sendDataPointBool(entity, 101, value.toUpperCase() === 'ON', 0x04);
        },
        from: (value) => {
            return {false: 'OFF', true: 'ON'}[value];
        },
    },
    IrrigationTimer:{
        to: async (value, meta) => {
            const entity = meta.device.endpoints[0];
            await tuya.sendDataPointValue(entity, 111, value, 0x04);
        },
        from: (value) => {
            return value;
        },
    },
}**

const definition = {
    fingerprint: tuya.fingerprint('TS0049', ['_TZ3210_0jxeoadc']),
    model: 'TS0049',
    vendor: 'TuYa',
    description: 'Water Valve',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetLocalTime,
    configure: tuya.configureMagicPacket,
    exposes: [
        tuya.exposes.errorStatus(),
        tuya.exposes.switch(),
        tuya.exposes.countdown().withValueMin(0).withValueMax(255).withUnit('minutes')
            .withDescription('Max on time in minutes'),
        tuya.exposes.batteryState(),
    ],
    meta: {
        tuyaDatapoints: [
            [26, 'error_status', tuya.valueConverter.raw],
            [101, 'state', TS0049_ValueConverter.Switch],
            [111, 'countdown', TS0049_ValueConverter.IrrigationTimer],
            [115, 'battery_state', tuya.valueConverter.batteryState],
        ],
    },
};
module.exports = definition;

zenalien, 2023-08-01:
> I finally decided to go with Z2M because I have another device that is not supported, I was just at the beginning of my setup and estimated that I will spend less time setting up Z2M than trying to blindly fix the quirks.

Is the device fully functionnal with Z2M ?
By chance wou:ld you know what entities are there ?

Thank you .

coutadeurf, 2023-08-01:
Hi,

Yes, the device is working fine in Z2M. Here are the entities that has been
added to HA:
[image: image.png]
Regards

pnneil, 2023-08-06:
So with some trial and error, (mostly the latter :) ) and borrowing other bits of code I found a way to control the valve.  It is throwing warning messages which I haven't eliminated yet but it is a start...

```python

from typing import Any, Dict, Optional, Union

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl import foundation
from zigpy.zcl.clusters.general import Basic, Identify, OnOff, Ota, Time
from zigpy.zcl.clusters.smartenergy import Metering

from zhaquirks import DoublingPowerConfigurationCluster
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    TUYA_SEND_DATA,
#    TuyaLocalCluster,
)
from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    EnchantedDevice,
    TuyaMCUCluster,
    TuyaOnOff,
    TuyaPowerConfigurationCluster,
)

class TuyaValveFamilyCluster(TuyaMCUCluster):
    """On/Off Tuya family cluster with extra device attributes"""

    attributes = TuyaMCUCluster.attributes.copy()
    
    async def command(
        self,
        command_id: Union[foundation.GeneralCommand, int, t.uint8_t],
        *args,
        manufacturer: Optional[Union[int, t.uint16_t]] = None,
        expect_reply: bool = True,
        tsn: Optional[Union[int, t.uint8_t]] = None,
        **kwargs: Any,
    ):
        """Override the default Cluster command."""
        self.debug("Setting the NO manufacturer id in command: %s", command_id)
        return await super().command(
            TUYA_SEND_DATA,
            *args,
            manufacturer=foundation.ZCLHeader.NO_MANUFACTURER_ID,
            expect_reply=expect_reply,
            tsn=tsn,
            **kwargs,
        )    

    dp_to_attribute: Dict[int, DPToAttributeMapping] = {
        26: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "error_status",
        ),
        101: DPToAttributeMapping(
            TuyaOnOff.ep_attribute,
            "on_off",
        ),
        110: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "dp_110",
        ),        
        111: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "irrigation_time",
        ),
        115: DPToAttributeMapping(
            TuyaPowerConfigurationCluster.ep_attribute,
            "battery_state",
        ),
    }

    data_point_handlers = {
        26: "_dp_2_attr_update",
        101: "_dp_2_attr_update",
        110: "_dp_2_attr_update",
        111: "_dp_2_attr_update",
        115: "_dp_2_attr_update",
    }

class TuyaIrrigationValve(EnchantedDevice):
    """Tuya green irrigation valve device."""
    signature = {
        MODELS_INFO: [("_TZ3210_0jxeoadc", "TS0049")],
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=260 device_type=0
            # device_version=1
            # input_clusters=[0, 61184]
            # output_clusters=[10, 25]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    TuyaValveFamilyCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    TuyaOnOff,
                    TuyaPowerConfigurationCluster,
                    TuyaValveFamilyCluster
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
        },
    }

danpeig, 2023-08-08:
I will try it out If I have some time during the weekend.... what sort of warning messages do you get?

pnneil, 2023-08-08:
I get these: 
2023-08-08 12:42:14.504 WARNING (MainThread) [zigpy.zcl] [0x739A:1:0xef00] Unknown cluster command 5 b'\x00\x8ee\x01\x00\x01\x01'
2023-08-08 12:42:14.614 WARNING (MainThread) [zigpy.zcl] [0x739A:1:0xef00] Unknown cluster command 5 b'\x00\x8fn\x00\x00\x02\x00\x01'

danpeig, 2023-08-09:
Hi,

I did some minor changes to expose the "irrigation_timer" in the cluster settings in ZHA. This way you can set a different maximum duration for the cycle directly from HA.

Still trying to understand the warnings....

```python

from typing import Any, Dict, Optional, Union

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl import foundation
from zigpy.zcl.clusters.general import Basic, Identify, OnOff, Ota, Time
from zigpy.zcl.clusters.smartenergy import Metering

from zhaquirks import DoublingPowerConfigurationCluster
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    TUYA_SEND_DATA,
#    TuyaLocalCluster,
)
from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    EnchantedDevice,
    TuyaMCUCluster,
    TuyaOnOff,
    TuyaPowerConfigurationCluster,
)

class TuyaValveFamilyCluster(TuyaMCUCluster):
    """On/Off Tuya family cluster with extra device attributes"""

    attributes = TuyaMCUCluster.attributes.copy()
    attributes.update(
        {
            0xEF01: ("irrigation_time", t.uint32_t, True),
            0xEF02: ("dp_110", t.uint32_t, True),
            0xEF03: ("error_status", t.uint32_t, True),
        }
    )
    
    async def command(
        self,
        command_id: Union[foundation.GeneralCommand, int, t.uint8_t],
        *args,
        manufacturer: Optional[Union[int, t.uint16_t]] = None,
        expect_reply: bool = True,
        tsn: Optional[Union[int, t.uint8_t]] = None,
        **kwargs: Any,
    ):
        """Override the default Cluster command."""
        self.debug("Setting the NO manufacturer id in command: %s", command_id)
        return await super().command(
            TUYA_SEND_DATA,
            *args,
            manufacturer=foundation.ZCLHeader.NO_MANUFACTURER_ID,
            expect_reply=expect_reply,
            tsn=tsn,
            **kwargs,
        )    

    dp_to_attribute: Dict[int, DPToAttributeMapping] = {
        26: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "error_status",
        ),
        101: DPToAttributeMapping(
            TuyaOnOff.ep_attribute,
            "on_off",
        ),
        110: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "dp_110",
        ),        
        111: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "irrigation_time",
        ),
        115: DPToAttributeMapping(
            TuyaPowerConfigurationCluster.ep_attribute,
            "battery_percentage_remaining",
        ),
    }

    data_point_handlers = {
        26: "_dp_2_attr_update",
        101: "_dp_2_attr_update",
        110: "_dp_2_attr_update",
        111: "_dp_2_attr_update",
        115: "_dp_2_attr_update",
    }

class TuyaIrrigationValve(EnchantedDevice):
    """Tuya green irrigation valve device."""
    signature = {
        MODELS_INFO: [("_TZ3210_0jxeoadc", "TS0049")],
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=260 device_type=0
            # device_version=1
            # input_clusters=[0, 61184]
            # output_clusters=[10, 25]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    TuyaValveFamilyCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    TuyaOnOff,
                    TuyaPowerConfigurationCluster,
                    TuyaValveFamilyCluster
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
        },
    }

```

pnneil, 2023-08-09:
Hi danpieg, After reloading, the new quirk, I didn't see any difference but after deleting the device and re-adding it I am getting a Battery reading, although it is only 2% (seems very low).  I haven't seen any additional Entities exposed regarding the Irrigation timer, or is this meant to be controlled another way?  The switch is definitely still working :)
Thanks for your help!

MiguelAngelLV, 2023-08-09:
The battery value is not  a percentage. it have 3 values:

Low = 0x00
Medium = 0x01
High = 0x02

So, is necessary a custom battery converter.

pnneil, 2023-08-09:
> The battery value is not a percentage. it have 3 values:
> 
> Low = 0x00 Medium = 0x01 High = 0x02
> 
> So, is necessary a custom battery converter.

Thanks, That explains why the battery values I was getting were 1 or 2.

danpeig, 2023-08-09:
Conversion from 0, 1, 2 to %:

```python

115: DPToAttributeMapping(
            TuyaPowerConfigurationCluster.ep_attribute,
            "battery_percentage_remaining",
            converter=lambda x: x * 50,  # Adjust percen
tage
        ),
```

MiguelAngelLV, 2023-08-09:
I would use (x+1)*33

HIGH = 99
MEDIUM = 66
LOW = 33

```
115: DPToAttributeMapping(
            TuyaPowerConfigurationCluster.ep_attribute,
            "battery_percentage_remaining",
            converter=lambda x: (x+1) * 33,  # Adjust percen
tage
        ),
```

MiguelAngelLV, 2023-08-09:
A little improvement battery

```python
from typing import Any, Dict, Optional, Union

import zigpy.types as t
from zhaquirks import DoublingPowerConfigurationCluster
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    TUYA_SEND_DATA, TuyaLocalCluster,
)
from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    EnchantedDevice,
    TuyaMCUCluster,
    TuyaOnOff,
)
from zigpy.profiles import zha
from zigpy.zcl import foundation
from zigpy.zcl.clusters.general import Basic, Ota, Time, PowerConfiguration

class TuyaValveFamilyBattery(TuyaLocalCluster, DoublingPowerConfigurationCluster):
    _values = [10, 50, 90]
    _CONSTANT_ATTRIBUTES = {
        PowerConfiguration.attributes_by_name["battery_quantity"].id: 4,
        PowerConfiguration.attributes_by_name["battery_size"].id: PowerConfiguration.BatterySize.AAA
    }

    def _update_attribute(self, attrid, value):
        if attrid == self.BATTERY_PERCENTAGE_REMAINING:
            value = self._values[value]
        super()._update_attribute(attrid, value)

class TuyaValveFamilyCluster(TuyaMCUCluster):
    """On/Off Tuya family cluster with extra device attributes"""

    attributes = TuyaMCUCluster.attributes.copy()
    attributes.update(
        {
            0xEF01: ("irrigation_time", t.uint32_t, True),
            0xEF02: ("dp_110", t.uint32_t, True),
            0xEF03: ("error_status", t.uint32_t, True),
        }
    )

    async def command(
            self,
            command_id: Union[foundation.GeneralCommand, int, t.uint8_t],
            *args,
            manufacturer: Optional[Union[int, t.uint16_t]] = None,
            expect_reply: bool = True,
            tsn: Optional[Union[int, t.uint8_t]] = None,
            **kwargs: Any,
    ):
        """Override the default Cluster command."""
        self.debug("Setting the NO manufacturer id in command: %s", command_id)
        return await super().command(
            TUYA_SEND_DATA,
            *args,
            manufacturer=foundation.ZCLHeader.NO_MANUFACTURER_ID,
            expect_reply=expect_reply,
            tsn=tsn,
            **kwargs,
        )

    dp_to_attribute: Dict[int, DPToAttributeMapping] = {
        26: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "error_status",
        ),
        101: DPToAttributeMapping(
            TuyaOnOff.ep_attribute,
            "on_off",
        ),
        110: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "dp_110",
        ),
        111: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "irrigation_time",
        ),
        115: DPToAttributeMapping(
            TuyaValveFamilyBattery.ep_attribute,
            "battery_percentage_remaining",
        ),
    }

    data_point_handlers = {
        26: "_dp_2_attr_update",
        101: "_dp_2_attr_update",
        110: "_dp_2_attr_update",
        111: "_dp_2_attr_update",
        115: "_dp_2_attr_update",
    }

class TuyaIrrigationValve(EnchantedDevice):
    """Tuya green irrigation valve device."""
    signature = {
        MODELS_INFO: [("_TZ3210_0jxeoadc", "TS0049")],
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=260 device_type=0
            # device_version=1
            # input_clusters=[0, 61184]
            # output_clusters=[10, 25]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    TuyaValveFamilyCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    TuyaOnOff,
                    TuyaValveFamilyBattery,
                    TuyaValveFamilyCluster
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
        },
    }
```

I added a rule for battery (low => 10%, medium => 50%, high => 90%), they can be modified easily.

And added the battery size and quantity too.

pnneil, 2023-08-10:
Thanks MiguelAngelLV. its working well!

pazzotranquillo, 2023-08-11:
you guys are amazing! Many thanks!

MiguelAngelLV, 2023-08-18:
I check that irrigation timer works, but don't show the input, you need go to configuration, open the clusters and set the value.

Any way to get a input in device config?

813648, 2023-08-30:
Está a funcionar 5*. Muito Obrigado!!! - It's working 5*.  Thank you very much!!!

813648, 2023-08-30:
Está a funcionar 5*. Muito Obrigado!!! - It's working 5*. Thank you very much!!!
[image]

raddacle, 2023-09-28:
The quirk works great! Though I'm having an issue where it automatically turns off after a minute every time. I believe this is a default setting that you could disable through the tuya app, but how can I disable it through ZHA?

Matze89x, 2023-11-08:
When will this be officially adopted? I'm currently still using the custom version in Home Assistant. Thank you very much.

MiguelAngelLV, 2023-11-08:
> When will this be officially adopted? I'm currently still using the custom version in Home Assistant. Thank you very much.

The quirk works, but, it is not finished...  I think that It needs "a little of love" before merge in official.

guciopl, 2023-11-21:
> Está a funcionar 5*. Muito Obrigado!!! - It's working 5*. Thank you very much!!! [image]

I have same Tuya valve but green and it needs RF433 gateway as it cannot connect to wifi on its own. Does this yellow require gateway as well or it is different model? I cant get my green one to work with HA at all.

coutadeurf, 2023-11-21:
Hi,

Yellow is zigbee powered so you need a zigbee adapter to make it work.

MiguelAngelLV, 2023-11-21:
@guciopl there are a lot os versions. We are talking about zigbee version (powered with AC or battery, but, zigbee).

If the yours use RF433, it is not valid version with ZHA.

moryoav, 2023-12-20:
> The quirk works great! Though I'm having an issue where it automatically turns off after a minute every time. I believe this is a default setting that you could disable through the tuya app, but how can I disable it through ZHA?
> 
> EDIT: Nevermind, found it. For anyone who runs into this too, go to Manage Zigbee Device, select cluster TuyaValveFamilyCluster, and then select attribute irrigation_time. From there you can set any maximum minutes that it will be on; and there's no manufacturer code needed.

It's very limiting that the irrigation time cannot be controlled directly like any other device control. Can it be added?

rdeva31, 2024-03-02:
@MiguelAngelLV when needs to be done in order for this  be turned into a PR?  I'm happy to take this forward if you tell me where the gaps are.

danpeig, 2024-03-03:
It was failing to pass verification checks. I have no proficiency with the python to fix the errors.

MiguelAngelLV, 2024-03-15:
Any one have problem with battery level? After 4 months show 90% yet and I don't think could be possible...

sada1410, 2024-04-03:
Hello, 

I used the Quirk and it worked great. However, after a while my 4 valves stopped working.

I can still read but can no longer write. I can also see the battery level. When I reconfigure the valves, I get the message that TuyaOnOff cannot be bound. 

What could be the reason for this? Can anyone help me? 

[image]

sada1410, 2024-04-13:
If I try to reconfigure the device the following error log appears: 

`2024-04-13 14:55:45.946 DEBUG (MainThread) [homeassistant.components.zha.core.cluster_handlers] [0x2D11:1:0x0006]: Failed to bind 'on_off' cluster: Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/zha/core/cluster_handlers/__init__.py", line 203, in bind
    res = await self.cluster.bind()
          ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/zhaquirks/tuya/__init__.py", line 572, in bind
    await self.spell_attribute_reads()
  File "/usr/local/lib/python3.12/site-packages/zhaquirks/tuya/__init__.py", line 586, in spell_attribute_reads
    await basic_cluster.read_attributes(attr_to_read)
  File "/usr/local/lib/python3.12/site-packages/zigpy/zcl/__init__.py", line 528, in read_attributes
    result = await self.read_attributes_raw(to_read, manufacturer=manufacturer)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/zigpy/zcl/__init__.py", line 377, in request
    return await self._endpoint.request(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/zigpy/endpoint.py", line 253, in request
    return await self.device.request(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/zigpy/device.py", line 339, in request
    await send_request()
  File "/usr/local/lib/python3.12/site-packages/zigpy/application.py", line 841, in request
    await self.send_packet(
  File "/usr/local/lib/python3.12/site-packages/bellows/zigbee/application.py", line 931, in send_packet
    raise zigpy.exceptions.DeliveryError(
zigpy.exceptions.DeliveryError: Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>`

It is possible to read and write the irrigation value. Only TuyaOnOff doesn't work.

sada1410, 2024-04-13:
I tried to connect the valve to an original tuya Zigbee gateway. The valve is working. So maybe something is missing in the communication and than the valve is unavailable. How can I debug the messages and how to check if commands are missing? Anyone has some ideas? I tried a lot of things in the quirck but nothing helped.
My gain is to have a working and reliable quirk for this valve.

narthollis, 2024-05-25:
I would like to quickly add that I have just used the custom profile posted by MiguelAngelLV on Aug 9, 2023 and that it just worked for me. Many thanks.

alexdaszek, 2024-05-28:
> The quirk works great! Though I'm having an issue where it automatically turns off after a minute every time. I believe this is a default setting that you could disable through the tuya app, but how [...]

Did this end up actually working for you? I do the same thing, but the value never sticks and always goes back to 1 after writing it. 

[image]

I'm using this for drip irrigation, ideally I want HA to turn it on and off, but because it works this way I want a 4 hour time, so I set it to 240 minutes and write, and read back and it says 240. But then if I check it again the next day it's back to a value of 1 minute. I checked the python code of the quirk, doesn't seem to be stored in there to change manually so not sure where this value lives if there's other options for changing it. Anything else I can try?

moryoav, 2024-05-29:
This works for me fine. Only if I re-pair the device with the zigbee router
it resets to default, but in day-to-day operations this works fine.
If you can't manage then you can always create an automation that'll run
daily and put that value again automatically.
Check out the service "zha.set_zigbee_cluster_attribute"

alexdaszek, 2024-05-30:
> This works for me fine. Only if I re-pair the device with the zigbee router it resets to default, but in day-to-day operations this works fine. If you can't manage then you can always create an [...]

I think I got that working, thank you. I'm still having issues with the device itself, it doesn't do anything when turning on or off. Originally it was working OK, there was a long lag before turning it on in HA and it actually turning on and now I can't get it to work at all. But that's another issue...

Here's what worked for me as far as having an automation that writes the irrigation_time value before turning the switch on:

[image]

[image]

[image]

I got the ieee value from ZHA, and all of the other data from the "manage device" menu I shared an image of in my original comment. It gives you the attribute and cluster IDs as hex, but the service needs them as an int so you have to convert them. 

Just an interesting side note that may or may not be relevant, is the debug node reports from the service that entity_id is undefined but it isn't one of the values listed by the service so not sure what's going on there. I assume it's optional, it has the ieee value after all.

[image]
[image]

moryoav, 2024-05-30:
I recently discovered another issue with zha and this valve, sometimes sending the command to turn it on doesn't turn it on. It shows as if it's on but it's not. Physically pressing the button on the valve turns it on if that happens, but this is not a good solution obviously. So what I realized was that if I send the command to turn it on and then off, repeat that a few times, then after a few times it successfully wakes up the device and turning it on succeeds. Anyone else seeing this behavior? I don't know how to "verify" it's really on, I've been seen a zigbee device that's reporting to be on when it's not (I have a few of these valves and I'm seeing this behavior in all of them, so this is not a unique device failure)

alexuco81, 2024-06-23:
https://github.com/zigpy/zha-device-handlers/issues/2377#issuecomment-2138725256
Same here. Any ideas?
Regards.

alexdaszek, 2024-06-29:
> I recently discovered another issue with zha and this valve, sometimes sending the command to turn it on doesn't turn it on. It shows as if it's on but it's not. Physically pressing the button on [...]

I experienced this too from my above issues, I think it just that the device doesn't have a good connection. It would work fine inside my house and be responsive, but when used outside for the garden as intended the signal just isn't good enough, even with boosters as close as I could manage.

moryoav, 2024-07-15:
This may be related to connection, but I'm not 100% convinced. I have a few of those devices, Today I noticed it had an LQI of 200 and RSSI of -50dbm, which should be really ok, it's relatively high, shouldn't be problematic at all. I had to switch the valve on and off like 5 times or more until it actually switched on. Now, connectivity issues are possible, definitely ones that are momentary before going back to normal, however, the big issue here is that the zha device reports as "on" when it is not. With other devices this is not the case, if my zha light switch loses connectivity, for example, or has a weak connectivity, then when trying to turn it on, zha will tell me it cannot turn it on. So I'm hoping it's some missing functionality in this quirk, where it doesn't properly report "success" or "failure" after switching it on or off, and zha therefore doesn't know the correct state of the device. I really hope someone who knows something about those quirks will be able to shed some light here. Otherwise I'm going to try switching to zigbee2mqtt to see what the situation over there with these valves.

moryoav, 2024-07-18:
1. I have a slight suspicion that the issues I have with this valve are related to the batteries, I added fresh new batteries to all my valves and will give it a few days to see if things improve.
2. In any case, I found a really cool solution to know if the valve is really open or not, and if not to just try again. It's using a zigbee vibration sensor placed on top of the valve. I created a detailed writeup about it here:
https://smarthome.yoavmor.com/home-assistant/ensuring-zigbee-valve-reliability-with-vibration-sensors-in-home-assistant/

gunarser, 2024-07-30:
Hi!

I've got a slightly different TS0049 water valve of different manufacturer. What do I need to change in order to get it work in HA?

The signature:
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0000",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe001"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3000_gjpgagal",
  "model": "TS0049",
  "class": "zigpy.device.Device"
}

danpeig, 2024-07-30:
If it using the same underlying firmware, just changing:

`MODELS_INFO: [("_TZ3210_0jxeoadc", "TS0049")],`

to 

`MODELS_INFO: [("_TZ3000_gjpgagal", "TS0049")],`

would work.

gunarser, 2024-07-30:
> If it using the same underlying firmware, just changing:
> 
> `MODELS_INFO: [("_TZ3210_0jxeoadc", "TS0049")],`
> 
> to
> 
> `MODELS_INFO: [("_TZ3000_gjpgagal", "TS0049")],`
> 
> would work.

Thanks for the quick reply!

I already tried this change but the quirk does not get loaded for the particular device. That's why I realized that I am perhaps missing something else.
I changed this section in the quirk:

class TuyaIrrigationValve(EnchantedDevice):
    """Tuya green irrigation valve device."""
    signature = {
        MODELS_INFO: **[("_TZ3000_gjpgagal", "TS0049")],**
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=260 device_type=0
            # device_version=1
            # input_clusters=[0, 61184]
            # output_clusters=[10, 25]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    TuyaValveFamilyCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
        },
    }

How can I find out the reason why the quirk is not loaded?

zaratustraelimpio, 2024-08-30:
Hi, 

I have a model like the one in the post bought on aliexpress (https://es.aliexpress.com/item/1005004599430685.html?spm=a2g0o.order_list.order_list_main.5.6de2194dKwzi7q&gatewayAdapt=glo2esp) but HA does not recognize the model:

[image]

If I install the zha from this post it does not work, no controls show, can anyone help? Can someone share the purchase link for the exact model TS0049?, 

thanks,

regards

moryoav, 2024-08-30:
> Hi,
>
> I have a model like the one in the post bought on aliexpress (https://es.aliexpress.com/item/1005004599430685.html?spm=a2g0o.order_list.order_list_main.5.6de2194dKwzi7q&gatewayAdapt=glo2esp) [...]

I think we have the exact the device (the green one?), it's the same id. But your screenshot looks like you opened the Tuya integration and not ZHA, are you sure you have ZHA and paired it correctly with your zigbee network? 
I created this guide based on my experience with this exact model:

https://smarthome.yoavmor.com/home-assistant/adding-ts0049-irrigation-valve-to-zigbee-home-automation/

zaratustraelimpio, 2024-08-30:
Hi, 

I have it integrated with the Tuya cloud (I have a Tuya Zigbee gateway), is it necessary to integrate it with a Zigbee device that is compatible with HA for it to recognize it? 

thanks, 

regards.

[40 more replies]

PR #3737 (2025-01-20): Add Tuya Rain Seer valve `_TZ3210_0jxeoadc`
## Proposed change

Adds Tuya Rain Seer valve _TZ3210_0jxeoadc

Closes: #2624
Closes: #2377

## Additional information

Based on: https://github.com/Koenkk/zigbee-herdsman-converters/blob/f42aa5080ee933e8cde9fc08c67a88bf886d71ff/src/devices/tuya.ts#L8974

## Checklist

- [ ] The changes are tested and work correctly
- [x] `pre-commit` checks pass / the code has been formatted using Black
- [ ] Tests have been added to verify that the new code works
closed 2025-01-20 by TheJulianJES