zigpy/zha-device-handlers #2378: [Device Support Request] ZY-M100

diogo-madeira, 2023-05-04:
### Problem description

It apears in HA, but without any sensor.

### Solution description

Any

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE204_sxm7l9xa",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}

### Diagnostic information

{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.5.0",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.11",
    "docker": true,
    "arch": "x86_64",
    "timezone": "America/Recife",
    "os_name": "Linux",
    "os_version": "6.1.25",
    "supervisor": "2023.04.1",
    "host_os": "Home Assistant OS 10.1",
    "docker_version": "23.0.3",
    "chassis": "vm",
    "run_as_root": true
  },
[... 355 lines cut ...]
        }
      }
    }
  }
}

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

b2un0, 2023-05-10:
relates to https://github.com/Koenkk/zigbee2mqtt/issues/17483

Woll3, 2023-05-10:
> Details
> 
> Nwk: `0x9205` Pairing Log: [home-assistant_zha_2023-05-10T14-48-30.769Z.log](https://github.com/zigpy/zha-device-handlers/files/11443230/home-assistant_zha_2023-05-10T14-48-30.769Z.log)
> 
> --
> 
> the device looks like the same as `_TZE204_ztc6ggyl ` which is available in
> 
> https://github.com/zigpy/zha-device-handlers/blob/a221e682e9baffe95588eb6708d096cfc481cd5d/zhaquirks/tuya/ts0601_motion.py#L352
> 
> duplicated the `ts0601_motion.py` to my quirks folder and added the `_TZE204_sxm7l9xa` under the `_TZE204_ztc6ggyl` line. But the quirk does not applied. don't know why..

For me, this doesn't work.
I tried the following two quirks:

- [BlitzWolf IS-3/Tuya motion rechargeable occupancy sensor](https://github.com/zigpy/zha-device-handlers/blob/a221e682e9baffe95588eb6708d096cfc481cd5d/zhaquirks/tuya/ts0601_motion.py#L352) - only the MmwRadarMotionGPP lluminance measurement worked
- [Human Presence Sensor 24GHz](https://github.com/zigpy/zha-device-handlers/issues/2059#issuecomment-1459991922) - switch appeared in the device info

programmieramt, 2023-05-15:
Hello, I just had the same issues with zigbee2mqtt:

The _TZE204_ztc6ggyl has different "datapoints" then the _TZE204_sxm7l9xa so it is not done with just adding the manufacture id.
For z2mqtt you see the differents here: https://github.com/Koenkk/zigbee2mqtt/issues/17483#issuecomment-1547504532

Never worked with quirks but will try to check if possible

programmieramt, 2023-05-15:
Here is the description for the datapoints:

104 - Lux
105 - Presence
106 - Motion sensitivity 
107 - Max detection distance
108 - Min detection distance
109 - Target distance
110 - Detection delay
111 - Fading time

Woll3, 2023-05-15:
`"""ZY-M100 Human Presence Sensor"""
#https://github.com/zigpy/zha-device-handlers/issues/2378

#based on
#"""BlitzWolf IS-3/Tuya motion rechargeable occupancy sensor."""
#https://github.com/zigpy/zha-device-handlers/blob/a221e682e9baffe95588eb6708d096cfc481cd5d/zhaquirks/tuya/ts0601_motion.py#L352

import math
from typing import Dict, Optional, Tuple, Union

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl import foundation
from zigpy.zcl.clusters.general import (
    AnalogInput,
    Basic,
    GreenPowerProxy,
    Groups,
    Identify,
    Ota,
    Scenes,
    Time,
)
from zigpy.zcl.clusters.measurement import (
    IlluminanceMeasurement,
    OccupancySensing,
    RelativeHumidity,
    TemperatureMeasurement,
)
from zigpy.zcl.clusters.security import IasZone

from zhaquirks import Bus, LocalDataCluster, MotionOnEvent
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    MOTION_EVENT,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    DPToAttributeMapping,
    TuyaLocalCluster,
    TuyaManufCluster,
    TuyaNewManufCluster,
)
from zhaquirks.tuya.mcu import TuyaMCUCluster

ZONE_TYPE = 0x0001

class TuyaOccupancySensing(OccupancySensing, TuyaLocalCluster):
    """Tuya local OccupancySensing cluster."""

class TuyaAnalogInput(AnalogInput, TuyaLocalCluster):
    """Tuya local AnalogInput cluster."""

class TuyaIlluminanceMeasurement(IlluminanceMeasurement, TuyaLocalCluster):
    """Tuya local IlluminanceMeasurement cluster."""

class TuyaTemperatureMeasurement(TemperatureMeasurement, TuyaLocalCluster):
    """Tuya local TemperatureMeasurement cluster."""

class TuyaRelativeHumidity(RelativeHumidity, TuyaLocalCluster):
    """Tuya local RelativeHumidity cluster."""

class NeoBatteryLevel(t.enum8):
    """NEO battery level enum."""

[... 168 lines cut ...]
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.OCCUPANCY_SENSOR,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    MmwRadarManufCluster,
                    TuyaOccupancySensing,
                    TuyaAnalogInput,
                    TuyaIlluminanceMeasurement,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }
    }`

I tried to modify this [file](https://github.com/zigpy/zha-device-handlers/blob/a221e682e9baffe95588eb6708d096cfc481cd5d/zhaquirks/tuya/ts0601_motion.py#L352) but the code is not working. The sensor doesn't use the quirk! Mhh! Where is there error?

programmieramt, 2023-05-16:
> 

Did you just added the manufacturer Id? _TZE204_sxm7l9xa I think we have to do a new matching for the datapoints in the new quirk

hiscorebob, 2023-05-17:
> I tried to modify this [file](https://github.com/zigpy/zha-device-handlers/blob/a221e682e9baffe95588eb6708d096cfc481cd5d/zhaquirks/tuya/ts0601_motion.py#L352) but the code is not working. The sensor doesn't use the quirk! Mhh! Where is there error?

Never used a quirk before, this is also new to me.
I tried using your quirk, and noticed the Logs get flooded with "unknown cluster command" messages.

dieneuser, 2023-05-20:
> > 
> 
> Did you just added the manufacturer Id? _TZE204_sxm7l9xa I think we have to do a new matching for the datapoints in the new quirk

I can confirm the behavior. The [Quirk](https://github.com/zigpy/zha-device-handlers/issues/2378#issuecomment-1548703481) does not seem to be used:
[image]
[zha-796e67a077f16237422834223f8280f9-_TZE204_sxm7l9xa TS0601-b56351faae6a353ffd0cbb4ae7e1bb53.json.txt](https://github.com/zigpy/zha-device-handlers/files/11522241/zha-796e67a077f16237422834223f8280f9-_TZE204_sxm7l9xa.TS0601-b56351faae6a353ffd0cbb4ae7e1bb53.json.txt)

programmieramt, 2023-05-21:
I also tried several quirks and stuff but the device doesn’t use them

rofo69, 2023-05-22:
Really not sure which of the many threads about this device to post on, but I've managed to get the ZHA quirk working for me with the basic occupancy sensor and illuminance using this quirk that I put together from various others floating around:-

Its based on the _TZE204_sxm7l9xa variant.

If anyone can extend it to expose the configuration options that would be amazing.

`"""ZY-M100 Human Presence Sensor"""
#https://github.com/zigpy/zha-device-handlers/issues/2378

#based on
#"""BlitzWolf IS-3/Tuya motion rechargeable occupancy sensor."""
#https://github.com/zigpy/zha-device-handlers/blob/a221e682e9baffe95588eb6708d096cfc481cd5d/zhaquirks/tuya/ts0601_motion.py#L352

import math
from typing import Dict, Optional, Tuple, Union

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl import foundation
from zigpy.zcl.clusters.general import (
    AnalogInput,
    Basic,
    GreenPowerProxy,
    Groups,
    Identify,
    Ota,
    Scenes,
    Time,
)
from zigpy.zcl.clusters.measurement import (
    IlluminanceMeasurement,
    OccupancySensing,
    RelativeHumidity,
    TemperatureMeasurement,
)
from zigpy.zcl.clusters.security import IasZone

from zhaquirks import Bus, LocalDataCluster, MotionOnEvent
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    MOTION_EVENT,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    DPToAttributeMapping,
    TuyaLocalCluster,
    TuyaManufCluster,
    TuyaNewManufCluster,
)
from zhaquirks.tuya.mcu import TuyaMCUCluster

ZONE_TYPE = 0x0001

class TuyaOccupancySensing(OccupancySensing, TuyaLocalCluster):
    """Tuya local OccupancySensing cluster."""

class TuyaAnalogInput(AnalogInput, TuyaLocalCluster):
    """Tuya local AnalogInput cluster."""

class TuyaIlluminanceMeasurement(IlluminanceMeasurement, TuyaLocalCluster):
    """Tuya local IlluminanceMeasurement cluster."""

class TuyaTemperatureMeasurement(TemperatureMeasurement, TuyaLocalCluster):
    """Tuya local TemperatureMeasurement cluster."""

[... 172 lines cut ...]

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.OCCUPANCY_SENSOR,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    MmwRadarManufCluster,
                    TuyaOccupancySensing,
                    TuyaAnalogInput,
                    TuyaIlluminanceMeasurement,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }
    }`

NCC1941, 2023-05-23:
> Really not sure which of the many threads about this device to post on, but I've managed to get the ZHA quirk working for me with the basic occupancy sensor and illuminance using this quirk that I put together from various others floating around:-
> 
> Its based on the _TZE204_sxm7l9xa variant.
> 
> Its basically Wolf's altered quirk above but I've changed the device signature to work with the _TZE204_sxm7l9xa variant.
> 
> If anyone can extend it to expose the configuration options that would be amazing.
> 
> [ts0601_motion.zip](https://github.com/zigpy/zha-device-handlers/files/11537441/ts0601_motion.zip)

For whatever it's worth, the quirk you've provided didn't work with my _TZE204_sxm7l9xa. Similar result as above with the other quirk(s), my device just doesn't use the quirk.

In case it's relevant, I do also have a _TZE204_ztc6ggyl which I'm using with the other quirk that provides all of the configuration options. That one uses its quirk, and seems to work fine.

programmieramt, 2023-05-23:
> For whatever it's worth, the quirk you've provided didn't work with my _TZE204_sxm7l9xa. Similar result as above with the other quirk(s), my device just doesn't use the quirk.
>
> In case it's [...]

Same for me the quirk is not used

rofo69, 2023-05-23:
Can you post the device signature for the _TZE204_ztc6ggyl and your  _TZE204_sxm7l9xa and I'll try to see why it's not picking up.

programmieramt, 2023-05-23:
> Can you post the device signature for the _TZE204_ztc6ggyl and your _TZE204_sxm7l9xa and I'll try to see why it's not picking up.

I am not sure what is part of the signature:

`signature:
    node_descriptor: >-
      NodeDescriptor(logical_type=<LogicalType.Router: 1>,
      complex_descriptor_available=0, user_descriptor_available=0, reserved=0,
      aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>,
      mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice:
      142>, manufacturer_code=4417, maximum_buffer_size=66,
      maximum_incoming_transfer_size=66, server_mask=10752,
      maximum_outgoing_transfer_size=66,
      descriptor_capability_field=<DescriptorCapability.NONE: 0>,
      *allocate_address=True, *is_alternate_pan_coordinator=False,
      *is_coordinator=False, *is_end_device=False,
      *is_full_function_device=True, *is_mains_powered=True,
      *is_receiver_on_when_idle=True, *is_router=True,
      *is_security_capable=False)
    endpoints:
      '1':
        profile_id: '0x0104'
        device_type: '0x0051'
        input_clusters:
          - '0x0000'
          - '0x0004'
          - '0x0005'
          - '0xef00'
        output_clusters:
          - '0x000a'
          - '0x0019'
      '242':
        profile_id: '0xa1e0'
        device_type: '0x0061'
        input_clusters: []
        output_clusters:
          - '0x0021'
    manufacturer: _TZE204_sxm7l9xa
    model: TS0601`
    
    Is this enough?

rofo69, 2023-05-23:
Go to the device in zha click the the dots next to the reconfigure button. Then choose manage zigbee device then click signature and paste those for both your devices.

I have the _TZE204_sxm7l9xa working again. I must have broken something in the file I posted. I will upload in about an hour or so. I am away from my pc atm.

rofo69, 2023-05-23:
Can you also post the sig for _TZE204_ztc6ggyl

programmieramt, 2023-05-23:
> Can you also post the sig for _TZE204_ztc6ggyl

Here. This device works and also exposes the config values to the fronted
```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0107",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x000c",
        "0x000d",
        "0x0400",
        "0x0406",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": "0x0104",
      "device_type": "0x0007",
      "input_clusters": [
        "0x000d"
      ],
      "output_clusters": []
    },
    "3": {
      "profile_id": "0x0104",
      "device_type": "0x0007",
      "input_clusters": [
        "0x000d"
      ],
      "output_clusters": []
    },
    "4": {
      "profile_id": "0x0104",
      "device_type": "0x0007",
      "input_clusters": [
        "0x000d"
      ],
      "output_clusters": []
    },
    "5": {
      "profile_id": "0x0104",
      "device_type": "0x0007",
      "input_clusters": [
        "0x000d"
      ],
      "output_clusters": []
    }
  },
  "manufacturer": "_TZE204_ztc6ggyl",
  "model": "TS0601",
  "class": "presence_new.TuyaMmwRadarOccupancy"
}
```

rofo69, 2023-05-23:
Could you unload your working quirk on _TZE204_ztc6ggyl then reload HA and show me what the sig looks like without the quirk loaded. Then you can put it back again.

rofo69, 2023-05-23:
I'll post my working basic quirk for _TZE204_sxm7l9xa in about 45 minutes when I am back.

programmieramt, 2023-05-23:
> Could you unload your working quirk on _TZE204_ztc6ggyl then reload HA and show me what the sig looks like without the quirk loaded. Then you can put it back again.

```
{
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0107",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x000c",
        "0x0400",
        "0x0406",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE204_ztc6ggyl",
  "model": "TS0601",
  "class": "zhaquirks.tuya.ts0601_motion.MmwRadarMotion"
}
```
This is the signature without the quirk. But without the quirk it does not exposed the config to HA

programmieramt, 2023-05-23:
> Thats fine, i just wanted to compare both against _TZE204_sxm7l9xa, as I'm no quirk expert.
> 
> Here is the correct quirk that gives at least some functionality. I've also corrected the original link above.
> 
> [ts0601_motion.zip](https://github.com/zigpy/zha-device-handlers/files/11540754/ts0601_motion.zip)
> 
> I'll see if i can figure out how to expose the config entries.

Great. Looks good. The occupancy sensor is now exposed.

```

import math
from typing import Dict, Optional, Tuple, Union

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl import foundation
from zigpy.zcl.clusters.general import (
    AnalogInput,
    AnalogOutput,
    Basic,
    GreenPowerProxy,
    Groups,
    Identify,
    Ota,
    Scenes,
    Time,
)
from zigpy.zcl.clusters.measurement import (
    IlluminanceMeasurement,
    OccupancySensing
)
from zigpy.zcl.clusters.security import IasZone

from zhaquirks import Bus, LocalDataCluster, MotionOnEvent
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODEL,
    MOTION_EVENT,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)

from zhaquirks.tuya import (
    NoManufacturerCluster,
    TuyaLocalCluster,
    TuyaNewManufCluster,
)
from zhaquirks.tuya.mcu import (
    # TuyaDPType,
    DPToAttributeMapping,
    TuyaAttributesCluster,
    TuyaMCUCluster,
)

class TuyaMmwRadarSelfTest(t.enum8):
    """Mmw radar self test values."""
    TESTING = 0
    TEST_SUCCESS = 1
    TEST_FAILURE = 2
    OTHER = 3
    COMM_FAULT = 4
    RADAR_FAULT = 5

class TuyaOccupancySensing(OccupancySensing, TuyaLocalCluster):
    """Tuya local OccupancySensing cluster."""
[... 249 lines cut ...]
                INPUT_CLUSTERS: [
                    TuyaMmwRadarMaxRange,
                ],
                OUTPUT_CLUSTERS: [],
            },
            4: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.COMBINED_INTERFACE,
                INPUT_CLUSTERS: [
                    TuyaMmwRadarDetectionDelay,
                ],
                OUTPUT_CLUSTERS: [],
            },
            5: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.COMBINED_INTERFACE,
                INPUT_CLUSTERS: [
                    TuyaMmwRadarFadingTime,
                ],
                OUTPUT_CLUSTERS: [],
            },
        }
    }
```
This is the quirk which exposes the config

rofo69, 2023-05-23:
Ok, through a fair bit of trial and error, I think I have a working quirk with config options.
I can't get detection distance as a sensor working yet, but I might have a go.

I think you might need that sensor working for the min and max range config options to work (I'm guessing).

[modded_ts0601_motion.zip](https://github.com/zigpy/zha-device-handlers/files/11543903/modded_ts0601_motion.zip)

programmieramt, 2023-05-23:
> Ok, through a fair bit of trial and error, I think I have a working quirk with config options. I can't get detection distance as a sensor working yet, but I might have a go.
>
> I think you might need [...]

For me looks good. I am not sure if the value are used but I can configure it

diogo-madeira, 2023-05-23:
> Ok, through a fair bit of trial and error, I think I have a working quirk with config options. I can't get detection distance as a sensor working yet, but I might have a go.
>
> I think you might need [...]

I tried this quirk.... I can see all options, the illuminance sensor works, but the Occupancy sensor is not working....

rofo69, 2023-05-23:
Looks like there are some more tweaks to do.
The detected distance appears to be coming in meters:-
[image]

But the min and max are in cms, so that needs to be standardised:

rofo69, 2023-05-23:
Ok, having tested this device at home, whilst the illuminance and motion appear to work, I am utterly unconvinced the configuration settings don't work at all. 
I'm going to have to do some more digging or someone else can pick up what I have and fix/improve it.

rofo69, 2023-05-23:
I'm leaving my last effort here with a few more corrections. Sadly I still can't get the configs when set in the UI, to apply to the device.
I think it might be to do with adding the updates to the UI bus, but I'm guessing at this point.

[final_try.zip](https://github.com/zigpy/zha-device-handlers/files/11547413/final_try.zip)

Woll3, 2023-05-23:
> For me looks good. I am not sure if the value are used but I can configure it

I tried the quirk but it doesn't work for me! The log file is attached!
The controls and sensors are unavailable.

rofo69, 2023-05-23:
Are you using the _TZE204_sxm7l9xa variant ?
Have you cleared your _pycache_ folder ?

As I say... its 'works', but the config entities dont seem to set on the device, but the motion and illuminance sensors do work.

hiscorebob, 2023-05-23:
> I'm leaving my last effort here with a few more corrections. Sadly I still can't get the configs when set in the UI, to apply to the device. I think it might be to do with adding the updates to the [...]

Thank you! Presence Sensors are working now!

backcountrymountains, 2023-06-08:
Thanks for this! It got my _TZE204_sxm7l9xa working!

I can get it to work for Occupancy but I don't get any sensors for Target Distance or Illumination:
[image]

What is the magic incantation to get ZHA to add those sensors?
I can manually read the ```TuyaIlluminationMeasurement:measured_value``` and ```TuyaMmwRadarTargetDistance:present_value``` attributes using the ```Manage Zigbee Device``` menu, but the entities aren't created by ZHA in HA for me to look at.

What do I do?!?!?!

Thanks!

HA 2023.6.0
HUSUZB-1
Quirk: tuya.tuya_sxm_motion.MmwRadarMotionZYM100 (I renamed it, but it's final_try.py)

Woll3, 2023-06-08:
> Are you using the _TZE204_sxm7l9xa variant ? Have you cleared your _pycache_ folder ?
> 
> As I say... its 'works', but the config entities dont seem to set on the device, but the motion and illuminance sensors do work.

[image]

How do I clear the _pycache_ folder?
Thanks! 😊

wishie, 2023-06-17:
Occupancy and Illuminance both appear. Illuminance work fine, but my occupancy entity seems to never change back from "Detected" once its been triggered.

Zoriontsu, 2023-06-19:
> Occupancy and Illuminance both appear. Illuminance work fine, but my occupancy entity seems to never change back from "Detected" once its been triggered.

I am having the same issue. After playing around with multiple quirk iterations, I am stuck here

rofo69, 2023-06-21:
I only use the lux setting when its very low, to turn on a light in a room. For this use case, the sensor seems to work correctly.

As for putting the motion sensor outside, the mmwave type sensors are so sensitive, they will trigger on the slightest movement, so if there are any trees, leaves etc moving, its likely to trigger them. They really aren't designed for outdoor use. Perhaps a camera or PIR sensor would be a better candidate for you, depending on what you're trying to do.

As for the sensitivity setting, As I mentioned earlier in the thread, I'm not convinced that any of the settings are working correctly. I basically put together the quirk above by pasting together some others. Its now beyond my expertise to figure out what needs to be done.

Although I suspect, if any of the settings work, it is infact sensitivity, but I can't be certain of that.

Njamotep, 2023-06-23:
@rofo69 Thank you so much for the work you put into this. One question, and it's not really that important: I see in your screenshots that you have illuminance sensor working on  _TZE204_sxm7l9xa, but mine doesn't seem to work. What could be wrong? Have cleared pycache and installed from scratch several times. TIA!

Njamotep, 2023-06-23:
Nvm, like the 20th time I removed and re-paired the device it suddenly showed up! Thanks again!!

Riddlr, 2023-06-23:
Hi All, 
Here is what I use, seems to work okay.
`import math                                                                            
from typing import Dict, Optional, Tuple, Union                                        
                                                                                       
from zigpy.profiles import zha                                                         
from zigpy.quirks import CustomDevice                                                  
import zigpy.types as t                                                                
from zigpy.zcl import foundation                                                       
from zigpy.zcl.clusters.general import (                                               
    AnalogInput,                                                                       
    AnalogOutput,                                                                      
    Basic,                                                                             
    GreenPowerProxy,                                                                   
    Groups,                                                                            
    Identify,                                                                          
    Ota,                                                                               
    Scenes,                                                                            
    Time,                                                                              
)                                                                                      
from zigpy.zcl.clusters.measurement import (                                           
    IlluminanceMeasurement,                                                            
    OccupancySensing                                                                   
)                                                                                      
from zigpy.zcl.clusters.security import IasZone                                        
                                                                                       
from zhaquirks import Bus, LocalDataCluster, MotionOnEvent                             
from zhaquirks.const import (                                                          
    DEVICE_TYPE,                                                                       
    ENDPOINTS,                                                                         
    INPUT_CLUSTERS,                                                                    
    MODEL,                                                                             
    MOTION_EVENT,                                                                      
    OUTPUT_CLUSTERS,                                                                   
    PROFILE_ID,                                                                        
)                                                                                      
                                                                                       
from zhaquirks.tuya import (                                                           
    NoManufacturerCluster,                                                             
    TuyaLocalCluster,                                                                  
    TuyaNewManufCluster,                                                               
)                                                                                      
from zhaquirks.tuya.mcu import (                                                       
    # TuyaDPType,                                                                      
    DPToAttributeMapping,                                                              
    TuyaAttributesCluster,                                                             
    TuyaMCUCluster,                                                                    
)                                                                                      
                                                                                       
                                                                                       
class TuyaMmwRadarSelfTest(t.enum8):                                                   
    """Mmw radar self test values."""                                                  
    TESTING = 0                                                                        
    TEST_SUCCESS = 1                                                                   
    TEST_FAILURE = 2                                                                   
    OTHER = 3                                                                          
    COMM_FAULT = 4                                                                     
    RADAR_FAULT = 5                                                                    
                                                                                       
class TuyaOccupancySensing(OccupancySensing, TuyaLocalCluster):                        
    """Tuya local OccupancySensing cluster."""                                         
                                                                                       
class TuyaIlluminanceMeasurement(IlluminanceMeasurement, TuyaLocalCluster):            
[... cut ...]
                OUTPUT_CLUSTERS: [],                                                   
            },                
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },                                                         
        }                                                                              
    }                                                                                  `

rofo69, 2023-06-24:
I think the configs might be working with the file @Riddlr has posted. I've not tested them all so far, but I think min and max range do work.

I've noticed that the file seems to have alot of weird space characters in the blank lines, and at the end of the lines. I don't think it affects the quirk at all, but I purged them using this tool:

https://www.gillmeister-software.com/online-tools/text/trim_lines-remove_spaces.aspx

Njamotep, 2023-06-24:
Fantastic! Works great for my usecase with the improved quirk! Thanks @rofo69 and @Riddlr !

Goodie01, 2023-07-01:
Without wanting to be impatient... but being a little impatient; how far away is this from release and/or how would one go about adding it manually?

rofo69, 2023-07-01:
Google the forums for how to add a local zha quirk then use the last file in the thread above, then you can use it immediately.

evgenln, 2023-07-02:
@rofo69 work great for me, what about PR it?

rrooggiieerr, 2023-07-03:
Hi! I just received my ZY_M100 today and trying out the custom quirk. I've put `ts0601_motion.py` in `config/custom_zha_quirks/` on my HA, but it's only showing LQI and RSSI diagnostic sensors. Am I missing something?

johnwarne, 2023-07-03:
@rofo69 @Riddlr Works great on my end as well!

rrooggiieerr, 2023-07-04:
Hi! I received my ZY-M100 yesterday and got it working almost immediately, using the `ts0601_motion.py` by @rofo69  thanks for the good work.

Just some remarks from my side:
- I think the device is called a presence sensor, not an occupancy sensor. ZHA doesn't seem to have a PresenceSensing cluster, so the OccupancySensing is probably the right choice. Looking at the ZHA quirk for the Aqara FP1 the terp presence is also used while the OccupancySensing cluster is used for the implementation.
- Name of the Controls include underscores, e.g. `_TZE204_sxm7l9xa TS0601 Number detection_delay`. Also the Number part in the name could be gone I guess, its a but unnecessary. I can't really figure out where the name of the entity is set in the quirk.
- Icon is now a `mdi:home`, I think `mdi:motion-sensor` is more appropriate
- File uses dos/windows format line endings, I'm quite sure ZHA uses Unix line endings in it's coding standards.
- `ts0601_motion.py` is already an existing quirk in the `zha-device-handlers` package. Maybe rename it to `ts0601_presence.py` instead?

rofo69, 2023-07-04:
Thanks for the feedback. I'm in no way a ZHA dev and took on the original challenge by pasting together other quirks I'd found. Then @Riddlr provided the fundamentally working quirk that I tweaked in a minor way. If you want to take up the challenge and edit the remaining items you've highlighted, then please feel free to do so.

arbrandes, 2023-07-21:
I rounded up the changes proposed and ported them over to the (hopefully) proper location via https://github.com/zigpy/zha-device-handlers/pull/2485.

koenvangils, 2023-07-24:
> I rounded up the changes proposed and ported them over to the (hopefully) proper location via #2485.

I just tried this file and want to let you know that it also seems to work with the (new?) model _TZE204_qasjif9e after I manually edited it in your file.

arbrandes, 2023-07-25:
@koenvangils, if you add a comment to the proper line in the PR (for instance, [here](https://github.com/zigpy/zha-device-handlers/pull/2485/files#diff-54518cc8f6fb8f83044356d5403305d4a39716c0bccf89f79b4c51df1ab98a4cR568)), I can add that one in as well.

jbustdel, 2023-08-03:
I'm.sorry but for the new model model _TZE204_qasjif9e doesn't work. I have made the modifications and the sensors do not appear.

IlyaShcherbakov-TomTom, 2023-08-04:
I have the same as @jbustdel

backcountrymountains, 2023-08-04:
I have a ```_TZE204_sxm7l9xa``` that is mostly working with this quirk. 
I went through a bunch of these quirk files and sometimes the Illuminance measurement worked on pairing and sometimes it didn't. I don't really understand what makes the device or quirck finicky about exposing/creating the illuminance entity. 

Is it possible for people to list their device and what is working?

_TZE204_sxm7l9xa | Working | Working but </br> no HA entity | Not Working
-- | -- | -- | --
Controls
Detection Delay | ✅ |   |  
Fading Time | ✅  |   |  
Max Range | ✅  |   |  
Min Range | ✅  |   |  
Sensitivity | ✅  |   |  
Sensors
Illuminance | ✅  |   |  
Occupancy | ✅ |   |  
Motion |   |   | ❌ 
Target Distance |   | ❌  |  

For ```Target Distance``` I can query ```TuyaMmwRadarTargetDistance: present_value``` in ```Manage Zigbee Device```, and it returns a reasonable looking number, but there is no sensor in HA.

Also, one of these quirks seemed to have some logic to trigger a motion sensor entity, not just an occupancy entity:
```
if hdr.command_id == 0x0001 and tuya_cmd.command_id == 1027:
            self.endpoint.device.motion_bus.listener_event(MOTION_EVENT)
```
As part of this code:
```
class MotionCluster(LocalDataCluster, MotionOnEvent):
    """Tuya Motion Sensor."""

    _CONSTANT_ATTRIBUTES = {ZONE_TYPE: IasZone.ZoneType.Motion_Sensor}
    reset_s = 15

class TuyaManufacturerClusterMotion(TuyaManufCluster):
    """Manufacturer Specific Cluster of the Motion device."""

    def handle_cluster_request(
        self,
        hdr: foundation.ZCLHeader,
        args: Tuple[TuyaManufCluster.Command],
        *,
        dst_addressing: Optional[
            Union[t.Addressing.Group, t.Addressing.IEEE, t.Addressing.NWK]
        ] = None,
    ) -> None:
        """Handle cluster request."""
        tuya_cmd = args[0]
        self.debug("handle_cluster_request--> hdr: %s, args: %s", hdr, args)
        if hdr.command_id == 0x0001 and tuya_cmd.command_id == 1027:
            self.endpoint.device.motion_bus.listener_event(MOTION_EVENT)
```
Does anybody understand how that works? I would like to have a motion sensor entity in addition to the occupancy sensor, if possible.

Thanks to everyone who has helped make this quirk work!

evgenln, 2023-08-04:
You need to restart HA to get illuminance, it's limitations of this code. If it doesn't work try restart HA host, with HA option restart is not always successful, e.g. to missed Bluetooth device

backcountrymountains, 2023-08-04:
@evgenln Does a target distance measurement sensor show up for you?

egelinotte, 2023-08-04:
Nope

evgenln, 2023-08-04:
> @evgenln Does a target distance measurement sensor show up for you?

No, there is a duplication that seems to block it. It only works via query

evgenln, 2023-08-04:
Also, to get it as a sensor it should be presented as a sensor in HA, it cannot be done only with quirk. E.g. see attempt here https://github.com/zigpy/zha-device-handlers/pull/2275

backcountrymountains, 2023-08-04:
Hmm. Is there not a way to have zha/zhaquirk create an arbitrary sensor in HA? I don't care if it shows up as watts, Celsius, MJ/nS or whatever I just want a sensor entity so I can make a template sensor from it.

jbustdel, 2023-08-05:
For me and my _TZE204_qasjif9e the results are:

@rofo69 py shows all attributes but occupancy sensor always not detected.
@arbrandes  py dont show any atributes
The 24 GHZ py for Smart Plug sensor I posted up, occupancy sensor works ok, but not light sensor and regulations are not possible.

I dont understand nothing.

[126 more replies]

PR #3736 (2025-01-20): Add Tuya motion `_TZE204_sxm7l9xa`, `_TZE204_e5m9c5hl`
## Proposed change

Adds:
- _TZE204_sxm7l9xa
- _TZE204_e5m9c5hl

Closes: #2485
Closes: #2378
Closes: #2546

## Additional information

Based on: https://github.com/Koenkk/zigbee-herdsman-converters/blob/f42aa5080ee933e8cde9fc08c67a88bf886d71ff/src/devices/tuya.ts#L7959C71-L7959C87

## Checklist

- [ ] The changes are tested and work correctly
- [x] `pre-commit` checks pass / the code has been formatted using Black
- [x] Tests have been added to verify that the new code works
closed 2025-01-20 by TheJulianJES