zigpy/zha-device-handlers #2383: [BUG] 2 Gang Tuya Light Switch - Toggle Switches Both Lights

JIBYC, 2023-05-06:
### Bug description

Bug is as per issue [zigpy/zha-device-handlers] ts0012 - new firmware (Issue #1744)

### Steps to reproduce

1. Add 2 gang Tuya switch to ZHA
2. Issue command from HA via ZHA to toggle one of the two switches
3. ZHA will toggle both switches

### Expected behavior

Toggling a single switch in ZHA should toggle the single corrisonding switch only. 

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006"
      ],
      "out_clusters": [
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0004",
        "0x0005",
        "0x0006"
      ],
      "out_clusters": []
    }
  },
  "manufacturer": "_TZ3000_18ejxno0",
  "model": "TS0012",
  "class": "zhaquirks.tuya.ts001x.TuyaDoubleNoNeutralSwitch_2"
}

```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "ieee": "**REDACTED**",
  "nwk": 60050,
  "manufacturer": "_TZ3000_18ejxno0",
  "model": "TS0012",
  "name": "_TZ3000_18ejxno0 TS0012",
  "quirk_applied": true,
  "quirk_class": "zhaquirks.tuya.ts001x.TuyaDoubleNoNeutralSwitch_2",
  "manufacturer_code": 4417,
  "power_source": "Battery or Unknown",
  "lqi": 36,
  "rssi": null,
  "last_seen": "2023-05-06T16:22:15",
  "available": true,
  "device_type": "EndDevice",
  "signature": {
    "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
    "endpoints": {
      "1": {
        "profile_id": 260,
        "device_type": "0x0100",
        "in_clusters": [
          "0x0000",
          "0x0003",
          "0x0004",
          "0x0005",
          "0x0006"
        ],
        "out_clusters": [
          "0x0019"
        ]
      },
      "2": {
        "profile_id": 260,
        "device_type": "0x0100",
        "in_clusters": [
          "0x0004",
          "0x0005",
          "0x0006"
        ],
        "out_clusters": []
      }
    }
  },
  "active_coordinator": false,
  "entities": [
    {
      "entity_id": "button.tz3000_18ejxno0_ts0012_identify",
      "name": "_TZ3000_18ejxno0 TS0012"
    },
    {
      "entity_id": "light.tz3000_18ejxno0_ts0012_light",
      "name": "_TZ3000_18ejxno0 TS0012"
    },
    {
      "entity_id": "light.tz3000_18ejxno0_ts0012_light_2",
      "name": "_TZ3000_18ejxno0 TS0012"
    }
  ],
  "neighbors": [],
  "routes": [],
  "endpoint_names": [
    {
      "name": "ON_OFF_LIGHT"
    },
    {
      "name": "ON_OFF_LIGHT"
    }
  ],
  "user_given_name": null,
  "device_reg_id": "cf35eb0d04f1bbf928d5e240c01aa321",
  "area_id": null,
  "cluster_details": {
    "1": {
      "device_type": {
        "name": "ON_OFF_LIGHT",
        "id": 256
      },
      "profile_id": 260,
      "in_clusters": {
        "0x0000": {
          "endpoint_attribute": "basic",
          "attributes": {
            "0x0004": {
              "attribute_name": "manufacturer",
              "value": "_TZ3000_18ejxno0"
            },
            "0x0005": {
              "attribute_name": "model",
              "value": "TS0012"
            }
          },
          "unsupported_attributes": {}
        },
        "0x0003": {
          "endpoint_attribute": "identify",
          "attributes": {},
          "unsupported_attributes": {}
        },
        "0x0004": {
          "endpoint_attribute": "groups",
          "attributes": {},
          "unsupported_attributes": {}
        },
        "0x0005": {
          "endpoint_attribute": "scenes",
          "attributes": {},
          "unsupported_attributes": {}
        },
        "0x0006": {
          "endpoint_attribute": "on_off",
          "attributes": {
            "0x0000": {
              "attribute_name": "on_off",
              "value": 1
            },
            "0x4001": {
              "attribute_name": "on_time",
              "value": 0
            },
            "0x4002": {
              "attribute_name": "off_wait_time",
              "value": 0
            },
            "0x8001": {
              "attribute_name": "backlight_mode",
              "value": 1
            },
            "0x8002": {
              "attribute_name": "power_on_state",
              "value": 0
            }
          },
          "unsupported_attributes": {}
        }
      },
      "out_clusters": {
        "0x0019": {
          "endpoint_attribute": "ota",
          "attributes": {},
          "unsupported_attributes": {}
        }
      }
    },
    "2": {
      "device_type": {
        "name": "ON_OFF_LIGHT",
        "id": 256
      },
      "profile_id": 260,
      "in_clusters": {
        "0x0004": {
          "endpoint_attribute": "groups",
          "attributes": {},
          "unsupported_attributes": {}
        },
        "0x0005": {
          "endpoint_attribute": "scenes",
          "attributes": {},
          "unsupported_attributes": {}
        },
        "0x0006": {
          "endpoint_attribute": "on_off",
          "attributes": {
            "0x0000": {
              "attribute_name": "on_off",
              "value": 1
            },
            "0x4001": {
              "attribute_name": "on_time",
              "value": 0
            },
            "0x4002": {
              "attribute_name": "off_wait_time",
              "value": 0
            }
          },
          "unsupported_attributes": {}
        }
      },
      "out_clusters": {}
    }
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

RED4911, 2023-05-07:
I have the same problem. Zigbee2MQTT solved the problem on their own. They haven't fixed it on ZHA yet.  https://github.com/Koenkk/zigbee2mqtt/issues/13656

NicolasIonut, 2023-05-11:
I have the same problem with some moes 2 gang switches. pls help

ahydrax, 2023-05-23:
Looks like I've got the same device as issue author.
```
 "signature": {
      "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
      "endpoints": {
        "1": {
          "profile_id": "0x0104",
          "device_type": "0x0100",
          "input_clusters": [
            "0x0000",
            "0x0003",
            "0x0004",
            "0x0005",
            "0x0006"
          ],
          "output_clusters": [
            "0x0019"
          ]
        },
        "2": {
          "profile_id": "0x0104",
          "device_type": "0x0100",
          "input_clusters": [
            "0x0004",
            "0x0005",
            "0x0006"
          ],
          "output_clusters": []
        }
      },
      "manufacturer": "_TZ3000_18ejxno0",
      "model": "TS0012"
    },
```
Already tried a bunch of quirks from the internet

MattWestb, 2023-05-23:
I was adding the tuya magic spell to the ZHA quirks in January and if using then it shall working OK but need adding the device new in ZHA.
If using custom quirk it must having the spell casting i it or its not working.
Also need deleting and waiting one minute and adding the device new or the device is not getting the spell and not working OK.
https://github.com/zigpy/zha-device-handlers/pull/2049

ahydrax, 2023-05-23:
@NicolasIonut thanks for a quick reply! Probably I know why we have to do this manually:
However `zhaquirks.tuya.ts001x.TuyaDoubleNoNeutralSwitch_2` already derives from `EnchantedDevice` https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts001x.py#L285 the spell will not be invoked as device uses `Identify.cluster_id` https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/__init__.py#L538

MattWestb, 2023-05-23:
Hola @javicalle Have you time taking one fast look how to fixing this device ??

Thank in advance !
MW

JIBYC, 2023-08-06:
Hi

Any joy on this one? It's the most popular ZigBee switch on ebay.

javicalle, 2023-08-07:
There are quite different devices signatures and not all match the `TS0012` quirk.
In the current version there are 5 quirks for the `TS0012` devices, all of them of type `EnchantedDevice`.

If your device is loading any quirk, you can try to fix it with the 'reconfigure' option from the device view.
If your device is not loading any quirk, open a new issue with the device signature or attach it here.

lightheaded, 2023-08-20:
Any hopes of this being fixed soon? -i am switching to Home Assistant Yellow and have to migrate my entire Zigbee network. Thought I'd take the plunge and switch to ZHA while I'm at it, but this is a dealbreaker for me.. 

Also makes me wonder if there are other traps like this with any of my other 61 devices 😅

TheJulianJES, 2023-08-20:
> Probably I know why we have to do this manually: However `zhaquirks.tuya.ts001x.TuyaDoubleNoNeutralSwitch_2` already derives from `EnchantedDevice` [...]

The test checks that every quirk merged into this repo would successfully cast a spell when used with Home Assistant.
However, the quirks in `ts001x.py` had `SKIP_CONFIGURATION` set which would prevent ZHA to bind any cluster (thus also not "casting the spell").
This was fixed with https://github.com/zigpy/zha-device-handlers/pull/2524 (and tests now check for that)
So for those devices (including the original post/issue), a fix should (hopefully) be in Home Assistant Core 2023.9.0

TheJulianJES, 2023-08-20:
Everyone who has the issue and does NOT have `"class": "zhaquirks.tuya.ts001x.` in the quirk (so either no `"class"` section or just `zigpy.device.Device` showing for a device which needs the spell), please open a new issue and fill-out the issue template.

dr3amr, 2023-09-05:
Works fine for me with both devices with 2gang TS0012 and 3gang TS0013. I had 1x2gang and 1x3gang (newer revisions) that when switched from ZHA it would switch on all lights of the same switch as mentioned here (working fine when triggered by touch). The older ones TS0011/TS0012/TS0013 all worked fine. Tried with the modified quirk as bellow, and now older ones and also new ones work as expected 👍
https://raw.githubusercontent.com/zigpy/zha-device-handlers/6a2693f9fd179124cf09d2f204c9b714ecc077f4/zhaquirks/tuya/ts001x.py

Frog90, 2023-10-10:
Yep finally the "new" Moes   2 Gang TS0012 works as it should  -  running HA 2023.10.1 
"older" TS0012 worked fine before - as others have said.  They still work now on 2023.10.1

Thank you soooo much

closed 2024-04-14 by github-actions[bot]