zigpy/zha-device-handlers #2390: Lutron aurora dimmer switch(Z3-1BRL) support request

Plausible0Bug, 2023-05-12:
### Problem description

I'm suffering from unreliable button press on Lutron aurora dimmer switch(Z3-1BRL).
The knob works properly.
My hardware is Home assistant yellow: [Silicon Labs MGM210P Mighty Gecko Module](https://www.mouser.com/new/silicon-labs/silicon-labs-mgm210p-mighty-gecko-lighting-module/) with Zigbee 3.0 firmware (EZSP).
I'm able to capture debug logs, hopefully someone can help take a look or guide me through creating a quirk, thank you.

[Z3-1BR_logs.zip](https://github.com/zigpy/zha-device-handlers/files/11426387/Z3-1BR_logs.zip)

Reference to exiting discussions:
https://github.com/zigpy/zha-device-handlers/issues/98
https://community.home-assistant.io/t/nortek-stick-conbee-2-and-the-lutron-aurora-dimmer-switch-button-push-issue/246992

### Solution description

I'm not sure why the coordinator is unable to capture the zha event, I wonder if a change in device handlers will change this.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4420, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0820",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x1000",
        "0xfc00"
      ],
      "output_clusters": [
        "0x0003",
        "0x0004",
        "0x0006",
        "0x0008",
        "0x0019",
        "0x1000"
      ]
    }
  },
  "manufacturer": "Lutron",
  "model": "Z3-1BRL",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

rpatterson, 2023-06-17:
I just received 2 of these and looking at the event logs through the web UI, it seems the ZHA Zigbee intgration is only emitting events for turning the dimmer wheel and I see *zero* events emitted when carefully pressing only the central button.  This can be tricky because it's easy to accidentally turn the wheel slightly when pressing the button.  I'm guessing that's what's behind the web search result I found where a user screenshots such an event and says it comes from pressing the button.

For reference, here'e the event payload for such an event, when I pressed the button sloppily and it turned the wheel slightly:

```
event_type: zha_event
data:
  device_ieee: ff:ff:00:0f:e7:fa:da:d4
  unique_id: ff:ff:00:0f:e7:fa:da:d4:1:0x0008
  device_id: 05aae6cf21415526c95e371b5dd91e8e
  endpoint_id: 1
  cluster_id: 8
  command: move_to_level_with_on_off
  args:
    - 255
    - 7
  params:
    level: 255
    transition_time: 7
origin: LOCAL
time_fired: "2023-06-17T21:35:58.746539+00:00"
context:
  id: 01H35MJK8T1KCF4XWKHSAN29DX
  parent_id: null
  user_id: null
```

I hope to get these working because I have toggle pole switches I can't replace and I like this device's approach to covering and locking the toggle pole.  It's elegant and has a far more minimal footprint and profile than other such devices.  Also, there are very few devices that cover toggle pole switches on the market these days and this one is the most readily available.

So I'm motivated to get these working and eager to help in any way I can.  LMK what you need!

tube0013, 2023-06-17:
The most recent firmware versions for the aurora's send the on off commands to a group. If you have an ezsp coordinator you won't see them unless you join the coordinator to the group. 

There are some existing quirks that set this up I think. But other non ezsp coordinators see the group commands without being a part of it.

rpatterson, 2023-06-19:
> ezsp coordinator

Some new terminology for me here.  [The ZHA docs](https://www.home-assistant.io/integrations/zha/) are a little ambiguous.  Sometimes it seems like a coordinator is a layer between the physical hardware, other times it seems to refer to the physical hardware itself.  Here's my device info from the web UI:

```
EZSP = Silicon Labs EmberZNet protocol: Elelabs, HUSBZB-1, Telegesis
by ZHA
Zigbee info
IEEE: ...
Nwk: 0x0000
Device Type: Coordinator
LQI: 255
RSSI: Unknown
Last Seen: 2023-06-19T13:02:03
Power Source: Mains
```

So I assume that means I do have an "ezsp coordinator".

The parts of the ZHA docs dealing with groups seem to be written from the perspective of someone familiar with the technical internals of Zigbee.  I poked around in the web UI and found the "BINDINGS" tab of the "Manage Zigbee Device" modal, but I'm unclear what to do there.  Are there any more end-user oriented docs on Zigbee grouping somewhere?

Plausible0Bug, 2023-06-19:
I noticed that turning on with center button never works for me, however if I use the dial to turn on the lights I can later use the button to turn lights off.

Jay2645, 2023-10-08:
Has anyone figured this out? I'm having the same issue. I've noticed it being a problem ever since I installed the SkyConnect; I simply don't get the button push notification.

I'm not sure what group I should be adding things to? The docs are very unclear, as the above poster mentioned.

rpatterson, 2023-10-08:
> Has anyone figured this out?

In case it helps, [here's how I'm using this button/switch](https://community.home-assistant.io/t/nortek-stick-conbee-2-and-the-lutron-aurora-dimmer-switch-button-push-issue/246992/14).

closed 2024-04-13 by github-actions[bot]

BRCook, 2025-06-11:
This is still an issue. Using an EZSP coordinator, the center button on the Aurora fails to work except in a few rare cases. Surely there is a workaround?