zigpy/zha-device-handlers #2396: [BUG] GL-C-007P `start_up_on_off` attribute doesn't work

niorko, 2023-05-14:
### Bug description

GL-C-007P does not respect `start_up_on_off` attribute.

### Steps to reproduce

- Change the behavior of Start-up Behavior to Off.
- Turn off the light in the HA.
- Unplug it from the electricity an wait 10 seconds
- Plug it in again

### Expected behavior

The light should be off. (The light is on, but HA is reporting off)

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "11": {
      "profile_id": "0x0104",
      "device_type": "0x010d",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0x1000"
      ],
      "output_clusters": [
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "GLEDPOPTO",
  "model": "GL-C-007P",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.5.2",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.11",
[... 879 lines cut ...]
      "242": {
        "device_type": {
          "name": "unknown",
          "id": 97
        },
        "profile_id": 41440,
        "in_clusters": {},
        "out_clusters": {
          "0x0021": {
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

### Additional information

I've tried to change in OnOff (Endpoint id: 11, Id: 0x0006, Type: in) (start_up_on_off (id: 0x4003)), but it does not work either. The lights are always on. I have multiple of these GL-C-007P and their work the same way.

TheJulianJES, 2023-11-10:
Yeah, this is an issue in the device firmware then.

niorko, 2024-02-02:
Is there a way to update the firmware to the latest one via ZHA, and are they available? I am unable to find any.

closed 2024-08-07 by github-actions[bot]