zigpy/zha-device-handlers #2397: [Device Support Request] _TZ3000_gjrubzje

rea-ducks, 2023-05-14:
### Problem description

Got a Tuya Switch module like https://zigbee.blakadder.com/Tuya_WHD02.html
Home Assistant ZHA reports it as "TS0001 by _TZ3000_gjrubzje"
It looks like there's a quirk loaded for it as HA repots "Quirk: zhaquirks.tuya.ts000x.Switch_1G_Metering"

But it's seen as light while it's a switch. So it might apply the wrong quirk.
Also it reports power meassurement sensors I don't believe it really has. At least it's not advertised with it and the sensors also don't display any values.

### Solution description

Recognize the device with correct device class and don't report any sensors it doesn't have.

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0702",
        "0x0b04",
        "0xe000",
        "0xe001"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3000_gjrubzje",
  "model": "TS0001",
  "class": "zhaquirks.tuya.ts000x.Switch_1G_Metering"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.5.2",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.11",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Berlin",
    "os_name": "Linux",
    "os_version": "6.1.25",
    "supervisor": "2023.04.1",
    "host_os": "Home Assistant OS 10.1",
    "docker_version": "23.0.3",
[... 510 lines cut ...]
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

MattWestb, 2023-08-22:
This device is one normal simple switch and have no power measuring if looking on Z2M commits https://github.com/Koenkk/zigbee-herdsman-converters/commit/4e69fea28832205ddf5c5cd3d90f4545ff58f9e6#diff-7ae23d3c13c2d7f7e37106590684c791aba8093f8be813b7c74a4f1484017c3bL1961-L1972.
If having other information pleas posting it in the issue.

closed 2024-02-25 by github-actions[bot]

Zoriontsu, 2024-10-26:
> ### Problem description
> Got a Tuya Switch module like https://zigbee.blakadder.com/Tuya_WHD02.html Home Assistant ZHA reports it as "TS0001 by _TZ3000_gjrubzje" It looks like there's a quirk loaded for it as HA repots "Quirk: zhaquirks.tuya.ts000x.Switch_1G_Metering"
> 
> But it's seen as light while it's a switch. So it might apply the wrong quirk. Also it reports power meassurement sensors I don't believe it really has. At least it's not advertised with it and the sensors also don't display any values.
> 
> ### Solution description
> Recognize the device with correct device class and don't report any sensors it doesn't have.
> 
> ### Screenshots/Video
> _No response_
> 
> ### Device signature
> Device signature
> ### Diagnostic information
> Diagnostic information
> ### Logs
> Logs
> ### Custom quirk
> Custom quirk
> ### Additional information
> _No response_

Did you find a quirk to address this?

omniwired, 2025-04-23:
Sorry for the necromancy, but this is definitely still happening in 2025