zigpy/zha-device-handlers #2399: [Device Support Request] TS0601 by _TZE200_f1pvdgoh "Tuya 2 in 1 PIR Motion Sensor+Brightness"

lightning200mph, 2023-05-15:
### Problem description

Requesting new device support

### Solution description

New device support:
https://www.aliexpress.us/item/3256803705327106.html?spm=a2g0o.order_list.order_list_main.5.336b1802JEJNUs&gatewayAdapt=glo2usa

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_f1pvdgoh",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.5.3",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.11",
    "docker": true,
    "arch": "x86_64",
    "timezone": "America/New_York",
    "os_name": "Linux",
    "os_version": "6.1.25",
    "supervisor": "2023.04.1",
    "host_os": "Home Assistant OS 10.1",
    "docker_version": "23.0.3",
    "chassis": "vm",
[... 278 lines cut ...]
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

lightning200mph, 2023-06-02:
Hopefully this will help.
Found someone working on a zigbee2mqtt new device support request.
[https://github.com/Koenkk/zigbee2mqtt/issues/12883](url)

EmanueleBenedettini, 2023-06-15:
@lightning200mph have you managed to get it working? Seems like that link doesn't work

lightning200mph, 2023-06-15:
I have not been able to get anything working yet.
I've not learned enough yet to make something work.  

The link isn't working but the text is the correct address.  

https://github.com/Koenkk/zigbee2mqtt/issues/12883

EmanueleBenedettini, 2023-07-20:
i copy/pasted some parts from different sources and this came out. it compiles but it's not being linked to the device :|

Hope someone has more knowledge than me about this and spots what I'm missing
[TS0601_pir_lux.zip](https://github.com/zigpy/zha-device-handlers/files/12113547/TS0601_pir_lux.zip)

RalfCzekallaHA, 2023-11-11:
I would like to support this request. 
I tried several quirks and tips for quirks I could find but couldn't get these to be taken by ZHA:

TS0601
 _TZE200_f1pvdgoh   

Zigbee Info:
IEEE: a4:c1:38:12:11:41:52:08
Nwk: 0x420c
Device Type: EndDevice
LQI: 255
RSSI: -28
Last seen: 2023-11-11T16:48:42
Energy source: Battery or Unknown 

Any chance to get this released as a quirk which works?
Thanks a lot!

closed 2024-05-16 by github-actions[bot]

grgizem, 2025-01-16:
is this still not supported? anyone figured out this?