zigpy/zha-device-handlers #2402: [BUG]

CU-tony, 2023-05-17:
### Bug description

[MOES (powered by tuya) scene switch 20/22](https://www.amazon.com/MOES-Wireless-Require-Battery-Automation/dp/B0BJK9SYNH?th=1)
Paired with HA yellow currently on version 2023.5.2

The scene controller paired and worked great for a week and then died. After replacing the battery the device is unresponsive (have tried multiple batteries) and either has a light stuck on switch 1 or a faint glow on switch 2. No amount of button presses seem to change the state. 

### Steps to reproduce

The issue is not repeatable since the controller wont connect again. 

### Expected behavior

I would expect a battery powered scene controller to last more than a week

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0006",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0006"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": "0x0104",
      "device_type": "0x0006",
      "input_clusters": [
        "0x0001",
        "0x0006"
      ],
      "output_clusters": []
    },
    "3": { 
     "profile_id": "0x0104",
      "device_type": "0x0006",
      "input_clusters": [
        "0x0001",
        "0x0006"
      ],
      "output_clusters": []
    },
    "4": {
      "profile_id": "0x0104",
      "device_type": "0x0006",
      "input_clusters": [
        "0x0001",
        "0x0006"
[... 412 lines cut ...]
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0006": {
            "endpoint_attribute": "TS004X_cluster",
            "attributes": {},
            "unsupported_attributes": {}
          }
        },
        "out_clusters": {}
      }
    }
  }
}
</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

MattWestb, 2023-05-17:
Sound like corrupted network setting or is the remote being broken
Try resetting it and see if its lighting if working normal then getting it in pairing mode.
If it do adding it to ZHA and if not its being broken.

TheJulianJES, 2023-09-04:
Fixed by https://github.com/zigpy/zha-device-handlers/pull/2288
Please reset and re-pair your Tuya remote controller on the latest HA version.

closed 2023-09-04 by TheJulianJES