zigpy/zha-device-handlers #2405: [Device Support Request] HEIMAN DoorBell-EF-3.0

HydrelioxGitHub, 2023-05-19:
### Problem description

This is a Doorbell device and HA provide a mouvement binary sensor (which never trigger). It should be a button.

### Solution description

Delete the mouvement sensor and add a button instead.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4619, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0402",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0500",
        "0x0b05"
      ],
      "out_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "HEIMAN",
  "model": "DoorBell-EF-3.0",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.4.1",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.10",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Paris",
    "os_name": "Linux",
    "os_version": "6.1.25",
    "supervisor": "2023.04.1",
    "host_os": "Home Assistant OS 10.1",
    "docker_version": "23.0.3",
    "chassis": "embedded",
    "run_as_root": true
[... 381 lines cut ...]
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

HydrelioxGitHub, 2023-05-30:
Does I have to provide anything else ?

lel-amri, 2023-09-28:
It's not even that great of a paperweight.

Mine shows the following for the IAS in_cluster endpoint:

```json
{
  "0x0500": {
    "endpoint_attribute": "ias_zone",
    "attributes": {
      "0x0010": {
        "attribute_name": "cie_addr",
        "value": "00:12:4b:00:25:9b:ba:1a"
      },
      "0x0002": {
        "attribute_name": "zone_status",
        "value": 32772
      },
      "0x0001": {
        "attribute_name": "zone_type",
        "value": 32769
      }
    },
    "unsupported_attributes": {}
  }
}
```

I'm very new to home automation stuff, but I'll try to come up with something. If nobody beats me to it, that is :p

HydrelioxGitHub, 2024-01-24:
Any update on this ?

closed 2024-07-29 by github-actions[bot]

HydrelioxGitHub, 2024-09-18:
Can this issue be reopened as it has not been solved yet ?