zigpy/zha-device-handlers #2410: [Device Support Request] TS011F _TZ3000_cfnprab5 variant - SM-SO306M (UseeLink 4 AC with 2 USB, universal plugs)

fonix232, 2023-05-24:
### Problem description

While the general device is supported by plethora of quirks, this specific model I got off of Amazon does not seem to match any of them.

<details><summary>Quirk finding failure logs</summary>

```
2023-05-24 10:27:33.823 DEBUG (MainThread) [zigpy.quirks.registry] Checking quirks for _TZ3000_cfnprab5 TS011F (70:b3:d5:2b:60:01:68:64)
2023-05-24 10:27:33.823 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.tuya.ts011f_switch.Tuya_1G_Switch'>
2023-05-24 10:27:33.823 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {1} {1, 2, 3, 4, 5}
2023-05-24 10:27:33.823 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.tuya.ts011f_switch.Tuya_2G_Switch'>
2023-05-24 10:27:33.823 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {1, 2, 242} {1, 2, 3, 4, 5}
2023-05-24 10:27:33.823 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.lidl.ts011f_plug.Lidl_Plug_3AC_4USB'>
2023-05-24 10:27:33.823 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {1, 242} {1, 2, 3, 4, 5}
2023-05-24 10:27:33.823 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.tuya.ts011f_plug.Plug_2AC_var05'>
2023-05-24 10:27:33.823 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {1, 2, 242} {1, 2, 3, 4, 5}
2023-05-24 10:27:33.823 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.tuya.ts011f_plug.Plug_2AC_var03'>
2023-05-24 10:27:33.823 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {1, 2} {1, 2, 3, 4, 5}
2023-05-24 10:27:33.823 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.tuya.ts011f_plug.Plug_v2'>
2023-05-24 10:27:33.824 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {1} {1, 2, 3, 4, 5}
2023-05-24 10:27:33.824 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.tuya.ts011f_plug.Plug_4AC_2USB_Metering'>
2023-05-24 10:27:33.824 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {242, 1, 2, 3} {1, 2, 3, 4, 5}
2023-05-24 10:27:33.824 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.tuya.ts011f_plug.Plug_4AC_2USB_cfnprab5'>
2023-05-24 10:27:33.824 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {1, 2, 3, 4, 5, 242} {1, 2, 3, 4, 5}
2023-05-24 10:27:33.824 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.tuya.ts011f_plug.Plug_TZ3210_1AC'>
2023-05-24 10:27:33.824 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {1, 242} {1, 2, 3, 4, 5}
2023-05-24 10:27:33.824 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.tuya.ts011f_plug.Plug_TZ3210_2AC'>
2023-05-24 10:27:33.824 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {1, 2, 242} {1, 2, 3, 4, 5}
2023-05-24 10:27:33.824 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.tuya.ts011f_plug.Plug_4AC_2USB'>
2023-05-24 10:27:33.824 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {1, 2, 3, 4, 5, 242} {1, 2, 3, 4, 5}
2023-05-24 10:27:33.824 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.tuya.ts011f_plug.Plug_3AC_4USB'>
2023-05-24 10:27:33.824 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {242, 1, 2, 3} {1, 2, 3, 4, 5}
2023-05-24 10:27:33.824 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.tuya.ts011f_plug.Plug_1AC'>
2023-05-24 10:27:33.824 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {242, 11} {1, 2, 3, 4, 5}
2023-05-24 10:27:33.824 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.tuya.ts011f_plug.Plug'>
2023-05-24 10:27:33.824 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {1, 242} {1, 2, 3, 4, 5}
2023-05-24 10:27:33.824 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.xiaomi.aqara.opple_switch.XiaomiOpple2ButtonSwitchFace2'>
2023-05-24 10:27:33.825 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {1, 2, 242} {1, 2, 3, 4, 5}
2023-05-24 10:27:33.825 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.xiaomi.aqara.opple_switch.XiaomiOpple2ButtonSwitchFace1'>
2023-05-24 10:27:33.825 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {1, 2, 41, 42, 242, 51, 21, 31} {1, 2, 3, 4, 5}
2023-05-24 10:27:33.825 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
2023-05-24 10:27:33.825 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {232, 230} {1, 2, 3, 4, 5}
2023-05-24 10:27:33.825 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
2023-05-24 10:27:33.825 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {232, 230} {1, 2, 3, 4, 5}
2023-05-24 10:27:33.825 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.tuya.ts0201.MoesTemperatureHumidtySensorWithScreen'>
2023-05-24 10:27:33.825 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {1} {1, 2, 3, 4, 5}
2023-05-24 10:27:33.825 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
2023-05-24 10:27:33.826 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {1} {1, 2, 3, 4, 5}
2023-05-24 10:27:33.826 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
2023-05-24 10:27:33.826 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {1} {1, 2, 3, 4, 5}
2023-05-24 10:27:33.826 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
2023-05-24 10:27:33.826 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {1} {1, 2, 3, 4, 5}
2023-05-24 10:27:33.826 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
2023-05-24 10:27:33.826 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {11, 13} {1, 2, 3, 4, 5}
```
</details>

### Solution description

An additional profile that matches the clusters of this device so that it is recognised.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature
[... 766 lines cut ...]
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

fonix232, 2023-05-24:
Upon further inspection, I've realised that the unit I have is quite messed up - no OTA cluster endpoint, no way to update the firmware, and while the switches work, that's about it. Guess it's going back to Amazon.

smartmatic, 2023-06-07:
It seems i have also a variant of this power cord. 
One piece shows correct with switches for all plugs and USB plus the option to switch on / off background light.

[image]

But the other one shows only as light switches without further entities:

[image]

[zha-adfa12b853984fcc33aa8d149a7606a3-_TZ3000_cfnprab5 TS011F-c7c7ecd0bf2d9aebe71f84a7f0f4392d.json.txt](https://github.com/zigpy/zha-device-handlers/files/11673292/zha-adfa12b853984fcc33aa8d149a7606a3-_TZ3000_cfnprab5.TS011F-c7c7ecd0bf2d9aebe71f84a7f0f4392d.json.txt)

Is there a chance to update the quirk?

MattWestb, 2023-06-07:
@smartmatic The one that working have loading the quirk and sent the tuya magic so its OK.
The other one is the same model but somthing is not OK with the data the system have getting from it so the quirk is not being loaded for it.
Delete the device , wait one minute, adding it new and look if its loading the quirk.
If not working restart HA and loos if its being OK.

smartmatic, 2023-06-07:
> @smartmatic
> Delete the device , wait one minute, adding it new and look if its loading the quirk.
> If not working restart HA and loos if its being OK.

I‘ve executed the steps. Deleted the device, waited more than one minute and re-added it and restarted HA.
But it does not load the quirk.

MattWestb, 2023-06-07:
Look in the diagnostic if the endpoints and clusters config is different that is shall not being then both looks being the sane device name / type.
Sometime ZHA is failing matching the quirk the adding the device but is working after restart of HA but it shall working the same for both devices.

smartmatic, 2023-06-07:
I've tried it now multiple times but without success. The quirk ist not loaded.

tpickle-py, 2023-07-28:
i got the same device and zha isnt able to control devices. 

I can provide logs if needed. i have other zigbee routers , maybe a conflict

MaxMSchneider, 2023-10-02:
Jumping in here, since I just got the same one. 

ZHA loads the quirk ts011f_plug.Plug_4AC_2USB_cfnprab5 and everything works, but the device does not exit pairing mode, meaning it reconnects multiple times before disconnecting completely (when the device runs into a pairing timeout I suppose).

[image]

DavidGeorge528, 2023-11-15:
I just bought 2 of these and appear to be having the same problem. The strange thing is I already had one which (after some trial and error) I managed to get to pair with the quirk active. 

That one had a different issue when paired, instead of showing 5 lights (which the 2 I bought are showing), operating any of the switches would turn all switches on/off, which I believe [this issue addresses](https://github.com/zigpy/zha-device-handlers/issues/1677). But I managed to get the quirk working in the end.

These 2 are different, giving the same problems outlined in this thread, all 5 switches show as lights, and no quirk is added to the device during pairing. I've tried everything mentioned in this thread so far and nothing has worked on either device. Is there any other advice on how to pair them with the quirk active?

closed 2024-05-20 by github-actions[bot]

woomiez, 2024-07-16:
I just received one of these cfnprab5 power strips (4 AC, 2 USB, Euro variant), and the pairing issue still persists. The device pairs, is controllable for about half a minute, then restarts the pairing process. This repeats indefinitely. What to do to fix?