zigpy/zha-device-handlers #2415: [BUG] Moes 4-gang switch TS0044 slow response

KoenHaemels, 2023-05-29:
### Bug description

I’ve ordered a Zigbee TS0044 gang switch of Moes to test it as a replacement of my switches.
However if a compare the speed of a press action with a Ikea Tradfri button, it takes over a second to perform the action.
The Ikea Tradfri button is an instant action while the TS0044 takes a while.

### Steps to reproduce

1. Interview went very fine.
2. Created an automation that turns light on.
3. Triggers are a Ikea Tradfri button and the Moes 4-gang.
4. Speed of the Ikea button is 1.5 seconds faster than Moes.

FYI, fresh zha and HA installation.

### Expected behavior

Same press / action speed.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

javicalle, 2023-05-30:
It would be great to have the device signature to see if it is a quirked device.

These Tuya devices have a `switch_mode` attribute. The accepted values are:
```python
class SwitchMode(t.enum8):
    """Tuya switch mode enum."""

    Command = 0x00
    Event = 0x01 
```

IIRC, in `Event` mode, the device can report multiple presses but it is slow (to check the current press type). The `Command` mode would be faster, but will report just one kind of button press (no double or triple press report).

MattWestb, 2023-05-31:
I think its the same problems i have observed with the TS004F in https://github.com/zigpy/zha-device-handlers/issues/1448.

I was not having time digging in it but ST looks having getting it working OK.

I think i shall being good if some user with TS004X devices that can sniffing the command sent and revived so we can debugging it and finding one way the system sending the command "tuya way" but the TS004F is little different but i think its the same problem in the end.

KoenHaemels, 2023-07-09:
Any step further on this issue?

wernerhp, 2023-08-23:
I have the same issue with the 2 gang Moes Smart. Takes up to 5 sec delay to turn on and 1 sec delay to turn off.
Dimmer Zigbee switch (TS0601 by _TZE200)
Quirk: zhaquirks.tuya.ts0601_dimmer.TuyaDoubleSwitchDimmerGP

closed 2023-10-29 by github-actions[bot]

wernerhp, 2023-10-30:
No solution?

KoenHaemels, 2023-10-30:
Never heard anything about it anymore. Asked the question on a different platform also, but less response…

morkiuz, 2023-11-12:
Same issue here… takes wayyy too long to turn on.

javicalle, 2023-11-12:
Once again:
the `TuyaSmartRemoteOnOffCluster` has a `switch_mode` attribute. Has anyone tried to get/change its value?