zigpy/zha-device-handlers #2416: frient Air Quality Sensor does not report VOC

smartmatic, 2023-05-30:
### Bug description

I've migrated from Z2M to ZHA using a ConBee II Zigbee radio. Since that the value for VOC is no longer reported. It's shown as unknown.

The Conbee II has the latest Firmware installed

### Steps to reproduce

I've removed the device several times from ZHA and re-added it again but issue still persist.

### Expected behavior

Show value for VOC

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]
[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

[zha-adfa12b853984fcc33aa8d149a7606a3-frient A_S AQSZB-110-2c61ed4b05930c1d2d11d15326c06fa7.json.txt](https://github.com/zigpy/zha-device-handlers/files/11596511/zha-adfa12b853984fcc33aa8d149a7606a3-frient.A_S.AQSZB-110-2c61ed4b05930c1d2d11d15326c06fa7.json.txt)

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

closed 2024-01-05 by github-actions[bot]

rozsap, 2024-04-01:
I can't believe this device still can't be used normally.

LinuxfarmerHH, 2025-02-06:
Bought this device, with ZHA it will be detected as device, but there is no include. It's running endless, have reset the device 10 times and tried again and again.

Core 2025.2.0
Supervisor 2025.02.0
Operating System 14.2
Frontend 20250205.0

Fliens, 2025-04-09:
I have the same issue
I've bought this sensor because of the voc values
So no activity on this issue means that I have to switch back to z2m