zigpy/zha-device-handlers #2417: [Device Support Request] TS0505A Lidl/Silvercrest RGBCCT light

FachKraft42, 2023-05-31:
### Problem description

I integrated two of them using ZHA with a HA SkyConnect. The color temperature slider is off, it is warm somewhere around the middle and cold at the ends of the slider. Setting a color doesn't work at all.

### Solution description

Color and color temperature setting in HA working correctly.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
1
2
3
4
5
6
7
8
9
10
11
12
13
14
15
16
17
18
19
20
21
22
23
24
25
26
27
28
29
30
31
32
33
34
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x010d",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0x1000"
[... 453 lines cut ...]
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

FachKraft42, 2023-06-03:
> One question do you have using ZHA-Tool kit and scanning attributes on the device ? I have the same problems onetime and also the dev for the tool kit and we was not getting it working OK until resetting the device and reading it. I think tuya is have made some magic that is being triggered if reading all possible attributes on the devices (that shall working OK if using find and bind as IKEA is doing in Dirigera but tuya is tuya).

> One question do you have using ZHA-Tool kit and scanning attributes on the device ? I have the same problems onetime and also the dev for the tool kit and we was not getting it working OK until resetting the device and reading it. I think tuya is have made some magic that is being triggered if reading all possible attributes on the devices (that shall working OK if using find and bind as IKEA is doing in Dirigera but tuya is tuya).

Hi, thanks for your reply. I tried scan device and tuya magic. Both did not change anything. I am new to HA and this are the only two devices I have added. Is there anything I need to do differently? Are there steps need to follow?

MattWestb, 2023-06-03:
LIDL lights is normally working out of the box and dont need tuya magic or other things only some need correcting the color settings that is wrong made is some devices and your device is using the right quirk for it so all things shall working OK.

Try deleting the light and restart HA and then its stable adding it new (resetting the light) and look if its working OK.

FachKraft42, 2023-06-03:
@MattWestb 

I did. Still no color control and temperature setting is still off.

MattWestb, 2023-06-03:
@TheJulianJES Do you having one of this devices that using quirk ?
I have one `TS0505  _TZ3000_odygigth`  and its not needing it and is working OK with both RGB and TC but the quirk is being used of _`TZ3000_dbou1ap4` that is reported working OK.

Elvanthyel, 2023-06-03:
Hi,
I have the exact same issue, but with a SONOFF Zigbee 3.0 USB Dongle, with ZHA.
Same lightbulb from LIDL, same device signature and same diagnostic information on my HA.

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant Supervised",
    "version": "2023.5.4",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.11",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Paris",
    "os_name": "Linux",
    "os_version": "5.10.0-21-amd64",
    "supervisor": "2023.04.1",
    "host_os": "Debian GNU/Linux 11 (bullseye)",
    "docker_version": "23.0.1",
    "chassis": "laptop",
    "run_as_root": true
  },
  "custom_components": {
    "samsungtv_encrypted": {
      "version": "2022.3.1",
      "requirements": [
        "wakeonlan>=1.1.6",
        "beautifulsoup4>=4.6.0",
        "netdisco>=2.8.3"
      ]
    },
    "nodered": {
      "version": "1.2.0",
      "requirements": []
    },
    "hacs": {
      "version": "1.32.1",
      "requirements": [
        "aiogithubapi>=22.10.1"
      ]
    },
    "liveboxtvuhd": {
      "version": "1.0.0",
      "requirements": [
        "fuzzywuzzy==0.18.0"
      ]
    }
  },
  "integration_manifest": {
    "domain": "zha",
    "name": "Zigbee Home Automation",
    "after_dependencies": [
      "onboarding",
      "usb"
    ],
    "codeowners": [
      "@dmulcahey",
      "@adminiuga",
      "@puddly"
    ],
    "config_flow": true,
    "dependencies": [
      "file_upload"
    ],
    "documentation": "https://www.home-assistant.io/integrations/zha",
    "iot_class": "local_polling",
[... 624 lines cut ...]
2023-06-03 16:04:12.314 DEBUG (MainThread) [zigpy.zcl] [0xAA8C:1:0x0300] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=78, command_id=10, *direction=<Direction.Client_to_Server: 1>)
2023-06-03 16:04:12.315 DEBUG (MainThread) [zigpy.zcl] [0xAA8C:1:0x0300] Decoded ZCL frame: LidlRGBCCTColorCluster:Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=uint8_t, value=254)), Attribute(attrid=0x0001, value=TypeValue(type=uint8_t, value=254)), Attribute(attrid=0xF001, value=TypeValue(type=uint8_t, value=32))])
2023-06-03 16:04:12.315 DEBUG (MainThread) [zigpy.zcl] [0xAA8C:1:0x0300] Received command 0x0A (TSN 78): Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=uint8_t, value=254)), Attribute(attrid=0x0001, value=TypeValue(type=uint8_t, value=254)), Attribute(attrid=0xF001, value=TypeValue(type=uint8_t, value=32))])
2023-06-03 16:04:12.316 DEBUG (MainThread) [zigpy.zcl] [0xAA8C:1:0x0300] Attribute report received: current_hue=254, current_saturation=254, 0xF001=32
2023-06-03 16:04:12.317 DEBUG (Thread-2) [aiosqlite] executing functools.partial(<built-in method commit of sqlite3.Connection object at 0x7f42bdb56640>)
2023-06-03 16:04:12.318 DEBUG (Thread-2) [aiosqlite] operation functools.partial(<built-in method commit of sqlite3.Connection object at 0x7f42bdb56640>) completed
2023-06-03 16:04:12.318 DEBUG (Thread-2) [aiosqlite] executing functools.partial(<built-in method execute of sqlite3.Connection object at 0x7f42bdb56640>, 'UPDATE devices_v11\n                    SET last_seen=:ts\n                    WHERE ieee=:ieee AND :ts - last_seen > :min_last_seen_delta', {'ts': 1685801052.313608, 'ieee': 60:a4:23:ff:fe:45:c3:dd, 'min_last_seen_delta': 30.0})
2023-06-03 16:04:12.319 DEBUG (Thread-2) [aiosqlite] operation functools.partial(<built-in method execute of sqlite3.Connection object at 0x7f42bdb56640>, 'UPDATE devices_v11\n                    SET last_seen=:ts\n                    WHERE ieee=:ieee AND :ts - last_seen > :min_last_seen_delta', {'ts': 1685801052.313608, 'ieee': 60:a4:23:ff:fe:45:c3:dd, 'min_last_seen_delta': 30.0}) completed
2023-06-03 16:04:12.319 DEBUG (Thread-2) [aiosqlite] executing functools.partial(<built-in method commit of sqlite3.Connection object at 0x7f42bdb56640>)
2023-06-03 16:04:12.320 DEBUG (Thread-2) [aiosqlite] operation functools.partial(<built-in method commit of sqlite3.Connection object at 0x7f42bdb56640>) completed
2023-06-03 16:04:12.320 DEBUG (Thread-2) [aiosqlite] executing functools.partial(<built-in method execute of sqlite3.Connection object at 0x7f42bdb56640>, 'INSERT INTO attributes_cache_v11 VALUES (?, ?, ?, ?, ?)\n                    ON CONFLICT (ieee, endpoint_id, cluster, attrid)\n                    DO UPDATE SET\n                        value=excluded.value WHERE value != excluded.value', (60:a4:23:ff:fe:45:c3:dd, 1, 768, 0, 254))
2023-06-03 16:04:12.321 DEBUG (Thread-2) [aiosqlite] operation functools.partial(<built-in method execute of sqlite3.Connection object at 0x7f42bdb56640>, 'INSERT INTO attributes_cache_v11 VALUES (?, ?, ?, ?, ?)\n                    ON CONFLICT (ieee, endpoint_id, cluster, attrid)\n                    DO UPDATE SET\n                        value=excluded.value WHERE value != excluded.value', (60:a4:23:ff:fe:45:c3:dd, 1, 768, 0, 254)) completed
2023-06-03 16:04:12.321 DEBUG (Thread-2) [aiosqlite] executing functools.partial(<built-in method commit of sqlite3.Connection object at 0x7f42bdb56640>)
[... cut ...]
2023-06-03 16:04:12.326 DEBUG (Thread-2) [aiosqlite] executing functools.partial(<built-in method commit of sqlite3.Connection object at 0x7f42bdb56640>)
2023-06-03 16:04:12.326 DEBUG (Thread-2) [aiosqlite] operation functools.partial(<built-in method commit of sqlite3.Connection object at 0x7f42bdb56640>) completed
2023-06-03 16:04:12.555 DEBUG (MainThread) [homeassistant.components.zha.entity] light.z_l_salon_light: transition complete - future attribute reports will write HA state
``` 
</details>

TheJulianJES, 2023-06-03:
No similar devices here.

Elvanthyel, 2023-06-03:
Hi there,
I have the exact same issue, but with a SONOFF Zigbee 3.0 USB Dongle V2, with ZHA.
Same lightbulb from LIDL.
(re-post from the issue openned here : https://github.com/home-assistant/core/issues/93746 )

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant Supervised",
    "version": "2023.5.4",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.11",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Paris",
    "os_name": "Linux",
    "os_version": "5.10.0-21-amd64",
    "supervisor": "2023.04.1",
    "host_os": "Debian GNU/Linux 11 (bullseye)",
    "docker_version": "23.0.1",
    "chassis": "laptop",
    "run_as_root": true
  },
  "custom_components": {
    "samsungtv_encrypted": {
      "version": "2022.3.1",
      "requirements": [
        "wakeonlan>=1.1.6",
        "beautifulsoup4>=4.6.0",
        "netdisco>=2.8.3"
      ]
    },
    "nodered": {
      "version": "1.2.0",
      "requirements": []
    },
    "hacs": {
      "version": "1.32.1",
      "requirements": [
        "aiogithubapi>=22.10.1"
      ]
    },
    "liveboxtvuhd": {
      "version": "1.0.0",
      "requirements": [
        "fuzzywuzzy==0.18.0"
      ]
    }
  },
  "integration_manifest": {
    "domain": "zha",
    "name": "Zigbee Home Automation",
    "after_dependencies": [
      "onboarding",
      "usb"
    ],
    "codeowners": [
      "@dmulcahey",
      "@adminiuga",
      "@puddly"
    ],
    "config_flow": true,
    "dependencies": [
      "file_upload"
    ],
    "documentation": "https://www.home-assistant.io/integrations/zha",
[... 629 lines cut ...]
2023-06-03 16:04:12.317 DEBUG (Thread-2) [aiosqlite] executing functools.partial(<built-in method commit of sqlite3.Connection object at 0x7f42bdb56640>)
2023-06-03 16:04:12.318 DEBUG (Thread-2) [aiosqlite] operation functools.partial(<built-in method commit of sqlite3.Connection object at 0x7f42bdb56640>) completed
2023-06-03 16:04:12.318 DEBUG (Thread-2) [aiosqlite] executing functools.partial(<built-in method execute of sqlite3.Connection object at 0x7f42bdb56640>, 'UPDATE devices_v11\n                    SET last_seen=:ts\n                    WHERE ieee=:ieee AND :ts - last_seen > :min_last_seen_delta', {'ts': 1685801052.313608, 'ieee': 60:a4:23:ff:fe:45:c3:dd, 'min_last_seen_delta': 30.0})
2023-06-03 16:04:12.319 DEBUG (Thread-2) [aiosqlite] operation functools.partial(<built-in method execute of sqlite3.Connection object at 0x7f42bdb56640>, 'UPDATE devices_v11\n                    SET last_seen=:ts\n                    WHERE ieee=:ieee AND :ts - last_seen > :min_last_seen_delta', {'ts': 1685801052.313608, 'ieee': 60:a4:23:ff:fe:45:c3:dd, 'min_last_seen_delta': 30.0}) completed
2023-06-03 16:04:12.319 DEBUG (Thread-2) [aiosqlite] executing functools.partial(<built-in method commit of sqlite3.Connection object at 0x7f42bdb56640>)
2023-06-03 16:04:12.320 DEBUG (Thread-2) [aiosqlite] operation functools.partial(<built-in method commit of sqlite3.Connection object at 0x7f42bdb56640>) completed
2023-06-03 16:04:12.320 DEBUG (Thread-2) [aiosqlite] executing functools.partial(<built-in method execute of sqlite3.Connection object at 0x7f42bdb56640>, 'INSERT INTO attributes_cache_v11 VALUES (?, ?, ?, ?, ?)\n                    ON CONFLICT (ieee, endpoint_id, cluster, attrid)\n                    DO UPDATE SET\n                        value=excluded.value WHERE value != excluded.value', (60:a4:23:ff:fe:45:c3:dd, 1, 768, 0, 254))
2023-06-03 16:04:12.321 DEBUG (Thread-2) [aiosqlite] operation functools.partial(<built-in method execute of sqlite3.Connection object at 0x7f42bdb56640>, 'INSERT INTO attributes_cache_v11 VALUES (?, ?, ?, ?, ?)\n                    ON CONFLICT (ieee, endpoint_id, cluster, attrid)\n                    DO UPDATE SET\n                        value=excluded.value WHERE value != excluded.value', (60:a4:23:ff:fe:45:c3:dd, 1, 768, 0, 254)) completed
2023-06-03 16:04:12.321 DEBUG (Thread-2) [aiosqlite] executing functools.partial(<built-in method commit of sqlite3.Connection object at 0x7f42bdb56640>)
2023-06-03 16:04:12.322 DEBUG (Thread-2) [aiosqlite] operation functools.partial(<built-in method commit of sqlite3.Connection object at 0x7f42bdb56640>) completed
2023-06-03 16:04:12.323 DEBUG (Thread-2) [aiosqlite] executing functools.partial(<built-in method execute of sqlite3.Connection object at 0x7f42bdb56640>, 'INSERT INTO attributes_cache_v11 VALUES (?, ?, ?, ?, ?)\n                    ON CONFLICT (ieee, endpoint_id, cluster, attrid)\n                    DO UPDATE SET\n                        value=excluded.value WHERE value != excluded.value', (60:a4:23:ff:fe:45:c3:dd, 1, 768, 1, 254))
2023-06-03 16:04:12.323 DEBUG (Thread-2) [aiosqlite] operation functools.partial(<built-in method execute of sqlite3.Connection object at 0x7f42bdb56640>, 'INSERT INTO attributes_cache_v11 VALUES (?, ?, ?, ?, ?)\n                    ON CONFLICT (ieee, endpoint_id, cluster, attrid)\n                    DO UPDATE SET\n                        value=excluded.value WHERE value != excluded.value', (60:a4:23:ff:fe:45:c3:dd, 1, 768, 1, 254)) completed
2023-06-03 16:04:12.324 DEBUG (Thread-2) [aiosqlite] executing functools.partial(<built-in method commit of sqlite3.Connection object at 0x7f42bdb56640>)
[... cut ...]
2023-06-03 16:04:12.555 DEBUG (MainThread) [homeassistant.components.zha.entity] light.z_l_salon_light: transition complete - future attribute reports will write HA state
``` 
</details> 

additionnal behavior :
- after some times, HA display the bulb as off, even if the bulb is still lit.
- some times, when accessing the device on HA, or when the webpage auto-refresh, it toggle the bulb state, but HA don't display the command on the logbook. The state displayed on HA stay the same as the previous one (if the bulb is off, when accessing the bulb device page on HA, the bulb lit itself, but HA display it as off, and no new line on the logbook).

pipiche38, 2023-10-23:
It looks like under certain circumstance only Color command  0x06 Move to Hue and Saturation works. In addition Tuya use a specific command on cluster 0x0300 to switch to RGB or White mode

Tuya Color Mode
Cluster 0x0300
Cmd 0xf0
Payload
0 for white light
1 for color light

Tuya move to Hue and Saturation
Cluster 0x0300
Cmd 0x06
Payload: hue + saturation + brightness

MattWestb, 2023-10-23:
Great findings @pipiche38 !!
Do you knowing if its for only "normal" B versions (NOT Zigbee certified ones) or also the A version that LIDL is using or opposite ?
Was scanning all attributes on one LIDL and it was not able going out color mode until resetting it so this sounds being the answer of what  it was not changing to CT.

Atiroocky, 2024-03-13:
Hello,

Same issue here. I have 6 `TS505A_TZ3000_odygigth` and and can’t apply color. Only white and the behavior is pretty weird, as described earlier in this thread.

But I also have three other RGB bulbs from LIDL like this one.
[image]

These are shown as `TS505A_TZ3000_kdpxju99`
They all work perfectly in ZHA with Sonoff Zigbee (type P). Both white and colors. And on/off states are reliable. 

Is there a way to fake the `TS505A_TZ3000_odygigth` with `TS505A_TZ3000_kdpxju99` integration ?

(ps : Is there any link with "quirk" method ?)

Thanks ;-)

MattWestb, 2024-03-13:
Try deleting the device and restart HA and adding it new.
I was having the same problems and also the dev of ZHA-toolkit after scanning the device for commands and attribute with toolkit but both was getting it working after adding it new.
tuya have make special commands color changes that is not standard but it looks working aslong only using standard commands on this devices (its one A series so its Zigbee certified that most tuya device is not).

Atiroocky, 2024-03-13:
> Try deleting the device and restart HA and adding it new. I was having the same problems and also the dev of ZHA-toolkit after scanning the device for commands and attribute with toolkit but both [...]

Delete devices, restart HA, re-pair and that's it !
Big thanks to you ;-)

closed 2024-09-16 by github-actions[bot]