zigpy/zha-device-handlers #2428: Support for MOES (Tuya) 2-gang and 3-gang touch switches

mitnits, 2023-06-11:
### Problem description

Using ZHA in home assistant (v2023.6.1) and tuya (MOES) multi-gang switches I can't control lights separately.

Device info from ZHA is below:

```
Device info
TS0012
by _TZ3000_18ejxno0
Connected via Zigbee Coordinator
Zigbee info
IEEE: a4:c1:38:4c:bc:c3:0e:43
Nwk: 0x8f60
Device Type: EndDevice
LQI: 116
RSSI: -71
Last Seen: 2023-06-10T10:38:53
Power Source: Battery or Unknown
Quirk: zhaquirks.tuya.ts001x.TuyaDoubleNoNeutralSwitch_2
```

LMK what, if any, debug information I can supply to help fix this.

### Solution description

I'd like to get an improved quirks file that would let me control the lights separately. I also have 3-gang switches from the same batch that exhibit an identical issue.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

javicalle, 2023-06-18:
IIRC you can try to fix it just with the 'reconfigure' button from the device view.
If not working, I would suggest you to remove the device, wait for a couple of minutes and then pair the device again.

marcelodf12, 2023-09-02:
I have the same problem. I bought 5 switches of 2 and 3 groups. I have the same problem with everyone.
'Reconfigure' or 'removing and re-pair' the device doesn't fix it

closed 2024-03-07 by github-actions[bot]