zigpy/zha-device-handlers #2429: [Device Support Request] Tuya Zigbee Human Presence Sensor. TS0601 by  _TZE204_sxm7l9xa

Agimgrue, 2023-06-12:
### Problem description

Only device exist, with any thing else.

TS0601
por _TZE204_sxm7l9xa
Conectado a través de [Zigbee Coordinator](http://192.168.88.27:8123/config/devices/device/0bbeb1a8967443d367f497938a79e249)
Zigbee info
IEEE: a4:c1:38:9f:d2:31:dd:b9
Nwk: 0x8b19
Device Type: Router
LQI: 111
RSSI: Desconocido
Visto por última vez: 2023-06-12T20:04:07
Fuente de alimentación: Mains

I've see an external converter js to run it on z2m.

### Solution description

I post it here if anyone can adapt it as a custom quirk or include it on ZHA library

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE204_sxm7l9xa",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information
[... 505 lines cut ...]
        exposes.numeric('fading_time', ea.STATE_SET).withValueMin(0).withValueMax(1500).withValueStep(1)
            .withDescription('Fading time').withUnit('s'),
        // exposes.text('cli', ea.STATE).withDescription('not recognize'),
        exposes.enum('self_test', ea.STATE, Object.values(tuya.tuyaHPSCheckingResult))
            .withDescription('Self_test, possible resuts: checking, check_success, check_failure, others, comm_fault, radar_fault.'),
    ],
    meta: {
        tuyaDatapoints: [
            [104, 'illuminance', tuya.valueConverter.raw],
            [105, 'presence', tuya.valueConverter.boolean],
            [106, 'radar_sensitivity', tuya.valueConverter.raw],
            [107, 'maximum_range', tuya.valueConverter.raw],
            [108, 'minimum_range', tuya.valueConverter.raw],
            [109, 'target_distance', tuya.valueConverter.raw],
            [110, 'detection_delay', tuya.valueConverter.raw],
            [111, 'fading_time', tuya.valueConverter.raw],
        ],
    },
};

module.exports = definition;

```

</details>

dipseth, 2023-06-12:
I have similiar issue but for `_TZE204_sooucan5`

blainfaulkner, 2023-06-15:
I've got the same issue here _TZE204_sxm7l9xa,  When added it only displays diagnostics.  Any information I can provide let me know.

dipseth, 2023-06-19:
FYI [this](https://github.com/zigpy/zha-device-handlers/issues/2059#issuecomment-1459991922) work for me for presence sensor entity and a switch:

niknik98, 2023-06-19:
I have the same issue with TZE204_sxm7l9xa

nictronik99, 2023-06-28:
same issue  with TZE204_sxm7l9xa, i've not tested the @dipseth suggestion yet but i'm new to this world and i don't know the difference between TZE204 with different _xxxxxxxx. someone who knows how to get working this sensor can reply please?

dipseth, 2023-06-29:
@nictronik99 If your device's signature matches what was originally mentioned (not counting the manufacturer code  ` TZE204_...` , but the look at the input_clusters clusters, profile_id, etc.) the link I provided should work I believe.  Elsewhere in that thread there are instructions for how to implement a custom quirk.

nictronik99, 2023-06-29:
> @nictronik99 If your device's signature matches what was originally mentioned (not counting the manufacturer code  ` TZE204_...` , but the look at the input_clusters clusters, profile_id, etc.) the [...]

I tried a different route.... I switched to zigbee2mqtt....

labaland, 2023-07-14:
I need for "_TZE204_sooucan5"   Would be nice if someone can add it. I dont know how :( 
[image]

monstergubl, 2023-08-05:
im interested too not work since yet

adey, 2023-08-18:
Any update?

Thanks.

labaland, 2023-08-18:
It works now! But they can make some things better! Presence works perfect
:)

racedo, 2023-08-20:
@labaland could you elaborate on what works? The lates HA still doesn't show any sensors.

I'm attaching what HA sees when adding the device, which still makes the device unusable.

Thanks.

[image]

Woll3, 2023-08-25:
Custom quirk available here: [[Device Support Request] ZY-M100 #2378](https://github.com/zigpy/zha-device-handlers/issues/2378)

Support will be added: [Add support for more TS0601 sensors and configuration](https://github.com/zigpy/zha-device-handlers/pull/2485)

sneakythr0ws, 2024-01-08:
don't forget _TZE204_sooucan5 pls

chimeranzl, 2024-04-10:
and _TZW204_mtoaryre please!

Amerigo1973, 2024-06-19:
Simeone can add TZE204_nbkshs6k plesse?

closed 2024-12-23 by github-actions[bot]