zigpy/zha-device-handlers #2433: [Device Support Request] MOES Zigbee 006 Thermostat Model TS0601 manufacturer_TZE204_aoclfnxz

bsfaxi, 2023-06-15:
### Problem description

I bought this Moes Thermostat to control an electric underfloor heating system : https://fr.aliexpress.com/item/1005004996552864.html
The existing quirk is only for TZE200 that has one Endpoint only. This one has 2. 

### Solution description

Adapt the existing tuya quirk "ts0601_electric_heating.py" to include this model.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]
[image]
[image]
[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE204_aoclfnxz",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.6.2",
    "dev": false,
    "hassio": true,
[... 341 lines cut ...]
}

```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

Yop1403, 2023-08-06:
I support this request - I have exactly the same device (and the same problem)

@bsfaxi if you change the signature in the Quirk of the TZE200 to:
`MODELS_INFO:` [("_TZE204_aoclfnxz", "TS0601")],`
--> are you getting any sensors at all or also nothing?

bsfaxi, 2023-08-07:
> 

Hi @Yop1403,
The Thermostat entity appears, but unfortunately I can't control it !
[image]
[image]

I have the following errors when I try to change the temperature :
```
2023-08-02 22:05:05.842 WARNING (MainThread) [homeassistant.helpers.entity] Updating state for sensor.thermostat_bureau_hvac_action (<class 'homeassistant.components.zha.sensor.ThermostatHVACAction'>) took 0.418 seconds. Please create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+zha%22
2023-08-04 00:15:49.871 WARNING (MainThread) [homeassistant.helpers.entity] Updating state for climate.thermostat_bureau (<class 'homeassistant.components.zha.climate.Thermostat'>) took 0.466 seconds. Please create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+zha%22

```

Yop1403, 2023-08-07:
Very interesting. You at least get further than I do... My HA instance doesn't want to see the Quirk:
[image]

Would you mind sharing what this entry looks like in your Quirk file?
`    signature = {
        MODELS_INFO: [("_TZE204_aoclfnxz", "TS0601")],
`

bsfaxi, 2023-08-14:
The custom quirk, that I'm using, is in the description of this request in "Custom quirk" section. :)

Yop1403, 2023-08-15:
> The custom quirk, that I'm using, is in the description of this request in "Custom quirk" section. :)

Ah sorry overlooked that, thanks. Let's see if anyone here has some mercy with us and helps us out getting this Moes Thermostat integrated into HA...

lotr, 2023-08-15:
https://github.com/Koenkk/zigbee2mqtt/issues/18097 z2m should have it already

bsfaxi, 2023-08-16:
> https://www.zigbee2mqtt.io/devices/BHT-006GBZB.html

Indeed, it's already supported there: https://www.zigbee2mqtt.io/devices/BHT-006GBZB.html.
I need it in ZHA. ;)

alexlinno, 2023-08-16:
Hi, the quirk in the the first comment did help in a sense that I see the temp and target temp, And the state, And I see the more useful clusters. I will be modifying the quirk to be able to control the device.
[image]
[image]

douglascrc-git, 2023-08-17:
I support this request too! I'm trying to make this thermostat work on ZHA, but no succes has been achieved so far ;C

bsfaxi, 2023-08-20:
For those who speaks Zigbee, I tried to generate logs during:
1- The pairing process
[TS0601_pairing_log.txt](https://github.com/zigpy/zha-device-handlers/files/12388603/TS0601_pairing_log.txt)

2- The following scenario:
- switch on
- set high temperature to activate the heat
- set low  temperature to deactivate the heat
- switch between manual and schedule
- set program schedule day by day
- switch off
[TS0601_scenario_log.txt](https://github.com/zigpy/zha-device-handlers/files/12388605/TS0601_scenario_log.txt)

Anyone can help please?

bsfaxi, 2023-08-20:
Here the 3 recurring commands that I see:

Command 2:
```
2023-08-20 19:30:06.305 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x9E07), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=None, profile_id=260, cluster_id=61184, data=Serialized[b'\t\xbf\x02\r\xf8\x18\x02\x00\x04\x00\x00\x00\xff'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=207, rssi=-68)
2023-08-20 19:30:06.305 DEBUG (MainThread) [zigpy.zcl] [0x9E07:1:0xef00] Received ZCL frame: b'\t\xbf\x02\r\xf8\x18\x02\x00\x04\x00\x00\x00\xff'
2023-08-20 19:30:06.306 DEBUG (MainThread) [zigpy.zcl] [0x9E07:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), tsn=191, command_id=2, *direction=<Direction.Client_to_Server: 1>)
2023-08-20 19:30:06.306 WARNING (MainThread) [zigpy.zcl] [0x9E07:1:0xef00] Unknown cluster command 2 b'\r\xf8\x18\x02\x00\x04\x00\x00\x00\xff'
2023-08-20 19:30:06.308 DEBUG (MainThread) [zigpy.zcl] [0x9E07:1:0xef00] Received command 0x02 (TSN 191): b'\r\xf8\x18\x02\x00\x04\x00\x00\x00\xff'
2023-08-20 19:30:06.309 DEBUG (MainThread) [zigpy.zcl] [0x9E07:1:0xef00] No explicit handler for cluster command 0x02: b'\r\xf8\x18\x02\x00\x04\x00\x00\x00\xff'

2023-08-20 19:35:41.504 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x9E07), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=None, profile_id=260, cluster_id=0, data=Serialized[b"\x08\x9b\n\xdf\xffB<\x81'u,i\x82'u,i\x80'u,i\x81'u,i\x81'u,i\x82'u,i\x83'u,i\x83'u,i\x84'u,i\x7f'u,i\x7f'u,i\x7f'u,i"], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=255, rssi=-57)
2023-08-20 19:35:41.505 DEBUG (MainThread) [zigpy.zcl] [0x9E07:1:0x0000] Received ZCL frame: b"\x08\x9b\n\xdf\xffB<\x81'u,i\x82'u,i\x80'u,i\x81'u,i\x81'u,i\x82'u,i\x83'u,i\x83'u,i\x84'u,i\x7f'u,i\x7f'u,i\x7f'u,i"
2023-08-20 19:35:41.505 DEBUG (MainThread) [zigpy.zcl] [0x9E07:1:0x0000] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True), tsn=155, command_id=10, *direction=<Direction.Client_to_Server: 1>)
2023-08-20 19:35:41.506 DEBUG (MainThread) [zigpy.zcl] [0x9E07:1:0x0000] Decoded ZCL frame: Basic:Report_Attributes(attribute_reports=[Attribute(attrid=0xFFDF, value=TypeValue(type=CharacterString, value="�'u,i�'u,i�'u,i�'u,i�'u,i�'u,i�'u,i�'u,i�'u,i\x7f'u,i\x7f'u,i\x7f'u,i"))])
2023-08-20 19:35:41.507 DEBUG (MainThread) [zigpy.zcl] [0x9E07:1:0x0000] Received command 0x0A (TSN 155): Report_Attributes(attribute_reports=[Attribute(attrid=0xFFDF, value=TypeValue(type=CharacterString, value="�'u,i�'u,i�'u,i�'u,i�'u,i�'u,i�'u,i�'u,i�'u,i\x7f'u,i\x7f'u,i\x7f'u,i"))])
2023-08-20 19:35:41.507 DEBUG (MainThread) [zigpy.zcl] [0x9E07:1:0x0000] Attribute report received: 0xFFDF="�'u,i�'u,i�'u,i�'u,i�'u,i�'u,i�'u,i�'u,i�'u,i\x7f'u,i\x7f'u,i\x7f'u,i"
2023-08-20 19:35:41.509 DEBUG (MainThread) [zigpy.zcl] [0x9E07:1:0x0000] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=155, command_id=<GeneralCommand.Default_Response: 11>, *direction=<Direction.Client_to_Server: 1>)
2023-08-20 19:35:41.510 DEBUG (MainThread) [zigpy.zcl] [0x9E07:1:0x0000] Sending reply: Default_Response(command_id=10, status=<Status.SUCCESS: 0>)
```

Command 10:
```
2023-08-20 19:36:38.523 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x9E07), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=None, profile_id=260, cluster_id=0, data=Serialized[b"\x08\x00\n\xdf\xffB<\xbb'u,i\xbb'u,i\xbc'u,i\xbc'u,i\xb7'u,i\xb8'u,i\xb8'u,i\xb9'u,i\xb9'u,i\xba'u,i\xba'u,i\xba'u,i"], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=239, rssi=-64)
2023-08-20 19:36:38.524 DEBUG (MainThread) [zigpy.zcl] [0x9E07:1:0x0000] Received ZCL frame: b"\x08\x00\n\xdf\xffB<\xbb'u,i\xbb'u,i\xbc'u,i\xbc'u,i\xb7'u,i\xb8'u,i\xb8'u,i\xb9'u,i\xb9'u,i\xba'u,i\xba'u,i\xba'u,i"
2023-08-20 19:36:38.524 DEBUG (MainThread) [zigpy.zcl] [0x9E07:1:0x0000] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True), tsn=0, command_id=10, *direction=<Direction.Client_to_Server: 1>)
2023-08-20 19:36:38.525 DEBUG (MainThread) [zigpy.zcl] [0x9E07:1:0x0000] Decoded ZCL frame: Basic:Report_Attributes(attribute_reports=[Attribute(attrid=0xFFDF, value=TypeValue(type=CharacterString, value="�'u,i�'u,i�'u,i�'u,i�'u,i�'u,i�'u,i�'u,i�'u,i�'u,i�'u,i�'u,i"))])
[... cut ...] e_default_response=0, reserved=0, *is_cluster=True, *is_general=False), tsn=190, command_id=36, *direction=<Direction.Client_to_Server: 1>)
2023-08-20 19:30:05.903 WARNING (MainThread) [zigpy.zcl] [0x9E07:1:0xef00] Unknown cluster command 36 b'\xf7\r'
2023-08-20 19:30:05.905 DEBUG (MainThread) [zigpy.zcl] [0x9E07:1:0xef00] Received command 0x24 (TSN 190): b'\xf7\r'
2023-08-20 19:30:05.905 DEBUG (MainThread) [zigpy.zcl] [0x9E07:1:0xef00] No explicit handler for cluster command 0x24: b'\xf7\r'
```

mlouaze, 2023-08-20:
Hi, I just got the same thermostat and I also struggle to make it work. The custom quirks also allowed me to see the thermostat, but I also can't control it. I am a beginner in HA and ZHA, but I will also try to investigate.

bsfaxi, 2023-08-21:
When I applied the custom quirk provided in the description, I notice the following:

**FOR CLUSTER 02:**
Report room temperature 24°
`2023-08-21 10:05:58.369 DEBUG (MainThread) [zigpy.zcl] [0x990F:1:0xef00] Received command 0x02 (TSN 214): set_data_response(param=Command(status=9, tsn=122, command_id=536, function=0, data=[4, 0, 0, 0, 240]))`
Report room temperature 24.5°
`2023-08-21 10:06:57.385 DEBUG (MainThread) [zigpy.zcl] [0x990F:1:0xef00] Received command 0x02 (TSN 135): set_data_response(param=Command(status=9, tsn=180, command_id=536, function=0, data=[4, 0, 0, 0, 245]))`
You notice the command_id "536".

When nothing happens, I see a recurring message like the following:
`2023-08-21 10:05:55.953 DEBUG (MainThread) [zigpy.zcl] [0x990F:1:0xef00] Received command 0x02 (TSN 212): set_data_response(param=Command(status=9, tsn=122, command_id=1060, function=0, data=[1, 1]))`
You notice the command_id "1060".

When, I set the target temperature on the device to 20°, I see the following:
`2023-08-21 10:07:00.185 DEBUG (MainThread) [zigpy.zcl] [0x990F:1:0xef00] Received command 0x02 (TSN 162): set_data_response(param=Command(status=9, tsn=204, command_id=528, function=0, data=[4, 0, 0, 0, 20]))`
Even though I set the target temperature to 20.5°, the data still contains 20°:
`2023-08-21 10:23:44.910 DEBUG (MainThread) [zigpy.zcl] [0x990F:1:0xef00] Received command 0x02 (TSN 156): set_data_response(param=Command(status=13, tsn=115, command_id=528, function=0, data=[4, 0, 0, 0, 20]))`
Setting the target temperature to 21° shows 21 in the data:
`2023-08-21 10:29:30.481 DEBUG (MainThread) [zigpy.zcl] [0x990F:1:0xef00] Received command 0x02 (TSN 132): set_data_response(param=Command(status=14, tsn=67, command_id=528, function=0, data=[4, 0, 0, 0, 21]))`
You notice the command_id "528".

I see also a recurring command_id "536" every 1mn. But, I don't know to what it corresponds:
```
2023-08-20 21:05:28.659 DEBUG (MainThread) [zigpy.zcl] [0x990F:1:0xef00] Decoded ZCL frame: MoesBHTManufCluster:set_data_response(param=Command(status=91, tsn=38, command_id=536, function=0, data=[4, 0, 0, 0, 225]))
2023-08-20 21:05:28.659 DEBUG (MainThread) [zigpy.zcl] [0x990F:1:0xef00] Received command 0x02 (TSN 123): set_data_response(param=Command(status=91, tsn=38, command_id=536, function=0, data=[4, 0, 0, 0, 225]))
2023-08-20 21:06:28.685 DEBUG (MainThread) [zigpy.zcl] [0x990F:1:0xef00] Decoded ZCL frame: MoesBHTManufCluster:set_data_response(param=Command(status=91, tsn=112, command_id=536, function=0, data=[4, 0, 0, 0, 225]))
2023-08-20 21:06:28.685 DEBUG (MainThread) [zigpy.zcl] [0x990F:1:0xef00] Received command 0x02 (TSN 120): set_data_response(param=Command(status=91, tsn=112, command_id=536, function=0, data=[4, 0, 0, 0, 225]))
2023-08-20 21:07:28.720 DEBUG (MainThread) [zigpy.zcl] [0x990F:1:0xef00] Decoded ZCL frame: MoesBHTManufCluster:set_data_response(param=Command(status=91, tsn=146, command_id=536, function=0, data=[4, 0, 0, 0, 230]))
2023-08-20 21:07:28.720 DEBUG (MainThread) [zigpy.zcl] [0x990F:1:0xef00] Received command 0x02 (TSN 67): set_data_response(param=Command(status=91, tsn=146, command_id=536, function=0, data=[4, 0, 0, 0, 230]))
2023-08-20 21:08:28.758 DEBUG (MainThread) [zigpy.zcl] [0x990F:1:0xef00] Decoded ZCL frame: MoesBHTManufCluster:set_data_response(param=Command(status=91, tsn=188, command_id=536, function=0, data=[4, 0, 0, 0, 230]))
2023-08-20 21:08:28.759 DEBUG (MainThread) [zigpy.zcl] [0x990F:1:0xef00] Received command 0x02 (TSN 13): set_data_response(param=Command(status=91, tsn=188, command_id=536, function=0, data=[4, 0, 0, 0, 230]))
2023-08-20 21:09:28.791 DEBUG (MainThread) [zigpy.zcl] [0x990F:1:0xef00] Decoded ZCL frame: MoesBHTManufCluster:set_data_response(param=Command(status=91, tsn=220, command_id=536, function=0, data=[4, 0, 0, 0, 230]))
2023-08-20 21:09:28.791 DEBUG (MainThread) [zigpy.zcl] [0x990F:1:0xef00] Received command 0x02 (TSN 216): set_data_response(param=Command(status=91, tsn=220, command_id=536, function=0, data=[4, 0, 0, 0, 230]))
2023-08-20 21:10:28.823 DEBUG (MainThread) [zigpy.zcl] [0x990F:1:0xef00] Decoded ZCL frame: MoesBHTManufCluster:set_data_response(param=Command(status=92, tsn=17, command_id=536, function=0, data=[4, 0, 0, 0, 235]))
2023-08-20 21:10:28.824 DEBUG (MainThread) [zigpy.zcl] [0x990F:1:0xef00] Received command 0x02 (TSN 162): set_data_response(param=Command(status=92, tsn=17, command_id=536, function=0, data=[4, 0, 0, 0, 235]))
2023-08-20 21:11:29.363 DEBUG (MainThread) [zigpy.zcl] [0x990F:1:0xef00] Decoded ZCL frame: MoesBHTManufCluster:set_data_response(param=Command(status=92, tsn=71, command_id=536, function=0, data=[4, 0, 0, 0, 235]))
2023-08-20 21:11:29.364 DEBUG (MainThread) [zigpy.zcl] [0x990F:1:0xef00] Received command 0x02 (TSN 221): set_data_response(param=Command(status=92, tsn=71, command_id=536, function=0, data=[4, 0, 0, 0, 235]))
2023-08-20 21:12:29.398 DEBUG (MainThread) [zigpy.zcl] [0x990F:1:0xef00] Decoded ZCL frame: MoesBHTManufCluster:set_data_response(param=Command(status=92, tsn=94, command_id=536, function=0, data=[4, 0, 0, 0, 240]))
2023-08-20 21:12:29.398 DEBUG (MainThread) [zigpy.zcl] [0x990F:1:0xef00] Received command 0x02 (TSN 145): set_data_response(param=Command(status=92, tsn=94, command_id=536, function=0, data=[4, 0, 0, 0, 240]))
[... cut ...]
2023-08-21 10:19:56.623 DEBUG (MainThread) [zigpy.zcl] [0x990F:1:0xef00] Received command 0x24 (TSN 144): set_time_request(param=[132, 12])
2023-08-21 10:19:56.818 DEBUG (MainThread) [zigpy.zcl] [0x990F:1:0xef00] Received command 0x24 (TSN 144): set_time_request(param=[132, 12])
2023-08-21 10:19:57.018 DEBUG (MainThread) [zigpy.zcl] [0x990F:1:0xef00] Received command 0x24 (TSN 144): set_time_request(param=[132, 12])

2023-08-21 10:06:56.181 DEBUG (MainThread) [zigpy.zcl] [0x990F:1:0xef00] No explicit handler for cluster command 0x24: set_time_request(param=[179, 9])

```

mlouaze, 2023-08-21:
I guess the 536 is just sent by the thermostat every minute to notify the current  temperature. 536 is 218 in hexadecimal, and you can see it as MOESBHT_TEMPERATURE_ATTR = 0x0218 in the custom quirk. This seems to make sense since it is also the command_id you have when you report the room temperature.
Following the same idea 1060 is 424 in hexadecimal so MOESBHT_RUNNING_MODE_ATTR = 0x0424  n the custom quirk. And 528 is 0x210 so MOESBHT_TARGET_TEMP_ATTR = 0x0210 which also seems to make sense.

But maybe you had already figured this out ...

bsfaxi, 2023-08-21:
Another notice : The data sent by the COMMAND "0x24" every 1 minute contains the last TSN number and the Status from the **last** "command_id=1060".

mlouaze, 2023-08-21:
Pure Guess but 36 could be the number of information you just sent. 12 timeslot * 3 information per timeslot (hours, minutes and temperature).
Does your thermostat seems to receive/acknowledge and apply those command or not?

bsfaxi, 2023-08-21:
> Pure Guess but 36 could be the number of information you just sent. 12 timeslot * 3 information per timeslot (hours, minutes and temperature).

I think so. :)

bsfaxi, 2023-08-21:
Changing the status to heating:
`2023-08-21 17:47:17.145 DEBUG (MainThread) [zigpy.zcl] [0x990F:1:0xef00] Received command 0x02 (TSN 189): set_data_response(param=Command(status=14, tsn=150, command_id=1060, function=0, data=[1, 0]))`

changing the status to idle:
`2023-08-21 17:55:45.479 DEBUG (MainThread) [zigpy.zcl] [0x990F:1:0xef00] Received command 0x02 (TSN 252): set_data_response(param=Command(status=16, tsn=193, command_id=1060, function=0, data=[1, 1]))`

bsfaxi, 2023-08-21:
Changing the status to locked:
`2023-08-21 18:26:57.887 DEBUG (MainThread) [zigpy.zcl] [0x990F:1:0xef00] Received command 0x02 (TSN 125): set_data_response(param=Command(status=24, tsn=62, command_id=296, function=0, data=[1, 1]))`

Changing the status to unlocked:
`2023-08-21 18:32:01.506 DEBUG (MainThread) [zigpy.zcl] [0x990F:1:0xef00] Received command 0x02 (TSN 51): set_data_response(param=Command(status=25, tsn=142, command_id=296, function=0, data=[1, 0]))`

mlouaze, 2023-08-21:
I checked all your values, it seems they all follow the ATTR in the custom quirk. 
[image]

So to recap we have :
528 (210 in hexadec) for the value of the target temperature (three 0 then value in degree). ex : [4, 0, 0, 0, 20] for 20°
536 (218 in hexadec) for the value of the actual ambiant temperature (three 0 then value in decidegree). ex : [4, 0, 0, 0, 205] for 20.5°
1026 (402 in hexadec) for the value manual mode (1=false and 0=true). ex : [1, 0] for manual mode enabled
257 (101 in hexadec) for the enabling of the device (0=off and1=on). ex : [1, 1] for device on
1060 (424 in hexadec) for the heating status (1=idle and 0=heating). ex: [1, 1] for device not heating (idling)
296 (128 in hexadec) for the lock//childlock (0=unlocked and 1=locked). ex [1, 1] for device locked

Those command come in arrays with a first number indicating how many values are in the array
All those are already in the custom quirk and should then be working, I suppose.

The schedule command is not in the custom quirk already:
101 (65 in hexadec) for the temperature schedule. see https://github.com/zigpy/zha-device-handlers/issues/2433#issuecomment-1686336394 

This explains why we can see what is happening on the device from homeassistant. But we still need to figure out why we can't send values to the thermostat correctly.

bsfaxi, 2023-08-21:
Yes, all these are to GET the statuses.
I don't know how to catch the SET commands.

mlouaze, 2023-08-21:
[image]
When sending a target_temperature using this inteface (sorry it is in french), I get the following message in the log :
2023-08-21 19:47:04.750 DEBUG (MainThread) [homeassistant.components.zha.websocket_api] Set attribute for: cluster_id: [61184] cluster_type: [in] endpoint_id: [1] attribute: [528] value: [10] manufacturer: [None] response: [[[WriteAttributesStatusRecord(status=<Status.SUCCESS: 0>)]]]

This seems goog but after writing 10 and then reading the value again I still see 18.

bsfaxi, 2023-08-21:
Each time, I try from HA to switch the status from "on" to "off", I have the following message:
`2023-08-21 20:23:42.676 DEBUG (MainThread) [zhaquirks.tuya] [0x990f:1:0x0201] Mapping standard system_mode (0x001c) with value <SystemMode.Off: 0> to custom {257: 0}`

And from "off" to "on":
`2023-08-21 20:25:04.023 DEBUG (MainThread) [zhaquirks.tuya] [0x990f:1:0x0201] Mapping standard system_mode (0x001c) with value <SystemMode.Heat: 4> to custom {257: 1}`

Of course, I set the opposite on the device before testing from HA.

Is this normal?

mlouaze, 2023-08-22:
Just a question, do you also receive almost every 47ms a value for 0x0424 ?
Something like this: 

2023-08-22 12:03:48.426 DEBUG (MainThread) [zhaquirks.tuya] [0x1617:1:0xef00] Received value [1] for attribute 0x0424 (command 0x0002)
2023-08-22 12:03:48.427 DEBUG (MainThread) [homeassistant.components.zha.core.cluster_handlers] [0x1617:1:0x0201]: Attribute report 'MoesBHTThermostat'[running_mode] = RunningMode.Off
2023-08-22 12:03:48.428 DEBUG (MainThread) [homeassistant.components.zha.core.cluster_handlers] [0x1617:1:0x0201]: Attribute report 'MoesBHTThermostat'[running_state] = 0
2023-08-22 12:03:48.431 DEBUG (MainThread) [homeassistant.components.zha.entity] climate.tze204_aoclfnxz_ts0601_thermostat: Attribute 'running_mode' = RunningMode.Off update
2023-08-22 12:03:48.432 DEBUG (MainThread) [homeassistant.components.zha.entity] climate.tze204_aoclfnxz_ts0601_thermostat: Attribute 'running_state' = 0 update
2023-08-22 12:03:48.474 DEBUG (MainThread) [zhaquirks.tuya] [0x1617:1:0xef00] Received value [1] for attribute 0x0424 (command 0x0002)
2023-08-22 12:03:48.474 DEBUG (MainThread) [homeassistant.components.zha.core.cluster_handlers] [0x1617:1:0x0201]: Attribute report 'MoesBHTThermostat'[running_mode] = RunningMode.Off
2023-08-22 12:03:48.475 DEBUG (MainThread) [homeassistant.components.zha.core.cluster_handlers] [0x1617:1:0x0201]: Attribute report 'MoesBHTThermostat'[running_state] = 0
2023-08-22 12:03:48.477 DEBUG (MainThread) [homeassistant.components.zha.entity] climate.tze204_aoclfnxz_ts0601_thermostat: Attribute 'running_mode' = RunningMode.Off update
2023-08-22 12:03:48.478 DEBUG (MainThread) [homeassistant.components.zha.entity] climate.tze204_aoclfnxz_ts0601_thermostat: Attribute 'running_state' = 0 update

I find it really weird for the device to notify that often informations

bsfaxi, 2023-08-22:
on my installation, it's almost every second... :)

mlouaze, 2023-08-22:
That is surprising. I tried a lot of test to modify value but nothing seems to work. I seems that the value is sent correctly but the thermostat doesn't care at all. As if something else was mandatory to modify the value on it. I think the only way to figure this out would be to use the working z2m and reverse engineer was is send.

bsfaxi, 2023-08-22:
I tried a new dirty quirk to get rid off "MOESBHT_SCHEDULE_MODE_ATTR" and "MOESBHT_MANUAL_MODE_ATTR".
Now, I have only MOESBHT6_MODE_ATTR = 0x0402  # [0] manual [1] scheduled
I also changed some logic, like for example the inverted values, that are not relevant for this thermostat.
Here is the new quirk:
```
"""Map from manufacturer to standard clusters for electric heating thermostats."""
import logging

from zigpy.profiles import zha
import zigpy.types as t
from zigpy.zcl.clusters.general import Basic, Groups, Ota, Scenes, Time, GreenPowerProxy

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    TuyaManufClusterAttributes,
    TuyaThermostat,
    TuyaThermostatCluster,
    TuyaUserInterfaceCluster,
)

# info from https://github.com/zigpy/zha-device-handlers/pull/538#issuecomment-723334124
# https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/converters/fromZigbee.js#L239
# and https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/converters/common.js#L113
MOESBHT6_TARGET_TEMP_ATTR = 0x0210  # [0,0,0,21] target room temp (degree)
MOESBHT6_TEMPERATURE_ATTR = 0x0218  # [0,0,0,200] current room temp (decidegree)
MOESBHT6_MODE_ATTR = 0x0402  # [0] manual [1] scheduled
MOESBHT6_ENABLED_ATTR = 0x0101  # [0] off [1] on
MOESBHT6_RUNNING_MODE_ATTR = 0x0424  # [1] idle [0] heating /!\ inverted
MOESBHT6_CHILD_LOCK_ATTR = 0x0128  # [0] unlocked [1] child-locked

_LOGGER = logging.getLogger(__name__)

class MoesBHT6ManufCluster(TuyaManufClusterAttributes):
    """Manufacturer Specific Cluster of some electric heating thermostats."""

    attributes = {
        MOESBHT6_TARGET_TEMP_ATTR: ("target_temperature", t.uint32_t, True),
        MOESBHT6_TEMPERATURE_ATTR: ("temperature", t.uint32_t, True),
        MOESBHT6_MODE_ATTR: ("mode", t.uint8_t, True),
        MOESBHT6_ENABLED_ATTR: ("enabled", t.uint8_t, True),
        MOESBHT6_RUNNING_MODE_ATTR: ("running_mode", t.uint8_t, True),
        MOESBHT6_CHILD_LOCK_ATTR: ("child_lock", t.uint8_t, True),
    }

    def _update_attribute(self, attrid, value):
        super()._update_attribute(attrid, value)
        if attrid == MOESBHT6_TARGET_TEMP_ATTR:
            self.endpoint.device.thermostat_bus.listener_event(
                "temperature_change",
                "occupied_heating_setpoint",
                value * 100,  # degree to centidegree
            )
        elif attrid == MOESBHT6_TEMPERATURE_ATTR:
            self.endpoint.device.thermostat_bus.listener_event(
                "temperature_change",
                "local_temperature",
                value * 10,  # decidegree to centidegree
            )
        elif attrid == MOESBHT6_MODE_ATTR:
            if value == 0:  # manual
                self.endpoint.device.thermostat_bus.listener_event(
                    "program_change", "manual"
                )
[... 80 lines cut ...]
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.THERMOSTAT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    MoesBHT6ManufCluster,
                    MoesBHT6Thermostat,
                    MoesBHT6UserInterface,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        }
    }
```

bsfaxi, 2023-08-22:
Now, when I switch on/off from the device, I have data=[1,0] or data=[1,1].
But, when I try from HA, I have data=0 or data=1.
See below:
```
2023-08-22 22:29:07.726 DEBUG (MainThread) [zigpy.zcl] [0x3728:1:0xef00] Sending request: set_data(param=Command(status=0, tsn=207, command_id=257, function=0, data=0))
2023-08-22 22:34:02.048 DEBUG (MainThread) [zigpy.zcl] [0x3728:1:0xef00] Decoded ZCL frame: MoesBHT6ManufCluster:set_data_response(param=Command(status=3, tsn=86, command_id=257, function=0, data=[1, 0]))
2023-08-22 22:34:02.049 DEBUG (MainThread) [zigpy.zcl] [0x3728:1:0xef00] Received command 0x02 (TSN 233): set_data_response(param=Command(status=3, tsn=86, command_id=257, function=0, data=[1, 0]))
2023-08-22 22:36:15.600 DEBUG (MainThread) [zigpy.zcl] [0x3728:1:0xef00] Sending request: set_data(param=Command(status=0, tsn=148, command_id=257, function=0, data=1))
2023-08-22 23:26:04.942 DEBUG (MainThread) [zigpy.zcl] [0x3728:1:0xef00] Decoded ZCL frame: MoesBHT6ManufCluster:set_data_response(param=Command(status=13, tsn=158, command_id=257, function=0, data=[1, 1]))
2023-08-22 23:26:04.943 DEBUG (MainThread) [zigpy.zcl] [0x3728:1:0xef00] Received command 0x02 (TSN 238): set_data_response(param=Command(status=13, tsn=158, command_id=257, function=0, data=[1, 1]))
2023-08-22 23:26:17.012 DEBUG (MainThread) [zigpy.zcl] [0x3728:1:0xef00] Sending request: set_data(param=Command(status=0, tsn=194, command_id=257, function=0, data=0))
```

This is linked to this part of the quirk I think:
```
        if attribute == "system_mode":
            if value == self.SystemMode.Off:
                return {MOESBHT6_ENABLED_ATTR: 0}
            if value == self.SystemMode.Heat:
                return {MOESBHT6_ENABLED_ATTR: 1}
            self.error("Unsupported value for SystemMode")
```

Anyone could help here ?

bsfaxi, 2023-08-22:
Same when setting target temperature. The data is not a table!

```
2023-08-22 23:39:58.158 DEBUG (MainThread) [zigpy.zcl] [0x3728:1:0xef00] Decoded ZCL frame: MoesBHT6ManufCluster:set_data_response(param=Command(status=15, tsn=226, command_id=528, function=0, data=[4, 0, 0, 0, 29]))
2023-08-22 23:39:58.158 DEBUG (MainThread) [zigpy.zcl] [0x3728:1:0xef00] Received command 0x02 (TSN 108): set_data_response(param=Command(status=15, tsn=226, command_id=528, function=0, data=[4, 0, 0, 0, 29]))
2023-08-22 23:40:19.081 DEBUG (MainThread) [zigpy.zcl] [0x3728:1:0xef00] Sending request: set_data(param=Command(status=0, tsn=45, command_id=528, function=0, data=25))
```

mlouaze, 2023-08-22:
I think it has to do with the fact that the value sent isn't a uint32 coded on 4 bytes but a uint8 coded in 1 byte. But you also have to add the 4 at the beginning of the array.

mlouaze, 2023-08-22:
I will try to take a look at it tomorrow

bsfaxi, 2023-08-22:
For the switch on/off command_id=257, it's 1 bloc. So, it's normal to have uint8...
I found the below error:
```
2023-08-23 00:53:06.584 DEBUG (MainThread) [zhaquirks.tuya] [0x3728:1:0x0201] Mapping standard system_mode (0x001c) with value <SystemMode.Off: 0> to custom {257: 0}
2023-08-23 00:53:06.585 DEBUG (MainThread) [zigpy.zcl] [0x3728:1:0xef00] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=True, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), manufacturer=4417, tsn=58, command_id=0, *direction=<Direction.Server_to_Client: 0>)
2023-08-23 00:53:06.586 DEBUG (MainThread) [zigpy.zcl] [0x3728:1:0xef00] Sending request: set_data(param=Command(status=0, tsn=58, command_id=257, function=0, data=0))
2023-08-23 00:53:06.618 DEBUG (MainThread) [homeassistant.components.zha.core.cluster_handlers] [0x3728:1:0x0201]: wrote {'system_mode': <SystemMode.Off: 0>} attrs, Status: [[WriteAttributesStatusRecord(status=<Status.SUCCESS: 0>)]]
2023-08-23 00:53:06.619 DEBUG (MainThread) [homeassistant.components.zha.core.cluster_handlers] [0x3728:1:0x0201]: set system to SystemMode.Off
2023-08-23 00:53:06.623 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x3728), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=None, profile_id=260, cluster_id=61184, data=Serialized[b'\x18:\x0b\x00\x83'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=135, rssi=-77)
2023-08-23 00:53:06.624 DEBUG (MainThread) [zigpy.zcl] [0x3728:1:0xef00] Received ZCL frame: b'\x18:\x0b\x00\x83'
2023-08-23 00:53:06.624 DEBUG (MainThread) [zigpy.zcl] [0x3728:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=58, command_id=11, *direction=<Direction.Client_to_Server: 1>)
2023-08-23 00:53:06.625 DEBUG (MainThread) [zigpy.zcl] [0x3728:1:0xef00] Decoded ZCL frame: MoesBHT6ManufCluster:Default_Response(command_id=0, status=<Status.UNSUP_MANUF_CLUSTER_COMMAND: 131>)
2023-08-23 00:53:06.625 DEBUG (MainThread) [zigpy.zcl] [0x3728:1:0xef00] Received command 0x0B (TSN 58): Default_Response(command_id=0, status=<Status.UNSUP_MANUF_CLUSTER_COMMAND: 131>)
2023-08-23 00:53:07.589 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x3728), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=None, profile_id=260, cluster_id=61184, data=Serialized[b'\t\x15\x02\x1d\xe9$\x04\x00\x01\x01'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=135, rssi=-77)

```

bsfaxi, 2023-08-23:
Same error "UNSUP_MANUF_CLUSTER_COMMAND: 131" when trying to set target temperature.
Example below: target temperature 26 and then 27.

```
2023-08-23 09:31:04.101 DEBUG (MainThread) [zhaquirks.tuya] [0x3728:1:0x0201] Mapping standard occupied_heating_setpoint (0x0012) with value 2600 to custom {528: 26}
2023-08-23 09:31:04.104 DEBUG (MainThread) [zigpy.zcl] [0x3728:1:0xef00] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=True, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), manufacturer=4417, tsn=91, command_id=0, *direction=<Direction.Server_to_Client: 0>)
2023-08-23 09:31:04.105 DEBUG (MainThread) [zigpy.zcl] [0x3728:1:0xef00] Sending request: set_data(param=Command(status=0, tsn=91, command_id=528, function=0, data=26))
2023-08-23 09:31:04.566 DEBUG (MainThread) [homeassistant.components.zha.core.cluster_handlers] [0x3728:1:0x0201]: wrote {'occupied_heating_setpoint': 2600} attrs, Status: [[WriteAttributesStatusRecord(status=<Status.SUCCESS: 0>)]]
2023-08-23 09:31:04.573 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x3728), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=None, profile_id=260, cluster_id=61184, data=Serialized[b'\x18[\x0b\x00\x83'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=55, rssi=-87)
2023-08-23 09:31:04.574 DEBUG (MainThread) [zigpy.zcl] [0x3728:1:0xef00] Received ZCL frame: b'\x18[\x0b\x00\x83'
2023-08-23 09:31:04.575 DEBUG (MainThread) [zigpy.zcl] [0x3728:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=91, command_id=11, *direction=<Direction.Client_to_Server: 1>)
2023-08-23 09:31:04.577 DEBUG (MainThread) [zigpy.zcl] [0x3728:1:0xef00] Decoded ZCL frame: MoesBHT6ManufCluster:Default_Response(command_id=0, status=<Status.UNSUP_MANUF_CLUSTER_COMMAND: 131>)
2023-08-23 09:31:04.578 DEBUG (MainThread) [zigpy.zcl] [0x3728:1:0xef00] Received command 0x0B (TSN 91): Default_Response(command_id=0, status=<Status.UNSUP_MANUF_CLUSTER_COMMAND: 131>)

...
2023-08-23 09:40:39.766 DEBUG (MainThread) [zhaquirks.tuya] [0x3728:1:0x0201] Mapping standard occupied_heating_setpoint (0x0012) with value 2700 to custom {528: 27}
2023-08-23 09:40:39.767 DEBUG (MainThread) [zigpy.zcl] [0x3728:1:0xef00] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=True, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), manufacturer=4417, tsn=236, command_id=0, *direction=<Direction.Server_to_Client: 0>)
2023-08-23 09:40:39.768 DEBUG (MainThread) [zigpy.zcl] [0x3728:1:0xef00] Sending request: set_data(param=Command(status=0, tsn=236, command_id=528, function=0, data=27))
2023-08-23 09:40:39.793 DEBUG (MainThread) [homeassistant.components.zha.core.cluster_handlers] [0x3728:1:0x0201]: wrote {'occupied_heating_setpoint': 2700} attrs, Status: [[WriteAttributesStatusRecord(status=<Status.SUCCESS: 0>)]]
2023-08-23 09:40:39.801 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x3728), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=None, profile_id=260, cluster_id=61184, data=Serialized[b'\x18\xec\x0b\x00\x83'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=71, rssi=-85)
2023-08-23 09:40:39.801 DEBUG (MainThread) [zigpy.zcl] [0x3728:1:0xef00] Received ZCL frame: b'\x18\xec\x0b\x00\x83'
2023-08-23 09:40:39.802 DEBUG (MainThread) [zigpy.zcl] [0x3728:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=236, command_id=11, *direction=<Direction.Client_to_Server: 1>)
2023-08-23 09:40:39.802 DEBUG (MainThread) [zigpy.zcl] [0x3728:1:0xef00] Decoded ZCL frame: MoesBHT6ManufCluster:Default_Response(command_id=0, status=<Status.UNSUP_MANUF_CLUSTER_COMMAND: 131>)
2023-08-23 09:40:39.803 DEBUG (MainThread) [zigpy.zcl] [0x3728:1:0xef00] Received command 0x0B (TSN 236): Default_Response(command_id=0, status=<Status.UNSUP_MANUF_CLUSTER_COMMAND: 131>)
```

mlouaze, 2023-08-23:
I was able to control the thermostat remotely using z2m. I will try to see how datas are exchanged to mimic it in ZHA.

bsfaxi, 2023-08-23:
Below is another diiiiiiirty quirk. But, I can now control the on/off and the target temperature. 🥇 
There is only one missing thing: when I set a target temperature higher than the actual temperature, the thermostat is heating. But, his "running_mode" is not reflected into HA as "Heat" ! It still "idle". :-( 
Could someone test it please?
Ensure you delete the "__pycache__" subdirectory before restarting HA.

```
"""Map from manufacturer to standard clusters for electric heating thermostats."""
"""Tuya MCU based thermostat."""

import logging

from zigpy.profiles import zha
import zigpy.types as t
from zigpy.zcl.clusters.general import Basic, Groups, Ota, Scenes, Time, GreenPowerProxy

from typing import Dict, Optional, Union
from zigpy.zcl import foundation
from zigpy.zcl.clusters.hvac import Thermostat

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    TuyaManufClusterAttributes,
    TuyaThermostat,
    TuyaThermostatCluster,
    TuyaUserInterfaceCluster,
    NoManufacturerCluster,
    TUYA_MCU_COMMAND,
    TuyaLocalCluster,
)

from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    TuyaClusterData,
    # TuyaDPType,
    TuyaMCUCluster,
)

class TuyaTC(t.enum8):
    """Tuya thermostat commands."""
    OFF = 0x00
    ON = 0x01

class ZclTC(t.enum8):
    """ZCL thermostat commands."""
    OFF = 0x00
    ON = 0x01

TUYA2ZB_COMMANDS = {
    ZclTC.OFF: TuyaTC.OFF,
    ZclTC.ON: TuyaTC.ON,
}

# info from https://github.com/zigpy/zha-device-handlers/pull/538#issuecomment-723334124
# https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/converters/fromZigbee.js#L239
# and https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/converters/common.js#L113
MOESBHT6_TARGET_TEMP_ATTR = 0x0210  # [0,0,0,21] target room temp (degree)
MOESBHT6_TEMPERATURE_ATTR = 0x0218  # [0,0,0,200] current room temp (decidegree)
MOESBHT6_MODE_ATTR = 0x0402  # [0] manual [1] scheduled
MOESBHT6_ENABLED_ATTR = 0x0101  # [0] off [1] on
MOESBHT6_RUNNING_MODE_ATTR = 0x0424  # [1] idle [0] heating
MOESBHT6_CHILD_LOCK_ATTR = 0x0128  # [0] unlocked [1] child-locked

_LOGGER = logging.getLogger(__name__)
[... 163 lines cut ...]

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.THERMOSTAT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    MoesBHT6ManufCluster,
                    MoesBHT6Thermostat,
                    MoesBHT6UserInterface,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
			}
        }
    }
```

bsfaxi, 2023-08-23:
And finally a dirty but fully working quirk. The Sensor is working from "Idle" to "Heating" and vice-versa.
If someone could test it and confirm. ;)

```
"""Map from manufacturer to standard clusters for electric heating thermostats."""
"""Tuya MCU based thermostat."""

import logging

from zigpy.profiles import zha
import zigpy.types as t
from zigpy.zcl.clusters.general import Basic, Groups, Ota, Scenes, Time, GreenPowerProxy

from typing import Dict, Optional, Union
from zigpy.zcl import foundation
from zigpy.zcl.clusters.hvac import Thermostat

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    TuyaManufClusterAttributes,
    TuyaThermostat,
    TuyaThermostatCluster,
    TuyaUserInterfaceCluster,
    NoManufacturerCluster,
    TUYA_MCU_COMMAND,
    TuyaLocalCluster,
)

from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    TuyaClusterData,
    TuyaMCUCluster,
)

class TuyaTC(t.enum8):
    """Tuya thermostat commands."""
    OFF = 0x00
    ON = 0x01

class ZclTC(t.enum8):
    """ZCL thermostat commands."""
    OFF = 0x00
    ON = 0x01

TUYA2ZB_COMMANDS = {
    ZclTC.OFF: TuyaTC.OFF,
    ZclTC.ON: TuyaTC.ON,
}

MOESBHT6_TARGET_TEMP_ATTR = 0x0210  # [0,0,0,21] target room temp (degree)
MOESBHT6_TEMPERATURE_ATTR = 0x0218  # [0,0,0,200] current room temp (decidegree)
MOESBHT6_MODE_ATTR = 0x0402  # [0] manual [1] scheduled
MOESBHT6_ENABLED_ATTR = 0x0101  # [0] off [1] on
MOESBHT6_RUNNING_MODE_ATTR = 0x0424  # [1] idle [0] heating
MOESBHT6_RUNNING_STATE_ATTR = 0x0424  # [1] idle [0] heating
MOESBHT6_CHILD_LOCK_ATTR = 0x0128  # [0] unlocked [1] child-locked

_LOGGER = logging.getLogger(__name__)

class MoesBHT6ManufCluster(TuyaManufClusterAttributes, NoManufacturerCluster, TuyaLocalCluster):
    """Manufacturer Specific Cluster of some electric heating thermostats."""

    attributes = {
[... 174 lines cut ...]

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.THERMOSTAT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    MoesBHT6ManufCluster,
                    MoesBHT6Thermostat,
                    MoesBHT6UserInterface,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
			}
        }
    }
```

alexlinno, 2023-08-23:
Thank you for all your efforts! Excellent work. I can now, switch it on and off, and set the target temp

mlouaze, 2023-08-24:
Congrats, I think I will be staying with z2m since it was a nightmare to migrate and now everything is working pretty well, but thanks a lot for your work!

douglascrc-git, 2023-08-24:
> And finally a dirty but fully working quirk. The Sensor is working from "Idle" to "Heating" and vice-versa. If someone could test it and confirm. ;)
>
> ```
> """Map from manufacturer to standard [...]

Do you know if it works for cooling?

douglascrc-git, 2023-08-25:
> ### Problem description
> I bought this Moes Thermostat to control an electric underfloor heating system : https://fr.aliexpress.com/item/1005004996552864.html The device deosn't show ANY entity ! The existing quirk is only for TZE200 that has one Endpoint only. This one has 2.
> 
> ### Solution description
> Is it possible to adapt the existing tuya quirk "ts0601_electric_heating.py" to include this model ?
> 
> ### Screenshots/Video
> Screenshots/Video
> ### Device signature
> Device signature
> ### Diagnostic information
> Diagnostic information
> ### Logs
> Logs
> ### Custom quirk
> Custom quirk
> ### Additional information
> _No response_

I can't figure how to install the custom Quirk. 

So far, I have created a folder  /config/quirks and added this code in config.yaml:

```
zha:
  enable_quirks: true
  custom_quirks_path: /config/quirks            
s    
```        
Nevertheless, I don't know which name the quirk has to have.

bsfaxi, 2023-08-25:
Hello @douglascrc-git,
The thermostat I bought has only "heating" mode, not "cooling".
To try the custom quirk, you did the right configuration.
Please add the signature of your device at the end if it's different from mine!
Then restart your HA to take it into account.
The name of the file can be whatever you want, and should only end with ".py".
make sure to delete the "pycache" subdirectory under "quirks" before restarting.

douglascrc-git, 2023-08-25:
> Hello @douglascrc-git, The thermostat I bought has only "heating" mode, not "cooling". To try the custom quirk, you did the right configuration. Please add the signature of your device at the end if [...]

Thankyou! Really appreaciate all the work you have done! 

I just realice that your device is similar to my thermostat, however, mine's is dual (both heat, cool and fan). Also it can control a fan coil speed (Has 3 outputs for 3 speeds and two outpust for 2 valves [heat/coo]). 

In the other hand, your quirk worked correctly for heat operation, iddle and off. But still need the cooling operation and the control of the fan coil speed.

I will try to modify the quirk your published in order to make it work for my device.

bsfaxi, 2023-08-25:
I'm interested by your thermostat that does both cooling and heating.
Could you please send me the link where you bought it?
Thanks.

Deub31, 2023-08-25:
Thank you so much for your work!
It works well.
However, several functions are missing, such as weekly programming, manual or automatic mode, ...
I saw that all these functions are available recently on zigbee2mqtt.
How can they also be available on ZHA?
Is there a way of recovering the work already done on zigbee2mqtt and integrating it into ZHA?
Thanks for any help!

Deub31, 2023-08-25:
You can buy it here :
https://fr.aliexpress.com/item/1005004996552864.html

bsfaxi, 2023-08-25:
> Thank you so much for your work! It works well. However, several functions are missing, such as weekly programming, manual or automatic mode, ... I saw that all these functions are available [...]

We cannot "translate" from Z2M to ZHA, but inspire from there and vice-versa. ;)
I have already found the command for programming. See the beginning of this thread.
But, this needs some time for debugging and coding to expose this feature.
I'm thinking to adapt the scheduling program used [here](https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_trv.py) in an existing quirk.

douglascrc-git, 2023-08-25:
> I'm interested by your thermostat that does both cooling and heating. Could you please send me the link where you bought it? Thanks.

You can buy it here:  https://vi.aliexpress.com/item/1005005650680212.html?spm=a2g0o.detail.1000013.4.1eeajGnJjGnJJ5&gps-id=pcDetailBottomMoreThisSeller&scm=1007.13339.291025.0&scm_id=1007.13339.291025.0&scm-url=1007.13339.291025.0&pvid=7a7fae38-688e-451f-99e8-43741f211647&_t=gps-id:pcDetailBottomMoreThisSeller,scm-url:1007.13339.291025.0,pvid:7a7fae38-688e-451f-99e8-43741f211647,tpp_buckets:668%232846%238110%231995&isseo=y&pdp_npi=4%40dis%21EUR%2168.41%2132.15%21%21%2172.10%21%21%40211b423c16929636189772973e53d3%2112000033886533029%21rec%21ES%21%21A

The device in the link provided by @Deub31 is  different! That one is intented to floor heating.

Deub31, 2023-08-25:
@bsfaxi That would be great! At the very least, the link to the mode (auto, manual) is missing. You can always programme it directly on the device, but you need to be able to manage the operating mode. I don't know how I can help you but if you need I can try with my poor programming skills ;-)

LeFlairGoD, 2023-08-25:
> Und schließlich eine schmutzige, aber voll funktionsfähige Eigenart. Der Sensor arbeitet von „Leerlauf“ auf „Heizung“ und umgekehrt. Wenn es jemand testen und bestätigen könnte. ;)

Hi, I am still relatively new to HA.
Do I need to change other places in the code besides line 223
`    signature = {
        MODELS_INFO: [
            ("_TZE204_aoclfnxz", "TS0601"),`
to
`    signature = {
        MODELS_INFO: [
            ("_TZE204_5toc8efa", "TS0601"),`

Yop1403, 2023-08-26:
> > And finally a dirty but fully working quirk. The Sensor is working from "Idle" to "Heating" and vice-versa. If someone could test it and confirm. ;)

This finally works for me, thank you so much for your effort! I'm now getting these 2 sensors with my TS0601 / _TZE204_aoclfnxz:
[image]

A question: As the thermostat sensor is a combined sensor (temperature + HVAC action status) - how can I set a temperature in HA code? (I have an AirBnB apartment with some crazy guests sometimes switching to 29°C etc. - I want to set the temp. to 25°C if any temperature >25°C is set) - is that something anyone could help me with?

javicalle, 2023-08-27:
You should use the `climate.set_temperature` service:
https://www.home-assistant.io/integrations/climate/#service-climateset_temperature

Yop1403, 2023-08-28:
> You should use the `climate.set_temperature` service:
> https://www.home-assistant.io/integrations/climate/#service-climateset_temperature

You and this community are amazing - I've actually never come across this.

In case anyone is interested - this is my automation now - above 24°C the thermostat is automatically set to 24°C to avoid overheating of the apartment. Thanks @javicalle 

```
alias: Wohnzimmer_Automatisierung
description: ""
trigger:
  - platform: numeric_state
    entity_id: climate.tze204_aoclfnxz_ts0601_thermostat
    attribute: temperature
    above: 24
condition: []
action:
  - service: climate.set_temperature
    data:
      temperature: 24
    target:
      entity_id: climate.tze204_aoclfnxz_ts0601_thermostat
mode: single

LeFlairGoD, 2023-08-28:
> > Und schließlich eine schmutzige, aber voll funktionsfähige Eigenart. Der Sensor arbeitet von „Leerlauf“ auf „Heizung“ und umgekehrt. Wenn es jemand testen und bestätigen könnte. ;)
> 
> Hallo, ich bin noch relativ neu bei HA. Muss ich außer Zeile 223 noch andere Stellen im Code ändern? ` signature = { MODELS_INFO: [ ("_TZE204_aoclfnxz", "TS0601"),`
> 
> Zu
> 
> ` signature = { MODELS_INFO: [ ("_TZE204_5toc8efa", "TS0601"),`
> 
> Muss ich dies auch durch die Werte ersetzen, die in der Signatur meines Geräts stehen?
> 
> ```
> `replacement = {
>         ENDPOINTS: {
>             1: {
>                 PROFILE_ID: zha.PROFILE_ID,
>                 DEVICE_TYPE: zha.DeviceType.THERMOSTAT,
>                 INPUT_CLUSTERS: [
>                     Basic.cluster_id,
>                     Groups.cluster_id,
>                     Scenes.cluster_id,
>                     MoesBHT6ManufCluster,
>                     MoesBHT6Thermostat,
>                     MoesBHT6UserInterface,
>                 ],
>                 OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
>             },
>             242: {
>                 PROFILE_ID: 41440,
>                 DEVICE_TYPE: 97,
>                 INPUT_CLUSTERS: [],
>                 OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],`
> ```
> 
> mein Gerät:
> 
> ```
> {
>   "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
>   "endpoints": {
>     "1": {
>       "profile_id": "0x0104",
>       "device_type": "0x0051",
>       "input_clusters": [
>         "0x0000",
>         "0x0004",
>         "0x0005",
>         "0xef00"
>       ],
>       "output_clusters": [
>         "0x000a",
>         "0x0019"
>       ]
>     },
>     "242": {
>       "profile_id": "0xa1e0",
>       "device_type": "0x0061",
>       "input_clusters": [],
>       "output_clusters": [
>         "0x0021"
>       ]
>     }
>   },
>   "manufacturer": "_TZE204_5toc8efa",
>   "model": "TS0601",
>   "class": "zigpy.device.Device"
> }
> ```

I just cleared the log and reconfigured the device, and the following appears in the log:

Logger: zigpy.zcl
Source: runner.py:179
First occurred: 11:36:57 (15 occurrences)
Last logged: 11:37:00

[0x9EF4:1:0xef00] Unknown cluster command 36 b’\x9e\x00’
[0x9EF4:1:0xef00] Unknown cluster command 2 b’\x00\x9f\x18\x02\x00\x04\x00\x00\x00\xdc’
[0x9EF4:1:0xef00] Unknown cluster command 2 b’\x00\xa0f\x02\x00\x04\x00\x00\x00\x00’

**When reconnecting, the following appears:**

Logger: frontend.js.latest.202308021
Source: components/system_log/init.py:270
First occurred: 11:39:48 (2 occurrences)
Last logged: 11:39:49

[http://homeassistant.local:8123/frontend_latest/62575-6mNV-72kqCE.js:4:8121 1](http://homeassistant.local:8123/frontend_latest/62575-6mNV-72kqCE.js:4:8121) Uncaught TypeError: Cannot read properties of undefined (reading ‘get’)

Does anyone have any ideas?

MattWestb, 2023-08-29:
@LeFlairGoD ZHA have not loading the quirk for your device so its dont understanding the commands sent from cluster `0xef00`.
Look in the quirk that the device signature is matching your device if not its not being loaded.

LeFlairGoD, 2023-08-29:
> @LeFlairGoDZHA hat den Quirk für Ihr Gerät nicht geladen, daher versteht es die vom Cluster gesendeten Befehle nicht `0xef00`. Überprüfen Sie in der Eigenart, ob die Gerätesignatur mit Ihrem Gerät übereinstimmt, wenn nicht, wird es nicht geladen.

Made here. Do I need to change anything else?
https://github.com/zigpy/zha-device-handlers/issues/2433#issuecomment-1693578332

javicalle, 2023-08-29:
@LeFlairGoD yours seems to be a different device.
Not sure if already fixed but check here for support:
https://github.com/zigpy/zha-device-handlers/issues/2521

mrTuomoK, 2023-09-08:
Also just received my thermostats today and was already cursing them not showing up with entities. Then I found this thread and thank you very much @bsfaxi making a quirk that made it work 👍

Turntable67, 2023-09-14:
It would be nice to be able to read the external floorsensor that is supplied with the BHT-002GBLZB, now only the temperature of the unit itself (EndPoint id:1, id: 0x0201, type: in)  is read.

mrTuomoK, 2023-09-14:
I use only the floor sensor reading on the thermostat (you set it on the thermostat itself to use it) and now it should show the floor sensor reading. Of course if you use the integrated one or both, then I don't know?

zdeneksvarc, 2023-10-23:
> I use only the floor sensor reading on the thermostat (you set it on the thermostat itself to use it) and now it should show the floor sensor reading. Of course if you use the integrated one or both, then I don't know?

The thermostat disply (16A electrical underfloor heating version) shows the temperature only from the room sensor integrated in the thermostat. It does not show the value from the external 10k NTC sensor. Or am I missing something? The readings from the external sensor should definitely be available for safe operation of the heating system, as it protects the top of the floor against damage from overheating. I'll look into it as soon as I have a moment. Unless someone's faster.

[66 more replies]