zigpy/zha-device-handlers #2435: [Device Support Request] TS0011 by _TZ3000_7jsk6lxz (Mercator Ikuü Switch Mechanism)

greyman12, 2023-06-18:
### Problem description

Requesting support for Mercator Ikuü Switch Mechanism 

Product Data Sheet [https://www.ikuu.com.au/downloadable-assets/product-specifications/MIDS000025Z%20Data%20Sheet.pdf](url)

### Solution description

Update switch status without delay.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006"
      ],
      "output_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3000_7jsk6lxz",
  "model": "TS0011",
  "class": "ts0011.TuyaSingleNoNeutralSwitch_3"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

### Additional information

The switch will pair but the status of the switch is delayed in updating of the state, I don't know if this is because it is being reported as Power Source: Battery of Unknown. The device also only reports as an enddevice not sure if it is capable as a router, I do have the dimmer of the same brand [https://www.ikuu.com.au/product/switch-mechanism-3/](url) and it does report as a router.

javicalle, 2023-06-18:
I think that do you have a local quirk for that device.
Can you remove it and test it with the stock version?

greyman12, 2023-06-18:
You are quite right I was playing around with a local quirk which actually was just a copy of the stock. I have now removed the local quirk and updated the device signature, same issues still apply.

javicalle, 2023-06-19:
Can you check which value has the `switch_mode`. I can't recall if there is a device entity for these or you need to get it from the clusters attributes (attribute 0x8004 from the TuyaZBOnOffAttributeCluster).

> The device also only reports as an enddevice not sure if it is capable as a router

That behavior can't be modified.

greyman12, 2023-06-19:
Hi @javicalle I found the switch_mode attribute in TuyaZBOnOffAttributeCluster with a value of none.

javicalle, 2023-06-20:
Probably the device wouldn't accept it, but I would suggest to you to try to set a 0 value:
https://github.com/zigpy/zha-device-handlers/blob/7d6576888136b9d1c2b486f83e29c0e4c4d973b8/zhaquirks/tuya/__init__.py#L974-L978

Some devices can operate in 2 modes (command or event).
The `command` mode is faster than the `event` (but only triggers a few commands).

Other thing you can check is if there are some info in your logs.
Enable the debbug logging:
https://www.home-assistant.io/integrations/zha/#debug-logging

and check your logs while operating the device.
If the device reports are delayed, the quirk can't do nothing about that.

greyman12, 2023-06-21:
I attempted to set the value via the Manage Zigbee Device and as you suspected this failed. I did enable debug and looked through the logs but I could not see the device in the log when turning it on and off so not sending any status changes by the look of it.

javicalle, 2023-06-21:
You can try to enable logs from the ZHA integration (I forget that there is the option there):
https://user-images.githubusercontent.com/65863752/209584437-6f7e4c9b-43eb-41ca-9891-cc2470f92f0a.png

It seems to me that this is the behavior from the device but the logs should tell to us.

greyman12, 2023-06-21:
Turned the debug back on turned on the switch in Home Assistant and then manually turned off the switch and waited for the status to update. This is what I found in the log when it updated.

2023-06-21 20:05:21.510 DEBUG (MainThread) [zigpy.zcl] [0xEB5D:1:0x0006] Received ZCL frame: b'\x18\x8c\x01\x00\x00\x00\x10\x00'
2023-06-21 20:05:21.510 DEBUG (MainThread) [zigpy.zcl] [0xEB5D:1:0x0006] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=140, command_id=1, *direction=<Direction.Client_to_Server: 1>)
2023-06-21 20:05:21.510 DEBUG (MainThread) [zigpy.zcl] [0xEB5D:1:0x0006] Decoded ZCL frame: TuyaZBOnOffAttributeCluster:Read_Attributes_rsp(status_records=[ReadAttributeRecord(attrid=0x0000, status=<Status.SUCCESS: 0>,

javicalle, 2023-06-21:
How many time between the button press and the report?
It seems that device is not reporting the button press, just the device status:
https://forums.whirlpool.net.au/thread/3xkkrq89?p=4#r68

I don't know if there are any device configuration to change the behavior.

Tuya devices usually report the device status when you power it on. You can enable the debug logs and then power on the device. It will report all the device properties, maybe there is a property to configure the 'switch mode'.
Also, the Ikuu app will tell us the device capabilities but probably you will need to have a Ikuu gateway to connect the device.

greyman12, 2023-06-22:
Yes it does not seem to report in when turning on or off only periodically and that seems to very hit and miss. When you say report status when powering on, does than mean when power is first supplied to the device?

javicalle, 2023-06-22:
> When you say report status when powering on, does than mean when power is first supplied to the device?

Yes, I mean when the wires are connected to the device. Not an usual case of use, I know 😅

greyman12, 2023-06-22:
So my question is do I need to flip the breaker or can I remove the device and pair it again to see the device properties?

javicalle, 2023-06-23:
No need to pair again.
Just unplug the device from current and plug again. You can do it removing the power wires or if you have a separate circuit breaker for the device/line.
You need to keep the debug logging enabled during all the process to get the device power on report.

greyman12, 2023-07-05:
Thanks @javicalle I will hopefully get this done in the next couple of days.

greyman12, 2023-08-08:
Ended up trying with zigbee2mqtt and worked fine. Would still be good to get some ZHA support.

closed 2024-02-11 by github-actions[bot]