zigpy/zha-device-handlers #2436: [Device Support Request] loratap Zigbee Curtain/Light Switch TS0601 by _TZE204_pxbjch8m

Brembo427, 2023-06-19:
### Problem description

Hey Guys

is someone able to help with this device?

https://vi.aliexpress.com/item/1005005590013359.html?spm=a2g0o.order_list.order_list_main.39.24391802oXGwsJ&gatewayAdapt=glo2vnm

Added into ZHA no issues but with no entities, reports the device as TS0601
by _TZE204_pxbjch8m. 

Must be a new tuya chip im guessing that doesn't follow the usual standards.

### Solution description

added support for even just the light and or shutter cover.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

javicalle, 2023-06-19:
The issue template is there for a reason.
Attach as much information as possible to help maintainers to help you.

Brembo427, 2023-06-20:
Sorry I'm new to this, will try and fill it out as much as I can

closed 2024-02-02 by github-actions[bot]

guydvir2, 2024-10-07:
Same here.
any thing new ?