zigpy/zha-device-handlers #2437: Adding value for LiXee ZLinky_TIC

usky73, 2023-06-19:
### Problem description

This LiXee ZLinky_TIC is already integrated in ZHA but it is missing a critical parameter from Lixee device, it is the alarm message that are pushed by linky to advise about an overload.

This is the only solution for users to get quickly advised and to have time to remove some loads.

Is there a way to integrate this info.
I guess that it is in MSG1 parameter.

Is there a way to add it ?

### Solution description

add MSG1 in the ZHA integration

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

fairecasoimeme, 2023-06-23:
Hi,
the alarm message which is send when there is an overload is in the STGE label which is send to zigbee coordinator when it changes.
you have to look at this doc : Enedis-NOI-CPT_45E.pdf to understand how to get what you need.

usky73, 2023-06-24:
Hi, it is not on this doc.
anyway, even if I found, how to integrate it in ZHA ?

closed 2023-12-28 by github-actions[bot]