zigpy/zha-device-handlers #2439: [Device Support Request] Add Double clicks to Ikea Open/Close remote

okainov, 2023-06-22:
### Problem description

Ikea Open/Closed remote (https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/ikea/opencloseremote.py) doesn't report double clicks. It would be great to have ability to react on double clicks as well as on single ones to add more automation possibilities.

### Solution description

There should be possibility to choose "On Double Click Open"/"On Double Click Close" for the remote.

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

MattWestb, 2023-06-22:
You have 2 alternative to getting it working:
- 1 Then the device is not sending the command you can implementing it in your own custom firmware for it that is braking  all standard and zigbee certificate signing the firmware and so on.
-  2 Write your own logic in the automatons for getting the timing of the command of single press and stop interpreted and "making" double clicks you can using in your system.

Then the device is very "Zigbee standard" and you like doing it not its not one zha-quirk thing is one custom automation you is looking for.

okainov, 2023-06-22:
@MattWestb yeah, I have seen already some blog post making exactly option 2, I tried to make it working, got some kind of partial success, but I was thinking it would be great to have such a obvious possibility out-of-the-box to avoid custom automations...

MattWestb, 2023-06-22:
Im very sure is not one thaing that belongs in the ZHA-Quirks but perhaps in HA -Blueprints or one custom quirks if its possible implanting in them.
The OnOff and OpenClose is sharing the firmware and only have different data settings that make them sending light or clover commands very Zigbee standard. The OpenClose have implanting release from long press by remember the last long press and its made in the quirk. The only function that was not standard of IKEA controllers was the battery reporting and looks being fixed inn the last OTA for most devices so they dont need some quirk for working OK in ZHA but is not getting Device Automatons (DA) from the quirk is not loading  it.
If some line doing local quirk or bueprint for the function i think its great but i think is not going true the maintainers if try implanting in the ZHA-Quirks.
I think it  can being good for some user cases but in the ground is one blind controller and not one click click (ala Aqara / tuya) and shall being used for that purposes.

By the was the gen 2 2 button controllers is in the pipe (Rodert) and is having both OnOff, OpenClose and ShurtCutt buttons on the same hardware and is on some IKEA homepages (Sweden and some more) and i think its open one way converting them for sending more commands.

TheJulianJES, 2023-06-22:
We already have some logic for "virtual multi-presses" for the Hue remotes.
We could try to make that more generic and apply it to more/all remotes.

Doing this in HA with blueprints isn’t as easy as using a device trigger (I think?)

MattWestb, 2023-06-22:
> We already have some logic for "virtual multi-presses" for the Hue remotes. We could try to make that more generic and apply it to more/all remotes.
> 
> Doing this in HA with blueprints isn’t as easy as using a device trigger (I think?)

I was think ZHA was using the HUE special commands sent from the device and not using the Zigbee light commands.
https://github.com/zigpy/zha-device-handlers/blob/1e619d84a676347af3fa115f7f3af90f9550fc15/zhaquirks/philips/rdm001.py#L69
and then doing some "magic" from them.

TheJulianJES, 2023-06-22:
Also see: https://github.com/zigpy/zha-device-handlers/blob/1e619d84a676347af3fa115f7f3af90f9550fc15/zhaquirks/philips/__init__.py#L103-L220

okainov, 2023-12-19:
Nothing has changed here

okainov, 2024-06-16:
Come on, nothing has changed here and it would be still a nice addition.

MattWestb, 2024-06-16:
I think the easiest is getting one **IKEA SOMRIG shortcut button** its have short. double and long with release on both buttons (it can being IKEA is implanting more the its using Matter standard commands that is open for very many fast clicks).

closed 2024-12-20 by github-actions[bot]