zigpy/zha-device-handlers #2440: [BUG] Wrong battery status for ikea blinds and remotes

okainov, 2023-06-23:
### Bug description

Not sure why, but for all of my FURTUR/KADRIL blinds ZHA reports wrong battery status (opening/closing works fine). The same for at least one TRADFRI remote (see screenshots below)

I saw 255%, 120%, 146% and so on...

### Steps to reproduce

Not sure, for me it's always visible, so connect any TRADFRI or FYRTUR blind and check the battery reports?

### Expected behavior

Battery should be in the range 0-100

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

For one of the blinds:
```json

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4476, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0202",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0020",
        "0x0102",
        "0x1000",
        "0xfc7c"
      ],
      "output_clusters": [
        "0x0019",
        "0x1000"
      ]
    }
  },
  "manufacturer": "IKEA of Sweden",
  "model": "FYRTUR block-out roller blind",
  "class": "zhaquirks.ikea.blinds.IkeaTradfriRollerBlinds2"
}```

For one of remotes:
```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4476, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0203",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0009",
[... 743 lines cut ...]
            "unsupported_attributes": {}
          },
          "0x1000": {
            "endpoint_attribute": "lightlink",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

oscar-b, 2023-07-03:
#2418 ?

TheJulianJES, 2024-02-03:
This issue should be fixed in HA Core 2024.2.0 (currently in beta).
Relevant PR:
- https://github.com/zigpy/zha-device-handlers/pull/2948

If you still see new "double battery percentage" reports, wait a day.
You can also try to press a button on the device or manually read the `sw_build_id` attriubte. Re-pairing/reconfiguring should also work.

If it's not fixed for you, please create a new issue and provide a new diagnostics file.

closed 2024-02-03 by TheJulianJES