zigpy/zha-device-handlers #2442: [Device Support Request] Curtain Motor (Tuya, Zigbee) C-Type (3 Button) _TZE200_cpbo62rn

NemitkoIvan, 2023-06-26:
### Problem description

I bought new Roller Blind Cover here:
https://www.aliexpress.com/item/1005002954586484.html?spm=a2g0o.order_list.order_list_main.34.25771802eEeu0j

It has been successfully connected to Home Assistant, but there are no controls to control it.

### Solution description

I'd like to have ability at least to set min/max limits and open/close curtain

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

### Additional information

I found topic in zigbee2mqtt with some information for this device: https://github.com/Koenkk/zigbee2mqtt/issues/17436

mitchellsanders, 2023-10-17:
I think I've bricked mine. I have two of these, they were working fine, now one is having issues.

I'm hoping I can get the device working in Home Assistant. Right now I've added it to the smart things app and can't seem to delete the up and down position - the wheel only moves for a split second before stopping when attempting to relearn via the smart things app.

i8nemo, 2024-02-08:
these aren't that new, Im surprised no quirk exists for _TZE200_cpbo62rn TS0601

closed 2024-09-08 by github-actions[bot]

joshdaffern, 2024-10-15:
I also have this device and am looking for a quirk to handle it

salos1982, 2024-11-06:
Me too. I bought this device and could not do anything with it

1liminal1, 2025-03-05:
Did this even go anywhere?

lucasfirl, 2026-03-12:
https://gist.github.com/lucasfirl/3403fb85a8804e55a81d63901663251b

AnaRuxandra, 2026-09-03:
It drives me nuts, I forgot how to set it in calibration mode. Now it one moves one sec up or down and stops. Did anyone find the solution, please?