zigpy/zha-device-handlers #2443: [BUG] New Zigbee Zemismart / Tuya Light Smart Switches TS0004 and TS0003 switch on and off as a group in Home Assistant - Physical buttons work fine

incognitto-indago, 2023-06-26:
### Bug description

I have been using Zha in HA for a while and have more than 10 smart switches. Recently, I have bought new Zemismart/Tuya Zigbee Light Switches (TS0004 and TS0003). Unlike my previous ones, each one of these new ones are switching on and off as group when I use the entities in Home Assistant. If I press the physical buttons on the switch, they do work well, and they turn on and off just the one I pressed. If I try to turn on or off just one button via Home Assistant, all the buttons on that switch will also turn on or off.

I have same-model switches that work fine here, but they were bought about year ago.

### Steps to reproduce

1. Add the zigbee switch (TS0004 or TS0003) to the zha integration ("+ Add Device" button)
2. Turn on any entity on the switch in Home Assistant
3. All buttons/gangs turn on (relay turn on sound) and show they indeed have turned on with light-emitting diodes
4. In HA one can see that all entities/gangs/buttons on the switch are turned on too
5. Turn off any entity/gang/button on Home Assistant and they all turn off the same way.

### Expected behavior

If I turn on/off a single entity in Home Assistant, I expect to turn on/off only that respective one button at the switch

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>TS0004 Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe000",
        "0xe001"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe000",
        "0xe001"
      ],
      "output_clusters": []
    },
    "3": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0003",
        "0x0004",
[... 1349 lines cut ...]
        "in_clusters": {},
        "out_clusters": {
          "0x0021": {
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

TheJulianJES, 2023-06-26:
You can test this custom quirk for the `TS0004` ([how to install](https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274)):

(Copy + paste the Python stuff below into a file like `ts0004_custom_quirk.py`, put that in your custom quirk folder, restart HA, re-pair the switch and see if it works.)

<details>
<summary>Possible custom quirk to test</summary>

```python
"""tuya TS000X Switches (CUSTOM QUIRK TEST: https://github.com/zigpy/zha-device-handlers/issues/2443)."""

from zigpy.profiles import zgp, zha
from zigpy.zcl.clusters.general import (
    Basic,
    GreenPowerProxy,
    Groups,
    Identify,
    OnOff,
    Ota,
    Scenes,
    Time,
)

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODEL,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    TuyaZBE000Cluster,
    TuyaZBExternalSwitchTypeCluster,
    TuyaZBOnOffAttributeCluster,
)
from zhaquirks.tuya.mcu import EnchantedDevice

class Switch_4G_GPP_Var2(EnchantedDevice):
    """Tuya 4 gang switch module with."""

    signature = {
        MODEL: "TS0004",
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    0xE000,
                    0xE001,
                ],
                OUTPUT_CLUSTERS: [
                    Time.cluster_id,
                    Ota.cluster_id,
                ],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    0xE000,
                    0xE001,
                ],
                OUTPUT_CLUSTERS: [],
            },
            3: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    0xE000,
                    0xE001,
                ],
                OUTPUT_CLUSTERS: [],
            },
            4: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    0xE000,
                    0xE001,
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: zgp.PROFILE_ID,
                DEVICE_TYPE: zgp.DeviceType.PROXY_BASIC,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [
                    GreenPowerProxy.cluster_id,
                ],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaZBOnOffAttributeCluster,
                    TuyaZBE000Cluster,
                    TuyaZBExternalSwitchTypeCluster,
                ],
                OUTPUT_CLUSTERS: [
                    Time.cluster_id,
                    Ota.cluster_id,
                ],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaZBOnOffAttributeCluster,
                    TuyaZBE000Cluster,
                    TuyaZBExternalSwitchTypeCluster,
                ],
                OUTPUT_CLUSTERS: [],
            },
            3: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaZBOnOffAttributeCluster,
                    TuyaZBE000Cluster,
                    TuyaZBExternalSwitchTypeCluster,
                ],
                OUTPUT_CLUSTERS: [],
            },
            4: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaZBOnOffAttributeCluster,
                    TuyaZBE000Cluster,
                    TuyaZBExternalSwitchTypeCluster,
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: zgp.PROFILE_ID,
                DEVICE_TYPE: zgp.DeviceType.PROXY_BASIC,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [
                    GreenPowerProxy.cluster_id,
                ],
            },
        },
    }
```

</details>

incognitto-indago, 2023-06-27:
Hello.

Thanks for helping me.

About the custom quirk, after the restart, the quirk did not load properly, I believe. I got the following error in the logs:

`
Logger: zhaquirks
Source: custom_zha_quirks/ts0004_custom_quirk.py:3
First occurred: 8:14:51 AM (1 occurrences)
Last logged: 8:14:51 AM

Unexpected exception importing custom quirk 'ts0004_custom_quirk'
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/zhaquirks/__init__.py", line 454, in setup
    importer.find_module(modname).load_module(modname)
  File "<frozen importlib._bootstrap_external>", line 605, in _check_name_wrapper
  File "<frozen importlib._bootstrap_external>", line 1120, in load_module
  File "<frozen importlib._bootstrap_external>", line 945, in load_module
  File "<frozen importlib._bootstrap>", line 290, in _load_module_shim
  File "<frozen importlib._bootstrap>", line 721, in _load
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_zha_quirks/ts0004_custom_quirk.py", line 3, in <module>
    from zigpy.profiles import zgp, zha
ImportError: cannot import name 'zgp' from 'zigpy.profiles' (/usr/local/lib/python3.11/site-packages/zigpy/profiles/__init__.py)
`

MattWestb, 2023-06-27:
@incognitto-indago Is you running HA 2023.6.X ? 
The ZGP was implanted as one function in that and is needed for getting the ZGP cluster that your device is having.

TheJulianJES, 2023-06-27:
Ah, right. It'll only be in the next major HA version.

Try this modified version:

<details>
<summary>Possible custom quirk to test (for older HA versions)</summary>

```python
"""tuya TS000X Switches (CUSTOM QUIRK TEST: https://github.com/zigpy/zha-device-handlers/issues/2443)."""

from zigpy.profiles import zha
from zigpy.zcl.clusters.general import (
    Basic,
    GreenPowerProxy,
    Groups,
    Identify,
    OnOff,
    Ota,
    Scenes,
    Time,
)

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODEL,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    TuyaZBE000Cluster,
    TuyaZBExternalSwitchTypeCluster,
    TuyaZBOnOffAttributeCluster,
)
from zhaquirks.tuya.mcu import EnchantedDevice

class Switch_4G_GPP_Var2(EnchantedDevice):
    """Tuya 4 gang switch module with."""

    signature = {
        MODEL: "TS0004",
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    0xE000,
                    0xE001,
                ],
                OUTPUT_CLUSTERS: [
                    Time.cluster_id,
                    Ota.cluster_id,
                ],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    0xE000,
                    0xE001,
                ],
                OUTPUT_CLUSTERS: [],
            },
            3: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    0xE000,
                    0xE001,
                ],
                OUTPUT_CLUSTERS: [],
            },
            4: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    0xE000,
                    0xE001,
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaZBOnOffAttributeCluster,
                    TuyaZBE000Cluster,
                    TuyaZBExternalSwitchTypeCluster,
                ],
                OUTPUT_CLUSTERS: [
                    Time.cluster_id,
                    Ota.cluster_id,
                ],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaZBOnOffAttributeCluster,
                    TuyaZBE000Cluster,
                    TuyaZBExternalSwitchTypeCluster,
                ],
                OUTPUT_CLUSTERS: [],
            },
            3: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaZBOnOffAttributeCluster,
                    TuyaZBE000Cluster,
                    TuyaZBExternalSwitchTypeCluster,
                ],
                OUTPUT_CLUSTERS: [],
            },
            4: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaZBOnOffAttributeCluster,
                    TuyaZBE000Cluster,
                    TuyaZBExternalSwitchTypeCluster,
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }
```

</details>

incognitto-indago, 2023-06-27:
Yes, indeed, I am running 2023.6.3.

The enviroment is:

Home Assistant 2023.6.3
Supervisor 2023.06.2
Operating System 10.3
Frontend 20230608.0 - latest

incognitto-indago, 2023-06-27:
Will test it now.

> Ah, right. It'll only be in the next major HA version.
> 
> Try this modified version:
> 
> Possible custom quirk to test (for older HA versions)

MattWestb, 2023-06-27:
Your HA is having ziogpy 0.55.0 and the function was implanted in https://github.com/zigpy/zigpy/releases/tag/0.56.0 so i dont understanding way other quirks is working but your is not.
@TheJulianJES any clue ?

incognitto-indago, 2023-06-27:
It worked!!!!

> Will test it now.
> 
> > Ah, right. It'll only be in the next major HA version.
> > Try this modified version:
> > Possible custom quirk to test (for older HA versions)

incognitto-indago, 2023-06-27:
Now, can we apply the same reasoning for TS0003?

TheJulianJES, 2023-06-27:
Try to paste the below code at the end of your custom quirk file:

<details>
<summary>Possible additional code for TS0003</summary>

```python
class Switch_3G_GPP_Var2(EnchantedDevice):
    """Tuya 3 gang switch module."""

    signature = {
        MODEL: "TS0003",
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    0xE000,
                    0xE001,
                ],
                OUTPUT_CLUSTERS: [
                    Time.cluster_id,
                    Ota.cluster_id,
                ],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    0xE000,
                    0xE001,
                ],
                OUTPUT_CLUSTERS: [],
            },
            3: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    0xE000,
                    0xE001,
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaZBOnOffAttributeCluster,
                    TuyaZBE000Cluster,
                    TuyaZBExternalSwitchTypeCluster,
                ],
                OUTPUT_CLUSTERS: [
                    Time.cluster_id,
                    Ota.cluster_id,
                ],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaZBOnOffAttributeCluster,
                    TuyaZBE000Cluster,
                    TuyaZBExternalSwitchTypeCluster,
                ],
                OUTPUT_CLUSTERS: [],
            },
            3: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaZBOnOffAttributeCluster,
                    TuyaZBE000Cluster,
                    TuyaZBExternalSwitchTypeCluster,
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }
```

</details>

incognitto-indago, 2023-06-27:
It worked as well.

Thanks a lot again!

Should I do anything else? Will I always need the custom quirk?

> Try to paste the below code at the end of your custom quirk file:
> 
> Possible additional code for TS0003
> ```python
> class Switch_3G_GPP_Var2(EnchantedDevice):
>     """Tuya 3 gang switch module."""
> 
>     signature = {
>         MODEL: "TS0003",
>         ENDPOINTS: {
>             1: {
>                 PROFILE_ID: zha.PROFILE_ID,
>                 DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
>                 INPUT_CLUSTERS: [
>                     Basic.cluster_id,
>                     Identify.cluster_id,
>                     Groups.cluster_id,
>                     Scenes.cluster_id,
>                     OnOff.cluster_id,
>                     0xE000,
>                     0xE001,
>                 ],
>                 OUTPUT_CLUSTERS: [
>                     Time.cluster_id,
>                     Ota.cluster_id,
>                 ],
>             },
>             2: {
>                 PROFILE_ID: zha.PROFILE_ID,
>                 DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
>                 INPUT_CLUSTERS: [
>                     Identify.cluster_id,
>                     Groups.cluster_id,
>                     Scenes.cluster_id,
>                     OnOff.cluster_id,
>                     0xE000,
>                     0xE001,
>                 ],
>                 OUTPUT_CLUSTERS: [],
>             },
>             3: {
>                 PROFILE_ID: zha.PROFILE_ID,
>                 DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
>                 INPUT_CLUSTERS: [
>                     Identify.cluster_id,
>                     Groups.cluster_id,
>                     Scenes.cluster_id,
>                     OnOff.cluster_id,
>                     0xE000,
>                     0xE001,
>                 ],
>                 OUTPUT_CLUSTERS: [],
>             },
>             242: {
>                 PROFILE_ID: 41440,
>                 DEVICE_TYPE: 97,
>                 INPUT_CLUSTERS: [],
>                 OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
>             },
>         },
>     }
> 
>     replacement = {
>         ENDPOINTS: {
>             1: {
>                 PROFILE_ID: zha.PROFILE_ID,
>                 DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
>                 INPUT_CLUSTERS: [
>                     Basic.cluster_id,
>                     Identify.cluster_id,
>                     Groups.cluster_id,
>                     Scenes.cluster_id,
>                     TuyaZBOnOffAttributeCluster,
>                     TuyaZBE000Cluster,
>                     TuyaZBExternalSwitchTypeCluster,
>                 ],
>                 OUTPUT_CLUSTERS: [
>                     Time.cluster_id,
>                     Ota.cluster_id,
>                 ],
>             },
>             2: {
>                 PROFILE_ID: zha.PROFILE_ID,
>                 DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
>                 INPUT_CLUSTERS: [
>                     Identify.cluster_id,
>                     Groups.cluster_id,
>                     Scenes.cluster_id,
>                     TuyaZBOnOffAttributeCluster,
>                     TuyaZBE000Cluster,
>                     TuyaZBExternalSwitchTypeCluster,
>                 ],
>                 OUTPUT_CLUSTERS: [],
>             },
>             3: {
>                 PROFILE_ID: zha.PROFILE_ID,
>                 DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
>                 INPUT_CLUSTERS: [
>                     Identify.cluster_id,
>                     Groups.cluster_id,
>                     Scenes.cluster_id,
>                     TuyaZBOnOffAttributeCluster,
>                     TuyaZBE000Cluster,
>                     TuyaZBExternalSwitchTypeCluster,
>                 ],
>                 OUTPUT_CLUSTERS: [],
>             },
>             242: {
>                 PROFILE_ID: 41440,
>                 DEVICE_TYPE: 97,
>                 INPUT_CLUSTERS: [],
>                 OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
>             },
>         },
>     }
> ```

TheJulianJES, 2023-06-27:
If https://github.com/zigpy/zha-device-handlers/pull/2445 gets merged, the next `zha-quirks` release is made, the library gets bumped in the HA Core repo, and a new HA core version is released, the quirk will be included and you should delete your custom quirk.

We should be able to get it in for the next major HA beta releasing tomorrow (2023.7.x).

If anyone can have a quick look at the PR to see if everything's fine, that would be great too.

incognitto-indago, 2023-06-27:
Thanks! 

Should I close this issue now?

> If #2445 gets merged, the next `zha-quirks` release is made, the library gets bumped in the HA Core repo, and a new HA core version is released, the quirk will be included and you should delete your [...]

closed 2023-06-27 by TheJulianJES

airtonac, 2023-08-14:
> If #2445 gets merged, the next `zha-quirks` release is made, the library gets bumped in the HA Core repo, and a new HA core version is released, the quirk will be included and you should delete your [...]

Hello,

So I bought both switches (TS0003 and TS0004) from the Zemismart store on Ali Express. I managed to make the TS0004 work perfectly using the quirk given as example. I added a TS0004.py to the custom_zha_quirks folder and added this lines to configuration. yaml:

`zha:
  custom_quirks_path: /config/custom_zha_quirks/
  enable_quirks: true`

For some reason the quirk for the 3-gang version didn't work out of the box for me. Do I have to do something for ZHA to know I'm using a handler or the device should just work? Do I have to create any folder to be able to use handlers that were included to the main branch?  I've just moved to ZHA from Z2M, so I'm a bit puzzled.

I tried to do the same process I did to the 4-gang device to the 3-gang version, but the switches are still all grouped on HA. I even tried a TS000X.py file that contained instructions for both devices and still only the 4-gang one works as expected (switches turn on and off individually on HA). Please check below the file I replaced the TS0004.py with (TS000X.py):

FUNNY PART IS EVEN IF I DELETE ALL QUIRKS FROM THE custom_zha_quirks FOLDER AND ALSO CLEARED THE CACHE FOLDER, THE 4-GANG STILL WORKS PERFECTLY. (BUT NOT THE 3-GANG DEVICE) Very confusing...
Any help would be very much appreciated.

Thanks,
Airton

`"""tuya TS000X Switches."""

from zigpy.profiles import zgp, zha
from zigpy.zcl.clusters.general import (
    Basic,
    GreenPowerProxy,
    Groups,
    Identify,
    OnOff,
    Ota,
    Scenes,
    Time,
)

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODEL,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    TuyaZBE000Cluster,
    TuyaZBElectricalMeasurement,
    TuyaZBExternalSwitchTypeCluster,
    TuyaZBMeteringCluster,
    TuyaZBOnOffAttributeCluster,
)
from zhaquirks.tuya.mcu import EnchantedDevice

class Switch_1G_GPP(EnchantedDevice):
    """Tuya 1 gang switch module with restore power state support."""

    signature = {
        MODEL: "TS0001",
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=266 device_type=256
            # device_version=1
            # input_clusters=[0, 3, 4, 5, 6, 57344, 57345]
            # output_clusters=[10, 25]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
[... 1078 lines cut ...]
                    TuyaZBExternalSwitchTypeCluster,
                ],
                OUTPUT_CLUSTERS: [],
            },
            4: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaZBOnOffAttributeCluster,
                    TuyaZBE000Cluster,
                    TuyaZBExternalSwitchTypeCluster,
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: zgp.PROFILE_ID,
                DEVICE_TYPE: zgp.DeviceType.PROXY_BASIC,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }`

TheJulianJES, 2023-08-14:
@airtonac Please open a new issue for each switch, as it's easier to track (and provide the custom quirk code again).

The "spell" is only required once, hence the device now works without the quirk.
It should still have the quirk though, as the "spell" may get lost when re-pairing and/or power-cycling the device.

The three gang one might have an issue in the quirk (again, separate issue for this please).

PR #2445 (2023-06-27): Add Tuya `TS0003` and `TS0004` switch variants
## Proposed change

Adds Tuya switch variants (which require the "Tuya spell").
Fixes #2443

## Additional information

See #2443

## Checklist

- [x] The changes are tested and work correctly
- [x] `pre-commit` checks pass / the code has been formatted using Black
- [ ] Tests have been added to verify that the new code works