zigpy/zha-device-handlers #2447: Quirk for TZ3000_xkap8wtb TS000F

erkr, 2023-06-29:
### Problem description

I have a zigbee 3.0 wall switch with power monitoring. Switching and power readout work fine. 
The current readout is a factor 1000 to high though. Not sure if the usage readout is correct. Will report on that later.

### Solution description

I guess a quirk can translate the reading to the right value

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

This is the switch readout with a 40 watt bulb:
[image]
The current should be around 0.2A

</details>

### Device signature

<details><summary>Device signature</summary>

```json
Tbd
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

Na

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

Na

</details>
W

erkr, 2023-07-08:
Update: the total usage (kWh) readout is a factor 100 to high

erkr, 2023-07-17:
Update: I managed to make a custom quirk myself. The first time,  so not sure it is according to all the rules:
Fixes:
- Both 'RMS current' and 'Summation delivered' are scaled correctly (divided by a factor 1000 and 100) re-using existing Tuya metering classes.
- Changed the device from a light to a switch.

This gist contains the latest version of my custom quirk: https://gist.github.com/erkr/5d4addf1b4c8e18d68d2adb154c6be93

Maybe one the contributors can have a look and adopt my quirk in the official tuya list.

Best Eric

TheJulianJES, 2023-09-17:
Yep, quirk looks mostly good to me. Power factor entity shouldn't be created, as it's not supported, but it's a minor issue in ZHA for all power monitoring devices at the moment.
- I'd keep the custom cluster in the replacement, although it doesn't really matter.
- While I personally prefer to have the correct device type set in the quirk, it will break for people who currently use this switch semi-functional without a quirk (a new entity would be created). I'd revert that change. You can change this locally by including this in your `configuration.yaml`: https://www.home-assistant.io/integrations/zha/#modifying-the-device-type (IEEE + "-" + endpoint)

> Z2M additionally exposes switch type for this device and that is still missing here

That seems to be attribute `0xD030` on cluster `0xE001`. Try including `TuyaZBExternalSwitchTypeCluster` in the `replacement` and use the "Manage Zigbee device" dialog to check if writing that attribute (on that cluster) works.

erkr, 2023-09-18:
Thanks for reviewing my first custom quirk. Good feedback 🙏

The option to change the device type in configuration.yaml is new to me. Will use that to remove my custom quirk without breaking my statistics 😊

I will try to add TuyaZBExternalSwitchTypeCluster and report back if that works later this week👌

Best Eric

arsenyspb, 2023-10-01:
@erkr hola, and thank you for supporting the community and your ZHA contributions! I tried your custom quirk (...because I'm "lucky" to have same TZ3000_xkap8wtb as a pilot for in-line power meters I want to install for few circuits) but it either didn't pick up or mine didn't work. I still see ridiculous number of A in current and RMS is out of whack. Likely I didn't do something simple.

Anyways, do you plan doing a Pull Request to `zigpy:dev` similar to e.g. [this one](https://github.com/zigpy/zha-device-handlers/pull/2252)? If not, I can take over from you and tinker with it... 

Once again, huge thanks for your support! Looking forward to see your quirk working in main branch!

erkr, 2023-10-01:
@arsenyspb 
Thanks for trying my custom quirk. Those ridiculous values sound like the quirk is not really used (I guess you checked that)
I won't deliver the quirk to dev, as I lack the experience how to do that. 
So if you take that over from me would be great!
Please take the remark from Julian into account with respect to changing the device type (I changed it into a switch, Julian prefered to keep the original light type)
I hope it works for you as well!! Success. 
Eric

arsenyspb, 2023-10-01:
@erkr @TheJulianJES OK I think I've got it working, finally see sane numbers... As a reminder one needs to also delete wrong metrics' history (see [here](https://community.home-assistant.io/t/is-there-any-way-to-edit-a-sensors-history/519531/4)):

```
apk add sqlite
cd /config/
sqlite3 home-assistant_v2.db
```

now

```
sqlite> select * from states where metadata_id = (select metadata_id from states_meta where entity_id = 'sensor.<your_sensor_name_here>_summation_delivered');
sqlite> select * from states where metadata_id = (select metadata_id from states_meta where entity_id = 'sensor.<your_sensor_name_here>_rms_current');
```

This outputs the mad readings for power and Amps. Next, clean them up:

```
sqlite> delete from states where metadata_id = (select metadata_id from states_meta where entity_id = 'sensor.<your_sensor_name_here>_summation_delivered');
sqlite> delete from states where metadata_id = (select metadata_id from states_meta where entity_id = 'sensor.<your_sensor_name_here>_rms_current');
```

Profit.

arsenyspb, 2023-10-01:
@erkr Once again kudos to your contribution, I've tweaked few things and battled the linter, - waiting for LGTM on [pull/2613](https://github.com/zigpy/zha-device-handlers/pull/2613) , but please go ahead and see the final local quirk for cross-check, copy the code from [here](https://github.com/zigpy/zha-device-handlers/pull/2613/files).

erkr, 2023-10-01:
@arsenyspb  Great, thanks for organising a delivery. I think when the review remarks are processed it's ready to rock and roll 🤓
- The unknown cluster 0x1888 was added in the mean time by Julian.
- I will change the device type in configuration.Yaml as explained here https://www.home-assistant.io/integrations/zha/#modifying-the-device-type

erkr, 2023-10-01:
Nice it works for you as well😊

closed 2023-10-05 by TheJulianJES

erkr, 2023-10-21:
@arsenyspb It reached the 2023.10.4  release. Removed my custom version and it works great with the released one. Thx.

klibro, 2024-08-08:
I'm a bit confused, because from what I understand this fix made it into the 2023.10.4 release. However, I'm on 2024.8.0 and I have the same exact same symptoms with the Ts000f, namely current off by 1000 and energy used off by 100. Has this regressed or is there something I need to do to activate the fix?

erkr, 2024-08-09:
@klibro 
Delete the device and re-pairing can help. It should be using this quirk (device info -> zigbee info):

zhaquirks.tuya.ts000f_switch.Tuya_1G_Wall_Switch_Metering

klibro, 2024-08-09:
Yes, I realise that the quirk is part of the distribution and should be being picked up. The reported model is definitely TS000F, and that should be enough to trigger the quirk, shouldn't it? Other quirks for other ZHA devices I have are being picked up. I'm also successfully applying a custom quirk. I have tried copying the distribution quirk and putting it into my custom quirk directory, but still this is not being picked up after a restart. I have just installed two TS000F devices, so although I can believe one might have glitched, for two not to work seems like it is a deeper issue. Is there some way to troubleshoot why quirks don't get applied?

klibro, 2024-08-09:
I figured this out. The quirk in the release specifies a signature with endpoints 1 and 242, but upon inspecting my device, I found that it was only reporting a signature with the "1" endpoint. So the signature wasn't matching and hence the quirk wasn't being applied.
I copied the release quirk into my custom quirks directory and removed the 242 endpoint from the signature and also from the replacement (is that right?) and all seems to be working as expected now.

commit dc8d7448 (2023-10-05): Add 1CH mini Smart Switch Model FS-02ZE `_TS000F_xkap8wtb` (#2613)

Fix #2447

---------