zigpy/zha-device-handlers #2449: [BUG] GLEDOPTO GL-C-006P constantly re-pairing

MaxMSchneider, 2023-07-01:
### Bug description

Hey guys,

I recently got a GLEDOPTO GL-C-006P CCT Controller to upfit some light installation. Unfortunately, it does not pair right. For some reason it just keeps jumping back into pairing mode (sometimes). While it's connected, controls don't work. I can request some stuff via the "Manage Zigbee Device" menu.

### Steps to reproduce

1. Set Controller to pairing mode
2. Add new device in ZHA
3. wait

### Expected behavior

Pairing working as usual

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4687, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "11": {
      "profile_id": "0x0104",
      "device_type": "0x010c",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0x1000"
      ],
      "output_clusters": [
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "GLEDOPTO",
  "model": "GL-C-006P",
  "class": "zigpy.device.Device"
}```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Pairing LOG.txt](https://github.com/zigpy/zha-device-handlers/files/11926672/Pairing.LOG.txt)

```

</details>

TheJulianJES, 2023-07-01:
I think this might be the same cause as: https://github.com/Koenkk/zigbee2mqtt/issues/11202

Where and when did you buy your GL-C-006P?

MaxMSchneider, 2023-07-01:
@TheJulianJES I just got it off Amazon. Literally bought it two days ago.

MaxMSchneider, 2023-09-19:
@TheJulianJES 
Are there any news on this?

closed 2023-10-20 by MaxMSchneider