zigpy/zha-device-handlers #2451: [Device Support Request] Aqara Curtain Motor ZNCLDJ11LM

hastarin, 2023-07-02:
### Problem description

I'm using ZHA to try controlling this curtain motor and it seems to have a few issues.

1. The Cover is reporting the inverse state of "open"/"closed".
2. The Number control shows NaN and can't be dragged to a value.
3. It shows Occupancy and Opening sensors but neither seem to function. To my knowledge there is no Occupancy sensor and I'm unsure what an Opening sensor is meant to do but it never changed state in my testing.

### Solution description

Either correcting the open/closed state reporting and/or add a configuration option to invert it.

Hide the Occupancy/Opening sensors by default.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=0, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0202",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x000a",
        "0x000d",
        "0x0013",
        "0x0102",
        "0x0406"
      ],
      "output_clusters": [
        "0x0001",
        "0x0006",
        "0x000a",
        "0x000d",
        "0x0013",
        "0x0019",
        "0x0102",
        "0x0406"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.curtain",
[... 190 lines cut ...]
2023-07-02 17:25:17.088 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0xf5 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=AddressMode.NWK address=0x6738 endpoint=1>, status: 0x00
2023-07-02 17:25:19.911 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xEBA8](lumi.sen_ill.mgl01): Device seen - marking the device available and resetting counter
2023-07-02 17:25:19.911 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xEBA8](lumi.sen_ill.mgl01): Update device availability -  device available: True - new availability: True - changed: False
2023-07-02 17:25:20.202 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x0e6a000700aa00
2023-07-02 17:25:20.203 DEBUG (MainThread) [zigpy_deconz.api] Received command device_state_changed[<DeviceState.APSDE_DATA_INDICATION|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|130: 170>, 0]
2023-07-02 17:25:20.203 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.APSDE_DATA_INDICATION|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|130: 170>, 0]
2023-07-02 17:25:20.206 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, <DataIndicationFlags.Always_Use_NWK_Source_Addr: 1>)
2023-07-02 17:25:20.206 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x176a000800010001
2023-07-02 17:25:20.210 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x176a0028002100220200000102f7e90204010c000a0018450a5500396666674200afff9bb60e00be
2023-07-02 17:25:20.211 DEBUG (MainThread) [zigpy_deconz.api] Received command aps_data_indication[33, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=AddressMode.NWK address=0x0000>, 1, <DeconzAddress address_mode=AddressMode.NWK address=0xe9f7>, 2, 260, 12, b'\x18E\nU\x009ffgB', 0, 175, 255, 155, 182, 14, 0, -66]
2023-07-02 17:25:20.211 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [33, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=AddressMode.NWK address=0x0000>, 1, <DeconzAddress address_mode=AddressMode.NWK address=0xe9f7>, 2, 260, 12, b'\x18E\nU\x009ffgB', 0, 175, 255, 155, 182, 14, 0, -66]
2023-07-02 17:25:20.212 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xE9F7), src_ep=2, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=None, profile_id=260, cluster_id=12, data=Serialized[b'\x18E\nU\x009ffgB'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=255, rssi=-66)
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

hastarin, 2023-07-29:
I've made a start on the custom quirk but it's still not working the way I'd like so requires more work when time permits.

hastarin, 2023-07-29:
If anyone can tell me how to get the log output showing via Home Assistant it would be most appreciated. I thought just enabling debug logging for zhaquirks would do the trick but it's still not showing up no matter what I try putting in for the name of the logger.

hastarin, 2023-07-29:
Just adding some TS code for reference from https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/xiaomi.ts in case it helps me figure things out:
```ts
    {
        zigbeeModel: ['lumi.curtain'],
        model: 'ZNCLDJ11LM',
        description: 'Aqara curtain motor',
        vendor: 'Xiaomi',
        fromZigbee: [fz.xiaomi_basic, fz.xiaomi_curtain_position, fz.xiaomi_curtain_position_tilt],
        toZigbee: [tz.xiaomi_curtain_position_state, tz.xiaomi_curtain_options],
        onEvent: async (type, data, device) => {
            if (type === 'message' && data.type === 'attributeReport' && data.cluster === 'genBasic' &&
                data.data.hasOwnProperty('1028') && data.data['1028'] == 0) {
                // Try to read the position after the motor stops, the device occasionally report wrong data right after stopping
                // Might need to add delay, seems to be working without one but needs more tests.
                await device.getEndpoint(1).read('genAnalogOutput', ['presentValue']);
            }
        },
        exposes: [e.cover_position().setAccess('state', ea.ALL),
            e.binary('running', ea.STATE, true, false)
                .withDescription('Whether the motor is moving or not'),
            e.enum('motor_state', ea.STATE, ['stopped', 'opening', 'closing'])
                .withDescription('Motor state')],
        ota: ota.zigbeeOTA,
    },
```

hastarin, 2023-08-05:
Original comment has been updated with a working custom quirk. I'll need to clean it up at some point and get a pull request sorted.

hastarin, 2023-08-20:
No I'm guessing maybe you're got a newer variant?

[image]

mgwrd, 2023-08-20:
Yes, you're right. Mine seems to be manufacured in 2022 (HW-Versoin 17, app/ota-file version 36):
[image]

Programmierus, 2023-09-26:
Can confirm this quirk working with my device, thank you!

[image]

poldim, 2024-03-25:
I just migrated my curtains from the aqara gateway to ZHA and unfortunately ZHA is not correctly supporting these curtains.  I'm going to try this quirk. 

Is this difficult to PR into core HA?

gordio, 2024-08-16:
My version required some deletions about `PowerConfiguration.cluster_id` and `OccupancySensing.cluster_id`.
[image]
and final file looks like:
```
"""Aqara Curtain Driver B1 device."""
from __future__ import annotations

import logging
from typing import Any

from zigpy import types as t
from zigpy.profiles import zha
from zigpy.zcl import foundation
from zigpy.zcl.clusters.closures import WindowCovering
from zigpy.zcl.clusters.general import (
    AnalogOutput,
    Basic,
    Groups,
    Identify,
    MultistateOutput,
    OnOff,
    Ota,
    #PowerConfiguration,
    Scenes,
    Time,
)
from zigpy.zcl.clusters.measurement import OccupancySensing

from zhaquirks import CustomCluster
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.xiaomi import LUMI, XiaomiCustomDevice

_LOGGER = logging.getLogger(__name__)

class AnalogOutputB1(CustomCluster, AnalogOutput):
    """Analog output cluster, used to relay current_value to WindowCovering."""

    def __init__(self, *args: Any, **kwargs: Any) -> None:
        """Init."""
        _LOGGER.debug("AnalogOutputB1 Cluster init")
        super().__init__(*args, **kwargs)
        self._update_attribute(
            self.attributes_by_name["max_present_value"].id, 100.0
        )  # max_present_value
        self._update_attribute(
            self.attributes_by_name["min_present_value"].id, 0.0
        )  # min_present_value

    def _update_attribute(self, attrid: int, value: Any) -> None:
        _LOGGER.debug("AnalogOutput update attribute %04x to %s... ", attrid, value)
        super()._update_attribute(attrid, value)
        if attrid == self.attributes_by_name["present_value"].id:
            self.endpoint.window_covering._update_attribute(  # pylint: disable=protected-access
                WindowCovering.attributes_by_name[
                    "current_position_lift_percentage"
                ].id,
                (100 - value),
            )
            self.endpoint.on_off._update_attribute(  # pylint: disable=protected-access
                OnOff.attributes_by_name["on_off"].id, value > 0
            )

class WindowCoveringB1(CustomCluster, WindowCovering):
[... 126 lines cut ...]
        },
    }
    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.WINDOW_COVERING_DEVICE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    #PowerConfiguration.cluster_id,
                    OnOffB1,
                    AnalogOutputB1,
                    WindowCoveringB1,
                ],
                OUTPUT_CLUSTERS: [
                    OnOffB1,
                    Time.cluster_id,
                    AnalogOutputB1,
                    Ota.cluster_id,
                    WindowCoveringB1,
                ],
            },
        },
    }
```

gordio, 2024-08-16:
AND I change one parameter in Manage Zigbee device:
1. Clusters tab;
2. Select AnalogOutputB1 (Endpoint id: 1, Id: 0x000d, Type: in) in Clusters dropdown;
3. Select `max_present_value (id: 0x0041)` in Attributes of the selected cluster;

poldim, 2024-09-27:
> AND I changed one parameter in **Manage Zigbee device**:
> 
> 1. Clusters tab;
> 2. Select AnalogOutputB1 (Endpoint id: 1, Id: 0x000d, Type: in) in _Clusters_ dropdown;
> 3. Select `max_present_value (id: 0x0041)` in _Attributes of the selected cluster_;
> 4. Press READ ATTRIBUTE;
> 5. If Value loaded as 0 -> manually change to 100 and press WRITE ATTRIBUTE.

What did doing this fix/solve?

poldim, 2024-09-27:
> Original comment has been updated with a working custom quirk. I'll need to clean it up at some point and get a pull request sorted.

@hastarin did you do any more cleanup / change to this quirk?

- I've been using the custom quirk from your OP for the last year on 6 of these curtains.  
- It's generally working fine with a small issue that **sometimes** the curtains continue to report "opening" or "closing" indefinitely after they've been commanded to do something.  
- I'm also surprised that I didn't have to comment out `PowerConfiguration` & `OccupancySensing` as others have.  Wondering if I should try commenting these out...?

My device signature:
```json
{
  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 4447,
    "maximum_buffer_size": 127,
    "maximum_incoming_transfer_size": 100,
    "server_mask": 0,
    "maximum_outgoing_transfer_size": 100,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0202",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0006",
        "0x000d",
        "0x0102"
      ],
      "output_clusters": [
        "0x0006",
        "0x000a",
        "0x000d",
        "0x0019",
        "0x0102"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.curtain",
  "class": "aqara-curtain.CurtainB1"
}
```

Photo:
[image]

Device in HA:

[image]

hastarin, 2024-09-27:
No I haven't as it's been working for me and I barely understood things enough to do that at the time so I'm sticking with the old, if it ain't broke don't fix it.

ZHA used to show a switch which vanished with an update sometime ago now but I just changed anything that was using it and moved on.

You can certainly try commenting those things out it obviously doesn't have an occupancy sensor but I have no idea what power configuration might do.

Seeing that inverted option in your screenshot does make me wonder if perhaps it might work without the quirk and just setting that instead in the current ZHA but again if it ain't broke...

poldim, 2024-09-30:
Good stuff...

Also, I'm not sure what **identify** does, I don't see it doing anything noticeable

muzzy124, 2025-03-20:
I use this custom quirk but I can't get 'opening' sensor to work. It's always closed. Does it work anyway?

muzzy124, 2025-03-20:
how to add this part of z2m custom converter to this quirk?
e.enum('motor_state', ea.STATE, ['stopped', 'opening', 'closing'])
                .withDescription('Motor state')],